zephyrproject-rtos/zephyr #36928: Disconnecting ISO mid-send giver error in hci_num_completed_packets

Thalley, 2021-07-13:
**Describe the bug**
If we are in the middle of sending ISO data on an iso channel (i.e. we have sent data, but have not yet received the num completed packets) when an ISO (or an ACL) disconnects (i.e. we get a disconnect complete event from the controller), we see the following fail: 

```
		conn = bt_conn_lookup_handle(handle);
		if (!conn) {
			BT_ERR("No connection for handle %u", handle);
			continue;
		}
```

There's nothing stopping us from deallocating iso channels while there is an ongoing write. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create ACL and CIS connection between two devices with a low ISO interval
2. Reset or move one of the device out of range to trigger a disconnect
3. See the error in the log

**Expected behavior**
The expected behavior would be to set the iso channels in the `BT_ISO_DISCONNECT` state until we have received the num complete packet of a pending write, and then release/deallocate the iso channel. 

**Impact**
Low; the only problem caused is that the log will contain and error message. 

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - Commit SHA or Version used: 36abc60b5109b08a270c7df4e829bf67ecf3b96a

**Additional context**
Add any other context about the problem here.

Thalley, 2021-07-13:
I couldn't (haven't yet) figured out how this is done for ACL connections, but I would assume that a solution would require taking a reference count when sending and dereference it when sent. 

@Vudentz @jhedberg @joerchan Do you have any suggestions, or input on how this is done for ACL connections?

jhedberg, 2021-07-13:
@Thalley it's handled here: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/conn.c#L1454

Thalley, 2021-08-02:
I'm taking a look at this issue right now

Thalley, 2021-08-04:
@jhedberg @joerchan Been looking at this issue for a while. I can confirm that the state handling of ISO connections works similar to ACL connections, and that `process_unack_tx` is called (as @jhedberg referenced in https://github.com/zephyrproject-rtos/zephyr/issues/36928#issuecomment-879229964). 

However, I don't see how that function, or any other function anywhere, ensures that all `bt_hci_evt_num_completed_packets` are processed before we set the state of the connection to `BT_CONN_DISCONNECTED`. 

The issue is basically that we get a `bt_hci_evt_num_completed_packets` event after we get the disconnect complete event. I am starting to wonder if this is correct behavior of the controller (**OBS!** a 3rd party controller in this case!), since the `bt_hci_evt_num_completed_packets` contains a connection handle, and the host typically will consider a handle invalid after the disconnect complete event. 

I can't find anything in the core spec that explicitly states whether this is okay for the controller to do, or not. 

Do you guys have any input?

Thalley, 2021-08-04:
Update: Found this little nifty sentence in the core spec:
> The HCI_Number_Of_Completed_Packets event **shall not specify a given Connection_Handle** before the event indicating the corresponding connection has been created or **after an event indicating disconnection of the corresponding connection**.

So my conclusion is that this is an issue with the controller. If no objections, I'll close this issue.

Thalley, 2021-08-05:
Confirmed controller issue, closing.

closed 2021-08-05 by Thalley