zephyrproject-rtos/zephyr #36947: Tensorflow: Dedicated tflite-micro repository

JordanYates, 2021-07-14:
## Purpose

The tensorflow organisation has recently (June 25th) split out the tensorflow micro code into a dedicated repository, instead of being part of the parent tensorflow repo. The dedicated repository is considerably smaller (5.85MB instead of 680MB), and is where future development will take place. The old location of the tflite-micro code in the parent repository has been removed (https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro).

https://github.com/tensorflow/tensorflow
https://github.com/tensorflow/tflite-micro

We should swap over from the tensorflow fork (https://github.com/zephyrproject-rtos/tensorflow) to a fork of the tflite-micro repository.

carlescufi, 2021-07-14:
@laurenmurphyx64 is this something that you could take care of?

carlescufi, 2021-07-14:
No objections from the TSC.

galak, 2021-07-14:
I went ahead and created https://github.com/zephyrproject-rtos/tflite-micro

laurenmurphyx64, 2021-07-27:
While working on this issue, I found that the Zephyr build for the examples in the new TFLM repo is broken due to recent changes made to the TF build system files that generate the CMakeLists.txt used by the Zephyr build system. I've filed an [issue](https://github.com/tensorflow/tflite-micro/issues/332) and submitted a [PR](https://github.com/tensorflow/tflite-micro/pull/333). I am now waiting for Google / Intel to legally approve me as a contributor.

laurenmurphyx64, 2021-07-29:
After discussion in my PR to TensorFlow, I'm OK with making the Zephyr module. 

The module PR is here: https://github.com/zephyrproject-rtos/tflite-micro/pull/1. 
The Zephyr PR is here: https://github.com/zephyrproject-rtos/zephyr/pull/37318/

PR #37318 (2021-08-24): Add Tensorflow Lite Micro as a module
Adds TensorFlow Lite Micro as a module and makes some updates to reflect the change from using the larger TensorFlow repo to the new, smaller TensorFlow Lite Micro repo. Cannot be merged until [the module PR enabling Zephyr support](https://github.com/zephyrproject-rtos/tflite-micro/pull/1) is merged.

Fixes  #36947
closed 2021-08-24 by cfriedt