zephyrproject-rtos/zephyr #36948: Cluttering of logs on USB Console in Zephyr when CDC Shell is enabled

Bee-Vision, 2021-07-14:
I am experiencing the cluttering of logs on the USB Console NRF5340. I have already turned off the colors in prj.conf file, but still getting "[8D[J". Might be some escape sequence or so and this only happens if SHell is enabled with the USB console.
Here are my prj.conf file configurations

```
CONFIG_LOG=y
CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_BACKEND_SHOW_COLOR=n

CONFIG_USB=y
CONFIG_SERIAL=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_USB_DEVICE_PRODUCT="uart:~$ "

CONFIG_UART_LINE_CTRL=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
**CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"**
CONFIG_SHELL_VT100_COLORS=n
```

What have you tried to diagnose or workaround this issue?
If `CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0" ` USB shell is enabled then it's printing the number of characters we have in the UART prompt. By default it's CONFIG_SHELL_PROMPT_UART="uart:~$ ". So, it's printing [8D[J in the starting of every line.

It's reproducible with shell_module sample code on nrf5340dk kit.
1. Add the following conf in prj.conf (samples/subsys/shell/shell_module/)
```
CONFIG_USB=y
CONFIG_SERIAL=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_USB_DEVICE_PRODUCT="USB device"
CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_BACKEND_SHOW_COLOR=n
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"
CONFIG_SHELL_VT100_COLORS=n
CONFIG_SHELL_PROMPT_UART="uart:~$ "
```
2. Add the following code in the main.c  (samples/subsys/shell/shell_module/src)
#include <usb/usb_device.h>
void main(void)
{
    if (usb_enable(NULL)) {
        return;
    }
}

**Logs and console output**
[image]

**Impact**
The USB logs are important for a data logger application and cluttering of logs will be the showstopper of our product. As the user can't understand the current senor details of the device. 

**Environment:**
 - OS: (MacOS and Windows)
 - Toolchain (Zephyr SDK v2.4.99-ncs-branch )
 - Commit ID 4ce908a2342494c19df09f347ce79ab81b415185

jfischer-no, 2021-07-14:
My comment from slack chat:
>This is not related to VT100 color mode, or VT100 commands. This "cluttering" is coming from the host/terminal echoed back, probably caused by prompt printing before DTR-RTS. I am not sure what people would prefer, to drop or to buffer data until DTR signal.

jfischer-no, 2021-07-14:
>If CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0" USB shell is enabled...

>The USB logs are important for a data logger application and cluttering of logs will be the showstopper of our product.

Is the shell support mandatory or just logging over CDC ACM?

Bee-Vision, 2021-07-14:
Shell is mandory - for Sensor,Modem, GPIO access

carlescufi, 2021-07-15:
@jakub-uC could you also give your opinion here?

jakub-uC, 2021-07-15:
It looks like shell is not receiving all characters hence escape codes are not handled correctly. I would expect we have some bug in the backend. 
@nordic-krch  are you familiar with the USB backend?

jfischer-no, 2021-07-15:
> It looks like shell is not receiving all characters hence escape codes are not handled correctly. I would expect we have some bug in the backend.
> @nordic-krch are you familiar with the USB backend?

Backend is serial or serial over logging, if enabled. Maybe it makes sense to have a DTR check in serial backend.

Bee-Vision, 2021-07-15:
@jfischer-no - Do u mean, I have to change shell_uart.c?

carlescufi, 2021-07-16:
@jakub-uC and @nordic-krch maybe the shell and logging backends for UART should be optionally configured to wait for DTR before sending any data, like it is being done in this sample:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/usb/cdc_acm/src/main.c#L97

jakub-uC, 2021-07-16:
@carlescufi My understanding is that we have a problem with bytes reception by the shell (UART backend). The proposal you have linked seems to be solving a different issue where the shell receiver is not capable of receiving data. Or I am missing something basic here?

carlescufi, 2021-07-16:
> It looks like shell is not receiving all characters hence escape codes are not handled correctly. I would expect we have some bug in the backend.
> @nordic-krch are you familiar with the USB backend?

I don´t get it. The escape characters are sent by the shell, not received? so it looks to me like there's some characters sent by the shell backend to the computer that never make it through?

carlescufi, 2021-07-16:
@jakub-uC it might be the garbling actually comes from the terminal misunderstanding a sequence being sent by the shell because the shell sends it too early. Could  you try waiting for DTR before sending any output from the shell to the terminal so that the synch is kept?

jakub-uC, 2021-07-16:
@carlescufi 
Sure I will implement this shortly.

jfischer-no, 2021-07-16:
[image]

Trace from the beginning of the communication.

Bee-Vision, 2021-07-21:
@jfischer-no did you get a chance to look at it?

jfischer-no, 2021-07-22:
> @jfischer-no did you get a chance to look at it?

It is not what should be fixed in CDC ACM UART driver. The idea is to let shell subsystem to check DTR signal.
If CDC ACM is the only one class you have in your application, the workaround could be to enable USB device support before the shell system.

carlescufi, 2021-09-01:
> @carlescufi
> Sure I will implement this shortly.

@jakub-uC ping

jakub-uC, 2021-09-01:
@carlescufi We have tested the solution today and it is working. PR shall be today or tomorrow.

PR #38203 (2021-09-02): shell: uart: Add waiting on DTR signal before sending data
Problem:
In some cases, as described in: https://github.com/zephyrproject-rtos/zephyr/issues/36948
shell backend sends characters to output before serial device
is ready for it. It results in observing additional characters
inserted on the shell input after device boot.

Solution:
Added waiting on DTR signal before sending anything to the output.

Fixes: #36948
commit e9f23888 (2021-09-02): shell: uart: Add waiting on DTR signal before sending data

Problem:
In some cases, as described in: https://github.com/zephyrproject-rtos/zephyr/issues/36948
shell backend sends characters to output before serial device
is ready for it. It results in observing additional characters
inserted on the shell input after device boot.

Solution:
Added waiting on DTR signal before sending anything to the output.

Fixes: #36948
closed 2021-09-02 by nashif