zephyrproject-rtos/zephyr #36953: <err> lorawan: MlmeConfirm failed : Tx timeout

Andermutu, 2021-07-14:
**Describe the bug**
I have built LoRaWAN class_a sample and run on my Actinius Icarus Board connected via SPI to a sx1276. I have my own gateway(RAK 7243) very close. The gateway receives the uplink message and sends a downlink message as shown in this picture:

[image]

On the node live data I can see that join requests is accepted:

[image]

However, on my console output I get:

> *** Booting Zephyr OS build v2.6.0-rc1-ncs1  ***
> [00:00:00.200,286] <inf> sx127x: SX127x version 0x12 found
> [00:00:00.266,662] <inf> lorawan_class_a: Joining network over OTAA
> [00:00:04.321,166] <err> lorawan: MlmeConfirm failed : Tx timeout
> [00:00:04.321,228] <err> lorawan_class_a: lorawan_join_network failed: -116

I have found that this [issue](https://github.com/zephyrproject-rtos/zephyr/issues/29581) is similar to mine, however, I can not found a solution there.

My source code is class_a source code, only modifying app_eui, dev_eui and app_key.

My .overlay file is this one: 

> / {
> 	aliases {
> 		lora0 = &lora;
> 	};
> };
> 
> &spi3 {
> compatible = "nordic,nrf-spim";
> status = "okay";
> sck-pin = <20>;
> mosi-pin = <21>;
> miso-pin = <22>;
> cs-gpios = <&gpio0 19 GPIO_ACTIVE_LOW>;
> 
> lora: sx1276@0 {
> 	compatible = "semtech,sx1276";
> 	reg = <0>;
> 	label = "sx1276";
> 	reset-gpios = <&gpio0 30 GPIO_ACTIVE_LOW>;
> 	dio-gpios = <&gpio0 31 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
> 	spi-max-frequency = <1000000>;
> 	power-amplifier-output = "pa-boost";		
> };
> };
> 

Everything that is in the .overlay file is connected like that except the dio-gpios pin that is unconnected. If you need more detailed information please ask.

**To Reproduce**
Steps to reproduce the behavior:
1. build class_a LoRaWAN sample.
2. view behaviour.

**Expected behavior**
I expect the LoRa module to connect to the gateway and transmit data.

**Impact**
I am stuck building the LoRaWAN node application.

**Environment (please complete the following information):**
 - OS:  MacOS Catalina.
 - Toolchain : nRF Connect SDK v1.6.0, latest available.

tagunil, 2021-07-14:
As far as I know, DIO0 pin is actually used to signal end of transmission. Without it being connected the transmission will time out.
I think one also need DIO1 pin to handle reception timeouts.

Andermutu, 2021-07-15:
> As far as I know, DIO0 pin is actually used to signal end of transmission. Without it being connected the transmission will time out.
> I think one also need DIO1 pin to handle reception timeouts.

Hi, after connecting DIO0 and DIO1 pins OTAA works,  and I can see data in TTN.  However, ABP mode still gives me an error although I can also see the data in TTN. This is ABP sample output:

> [00:00:00.200,286] <inf> sx127x: SX127x version 0x12 found
> [00:00:00.267,883] <inf> lorawan_class_a: Joining network over ABP
> [00:00:00.268,096] <inf> lorawan_class_a: New Datarate: DR_0, Max Payload 51
> [00:00:00.268,096] <inf> lorawan: Datarate changed: DR_0
> [00:00:00.268,127] <inf> lorawan_class_a: Sending data...
> [00:00:06.432,495] <err> lorawan: McpsRequest failed : Rx 2 timeout
> [00:00:06.432,556] <err> lorawan_class_a: lorawan_send failed: -116

Does not really care for me as OTAA is more interesting.

closed 2021-07-16 by Andermutu

tcpipchip, 2022-11-08:
i am having this same problem...

[00:25:56.699,000] <dbg> sx126x: SX126xWriteRegisters: Writing 1 registers @ 0x736: 0x9 , ...
[00:25:56.699,000] <dbg> sx126x: SX126xWriteCommand: Issuing opcode 0x8c w. 6 bytes of data
[00:25:56.700,000] <dbg> sx126x: SX126xWriteCommand: Issuing opcode 0x8 w. 8 bytes of data
[00:25:56.701,000] <dbg> sx126x: SX126xSetOperatingMode: SetOperatingMode: RX (5)
[00:25:56.701,000] <dbg> sx126x: SX126xWriteRegisters: Writing 1 registers @ 0x8ac: 0x94 , ...
[00:25:56.701,000] <dbg> sx126x: SX126xWriteCommand: Issuing opcode 0x82 w. 3 bytes of data
[00:25:56.919,000] <dbg> sx126x: sx126x_dio1_irq_work_handler: Processing DIO1 interrupt
[00:25:56.919,000] <dbg> sx126x: SX126xReadCommand: Issuing opcode 0x12 (data size: 2)
[00:25:56.919,000] <dbg> sx126x: SX126xReadCommand: -> status: 0xa6
[00:25:56.919,000] <dbg> sx126x: SX126xWriteCommand: Issuing opcode 0x2 w. 2 bytes of data
[00:25:56.919,000] <dbg> sx126x: SX126xSetOperatingMode: SetOperatingMode: STDBY_RC (1)
[00:25:56.919,000] <dbg> sx126x: SX126xAntSwOff: No antenna switch configured
[00:25:56.919,000] <dbg> sx126x: SX126xWriteCommand: Issuing opcode 0x84 w. 1 bytes of data
[00:25:56.919,000] <dbg> sx126x: SX126xSetOperatingMode: SetOperatingMode: SLEEP (0)
[00:25:58.672,000] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:25:58.672,000] <err> lorawan: McpsRequest failed : Rx 2 timeout

qbsho, 2025-04-30:
Same here. 
Try to change the Spi Datarate but no success