zephyrproject-rtos/zephyr #36962: littlefs: Too small heap for file cache (again).

r2r0, 2021-07-14:
The issue [#31524](https://github.com/zephyrproject-rtos/zephyr/issues/31524) is back.
Tested with Zephyr at b61ea62b6f09136e08f7fe7cf8a1012b5c5042b3.

This time defining FC_HEAP_PER_ALLOC_OVERHEAD as 28 make problem disappear (at least for 32 bit ARM platform).

Maybe in case when CONFIG_FS_LITTLEFS_FC_HEAP_SIZE is not postive it would be reasonable to use slab instead of heap i.e.:

``` C
#if (CONFIG_FS_LITTLEFS_FC_HEAP_SIZE - 0) > 0
static K_HEAP_DEFINE(file_cache_heap, CONFIG_FS_LITTLEFS_FC_HEAP_SIZE);
#else
#undef CONFIG_FS_LITTLEFS_FC_HEAP_SIZE
static K_MEM_SLAB_DEFINE(file_cache_pool, CONFIG_FS_LITTLEFS_CACHE_SIZE,
                CONFIG_FS_LITTLEFS_NUM_FILES, 4);
#endif /* CONFIG_FS_LITTLEFS_FC_HEAP_SIZE */

static inline void *fc_allocate(size_t size)
{
	void *ret = NULL;

#if defined(CONFIG_FS_LITTLEFS_FC_HEAP_SIZE)
	ret = k_heap_alloc(&file_cache_heap, size, K_NO_WAIT);
#else
	(void) size;
	k_mem_slab_alloc(&file_cache_pool, &ret, K_NO_WAIT);
#endif

	return ret;
}

static inline void fc_release(void *buf)
{
#if defined(CONFIG_FS_LITTLEFS_FC_HEAP_SIZE)
	k_heap_free(&file_cache_heap, buf);
#else
	k_mem_slab_free(&file_cache_pool, &buf);
#endif
}
```

r2r0, 2021-09-02:
@de-nordic Any chance to look at this issue?
I will be happy to provide PR but just need some hints.

de-nordic, 2021-09-03:
@r2r0 Sorry for the delay, I am little busy right now. Give me a few days to look at this and get back to you.

de-nordic, 2021-09-10:
@r2r0 Apologies that it took me so long to look at your proposal.
I think the problem requires a little more sophisticated approach.
At first I have though that we should find a way to export the information of heap overhead or provide some macros that would allocate RAM for specified number of allocations of chunks of maximal size, for example 4 chunks of at mast 1024K; then I have noticed that the `fc_allocate` is not called with `CONFIG_FS_LITTLEFS_CACHE_SIZE` as parameter, but with `lfs->cfg->cache_size`.
And there is a reason for that: you may set that size at runtime, before mounting the file system. If you do not, the `lfs->cfg->cache_size` will be assigned `CONFIG_FS_LITTLEFS_CACHE_SIZE`.
This means that we can have complete different values used for heap allocation at compile time and then used at run-time.
Additionally we may have more than one littlefs file system mounted, each using different cache_size value, but using the same `file_cache_heap`, or `file_cache_pool` in you solution.

r2r0, 2021-09-28:
@de-nordic Thanks a lot for your analysis.
If extraction of heap overhead information will take long time then maybe it is worth to temporarily increase FC_HEAP_PER_ALLOC_OVERHEAD and create separate issue for heap overhead info activity.

de-nordic, 2021-09-28:
> 
> 
> @de-nordic Thanks a lot for your analysis. If extraction of heap overhead information will take long time then maybe it is worth to temporarily increase FC_HEAP_PER_ALLOC_OVERHEAD and create separate issue for heap overhead info activity.

@r2r0 Would you consider adding Kconfig option exposing that overhead? This will not fix the issue at all but will provide, for users, workaround for "tuning" it to more suitable value.

r2r0, 2021-09-29:
No problem, I will add PR with proposed option .