zephyrproject-rtos/zephyr #36966: sys: ring_buffer: returns incorrect pointer

JonathanVanWin, 2021-07-14:
**Describe the bug**
The bug happens when using the (put/get)_(claim/finish) ring_buffer API.
All the following calls are passed the same size parameter.
When you call twice in a row put_claim, then once put_finish, then get_claim, and get_finish, and then once again put_claim; the first and second call to put_claim return a different pointer (consecutive), and then after all the other calls, the third time calling put_claim, it returns the same address as the second one.

**To Reproduce**
Steps to reproduce the behavior:
RING_BUF_DECLARE(ring_buf, SIZE * 9);
1. ring_buf_put_claim(&ring_buf, &buf1, SIZE);
2. ring_buf_put_claim(&ring_buf, &buf2, SIZE);
3. print pointers buf1 and buf2
4. ring_buf_put_finish(&ring_buf, SIZE);
5. ring_buf_get_claim(&ring_buf, &buf, SIZE);
6. ring_buf_get_finish(&ring_buf, SIZE);
7. ring_buf_put_claim(&ring_buf, &buf3, SIZE);
8. print pointers buf2 and buf3

**Expected behavior**
I expect that when calling for a third time put_claim, it remembers that it gave me the second time the same buffer and gives me the next one.
If I didn't mark for the second buffer I finished, it means I'm still working on it, and need another one, not the same one.
I didn't mark the first one I finished writing to it, and the second call still gave me the next buffer.
So I expect to be able to call several consecutive times put_claim, and get the next buffer, just like as the first time.

**Impact**
It is important to solve this issue. I would like to be able to allocate several buffers at the same time, to use for example in uart async mode where you give two buffers, in order to switch with the DMA.
The ring_buffer is a useful module to use to allocate buffers of fixed size, and then later signal how many bytes from the allocated buffer have you written into. But this issue doesn't let you have this behavior.

**Logs and console output**
D: put_claim buffer 0x2000398 size 2048
D: put_claim buffer 0x2000b98 size 2048
D: get_claim buffer 0x2000398 size 2048
D: put_claim buffer 0x2000b98 size 2048

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Board: stm32f411e_disco
 - tag: v2.5.0

**Additional context**
I found this bug when using ring_buffer for giving buffers for rx in uart async api.

nordic-krch, 2021-07-19:
Unfortunately, currently it is a ring buffer feature. `ring_buf_put_finish` ends "session". What you are looking for is different use case when you want to claim more buffers and gradually finish them. I also encountered this issue some time ago and created #27276 but it died due to lack of reviewers and I workarouded it in my use case. I can bring it back as there is more than me that wants it.

carlescufi, 2021-07-20:
Converted to Enhancement as per @nordic-krch comments. @JonathanVanWin if you are interested in reviving #27276 please let @nordic-krch know in this issue and then perhaps you can help reviewing and testing it in order to implement the feature you request.

kartben, 2026-10-30:
The claim/finish ring buffer API has been replaced by ring_buf_put_ptr/ring_buf_get_ptr (see the 4.5 migration guide), so closing this. Please open a new issue if the new API does not fit your use case.

closed 2026-10-30 by kartben