zephyrproject-rtos/zephyr #36986: LittleFS mount fails (error -22)

lord-potato, 2021-07-15:
**Describe the bug**
Mounting a LittleFS filesystem fails with error code 22. The filesystem should be mounted from an SPI NOR flash, which I believe is properly configured, since using the flash_map API I can access and modify the contents.

**To Reproduce**
I'm trying to run the LittleFS sample propgram.

I'm using the Kinetis K22 platform, in the devicetree I've mounted the SPI NOR flash:
```
&spi1 {
	status = "okay";
	pinctrl-0 = <&SPI1_PCS0_PTD4>,
	            <&SPI1_SCK_PTD5>,
	            <&SPI1_SOUT_PTD6>,
	            <&SPI1_SIN_PTD7>;

	spiflash: sl25fl512s@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <50000000>;    // 50MHz
		label = "SPI Flash";
		jedec-id = [ 01 02 20  ];
		size = <0x20000000>;    // 512Mb
		status = "okay";

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			lfs1_part: partition@0 {
				label = "storage";
				reg = <0x00000000 0x04000000>;
			};
		};
	};
};
```

I'have also added an fstab entry:
```
/ {
	fstab {
		compatible = "zephyr,fstab";
		lfs1: lfs1 {
			compatible = "zephyr,fstab,littlefs";
			mount-point = "/lfs1";
			partition = <&lfs1_part>;
			read-size = <16>;
			prog-size = <16>;
			cache-size = <64>;
			lookahead-size = <32>;
			block-cycles = <512>;
		};
	};
};
```

My testing process:
1. Wipe the flash using the flash_map API, then verify it's empty (no problems here)
2. Try to mount the LittleFS filesystem, I get error 84, since the FS doesn't exist (still fine)
3. Format the given partition to LittleFS, this also succeeds (return code 0)
4. Try to mount the FS, which should exist now: error 22.

**Expected behavior**
Calling mount should mount the filesystem.

**Impact**
I cannot continue with the project without a filesystem.

**Logs and console output**
```
Area 0 at 0x0 on SPI Flash for 67108864 bytes
Erasing flash area ... 0
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3669:trace: lfs_mount(0x20000090, 0x20000020 {.context=0xfc4c, .read=0xbf6b, .prog=0xbf4f, .erase=0xbf3b, .sync=0xbeb9, .read_size=16, .prog_size=16, .block_size=65536, .block_count=1024, .block_cycles=512, .cache_size=64, .lookahead_size=32, .read_buffer=0x200006b4, .prog_buffer=0x20000674, .lookahead_buffer=0x20000654, .name_max=0, .file_max=0, .attr_max=0})
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3813:trace: lfs_unmount(0x20000090)
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3815:trace: lfs_unmount -> 0
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3808:trace: lfs_mount -> -84
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3588:trace: lfs_format(0x20000090, 0x20000020 {.context=0xfc4c, .read=0xbf6b, .prog=0xbf4f, .erase=0xbf3b, .sync=0xbeb9, .read_size=16, .prog_size=16, .block_size=65536, .block_count=1024, .block_cycles=512, .cache_size=64, .lookahead_size=32, .read_buffer=0x200006b4, .prog_buffer=0x20000674, .lookahead_buffer=0x20000654, .name_max=0, .file_max=0, .attr_max=0})
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3664:trace: lfs_format -> 0
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3669:trace: lfs_mount(0x20000090, 0x20000020 {.context=0xfc4c, .read=0xbf6b, .prog=0xbf4f, .erase=0xbf3b, .sync=0xbeb9, .read_size=16, .prog_size=16, .block_size=65536, .block_count=1024, .block_cycles=512, .cache_size=64, .lookahead_size=32, .read_buffer=0x200006b4, .prog_buffer=0x20000674, .lookahead_buffer=0x20000654, .name_max=0, .file_max=0, .attr_max=0})
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3813:trace: lfs_unmount(0x20000090)
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3815:trace: lfs_unmount -> 0
WEST_TOPDIR/modules/fs/littlefs/lfs.c:3808:trace: lfs_mount -> -22
[00:02:00.053,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:997: Corrupted dir pair at {0x0, 0x1}
[00:02:00.055,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:02:01.229,000] <err> littlefs: remount after format failed (LFS -22)
[00:02:01.229,000] <err> fs: fs mount error (-22)
FAIL: mount id 0 at /lfs1: -22
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.12.4
 - Commit SHA or Version used: zephyr-v2.6.0-694-g3221a1e8e703

**Additional context**
Add any other context about the problem here.

MaureenHelm, 2021-07-20:
Is this an out-of-tree board? The in-tree `frdm_k22f` board doesn't have a SPI NOR flash.

Have you configured the SPI pinmuxes? Note that although we have started populating Kinetis board devicetrees with pinmux information, the drivers don't use it yet. You'll need to configure pinmuxes in `boards/arm/<your board>/pinmux.c`. See the `frdm_k64f` board for an example.

If you've done this already, can you try increasing the SPI logging level and verifying bus activity with a logic analyzer?

lord-potato, 2021-07-21:
Thanks for your reply! Yes, this is an out-of-tree board, and I've already modified my pinmux.c, to have SPI on the correct pins, and SPI debug log shows a lot of activity.

Unfortunately I don't have a logic analyzer on hand so I cannot easily check the content of the packets, but with a four channel scope I was able to verify that bidirectional communication happens between the MCU and the Flash.

I can also read and write the contents of the flash using the flash_map API

lord-potato, 2021-08-03:
@MaureenHelm any updates on this?

MaureenHelm, 2021-08-09:
@de-nordic Any guidance? Can you take a look at the LittleFS configuration?

de-nordic, 2021-08-16:
@marcsekz Would you be able to dump the SPI NOR, after format, and attach it here?

dleach02, 2021-09-14:
@marcsekz have you had an opportunity to try the suggestion?

dleach02, 2021-12-13:
Closing due to lack of response from submitter per policy

closed 2021-12-13 by dleach02