zephyrproject-rtos/zephyr #36990: Memory misalignment ARM Cortex-M33

mehrdadh, 2021-07-15:
**Describe the bug**
I have built a model with TVM on nRF5340DK using Zephyr RTOS and I ran into a memory misalignment. In TVM we define a large memory that we use as the TVM stack and we make sure the memory allocation always has alignment of 4 bytes.
```
static uint8_t g_aot_memory[WORKSPACE_SIZE];
```
However we ran into this bug where the initial address of the memory was misaligned and causes a CPU exception on FP32 operation. To fix this I added an attribute which fixed this problem, but my assumption was that zephyr would take care of this in the first place.
```
static uint8_t __attribute__((aligned(4))) g_aot_memory[WORKSPACE_SIZE];
```

TVM Issue: https://github.com/apache/tvm/issues/8474

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr v2.5-branch at this commit: dabf237584
 - Zephyr SDK: 0.12.3

I wasn't sure if this is a bug or I missed a `CONFIG` in my project file. So any help here is appreciated

cfriedt, 2021-07-20:
I would strongly suggest using our predefined stack declaration macro [K_KERNEL_STACK_DEFINE](https://docs.zephyrproject.org/latest/reference/kernel/threads/index.html#c.K_KERNEL_STACK_DEFINE) but this does not seem to be a bug in Zephyr itself.

closed 2021-07-20 by cfriedt

mehrdadh, 2021-07-20:
This PR fixed this issue in TVM:
https://github.com/apache/tvm/pull/8487