zephyrproject-rtos/zephyr #36994: eSPI emulator has little documentation and no sample

yperess, 2021-07-15:
**Is your enhancement proposal related to a problem? Please describe.**
Commit https://github.com/zephyrproject-rtos/zephyr/commit/d1948dc16457ca12978fd433249d628afa77d53b added the eSPI emulator. The emulator appears to require child nodes to be added to `zephyr,espi-emul-controller` compatible nodes, but there's no documentation or binding associated with these.

**Describe the solution you'd like**
A sample test that uses the eSPI emulator

@semihalf-niedzwiecki-dawid

albertofloyd, 2024-02-22:
@yperess  does this cover the need? https://github.com/zephyrproject-rtos/zephyr/pull/66912
Any inputs on this @aaronemassey

aaronemassey, 2024-02-22:
We have a demo with [emul_espi_host.c](https://github.com/zephyrproject-rtos/zephyr/blob/a48c958c8fb85a0985880482df9c79407a8fc55e/subsys/emul/espi/emul_espi_host.c).

IIUC I think @yperess was reporting this on behalf of @semihalf-niedzwiecki-dawid ?

However, It's true that there aren't a lot of examples of the espi emulator though.

albertofloyd, 2024-02-22:
@yperess  @aaronemassey  could the vehicle for reference device tree be the samples/drivers/espi code adding e.g. adding a  qemu board overlay?

aaronemassey, 2024-02-22:
@albertofloyd

> could the vehicle for reference device tree be the samples/drivers/espi code adding e.g. adding a qemu board overlay?

Not sure I understand the question. Are you asking if we should add an example test using the host emulator with a qemu board overlay? If so, we can just add the qemu overlay to the existing test.

albertofloyd, 2024-02-23:
@aaronemassey Yes
@yperess is asking for a device binding sample code for eSPI emulator, I'm unsure which are main board supported, but assuming qemu or similar can be used.

yperess, 2024-02-23:
I think the steps are to identify a peripheral that uses the eSPI bus, then implement an emulator and a single test that uses it. This would be enough as an example for how the eSPI bus transactions are handled in the emulator.

albertofloyd, 2024-02-23:
@yperess  Currently samples/drivers/espi is the only eSPI driver client test code inside Zephyr code tree.

albertofloyd, 2025-02-20:
@yperess re-open if anything else apart from sample drivers code is required

closed 2025-02-20 by albertofloyd