zephyrproject-rtos/zephyr #37006: tests: kernel: mem_protect: stack_random: enable qemu_riscv32

cfriedt, 2021-07-16:
**Describe the bug**
TL:DR - needed to disable that test only for `qemu_riscv32` because it broke CI
Details in #36996

**To Reproduce**
```
west build -p always -b qemu_riscv32 -t run tests/kernel/mem_protect/stack_random
```

It has only failed in CI so far, and I am unable to reproduce the failure locally.

**Expected behavior**
```
% west build -p always -b qemu_riscv32 -t run tests/kernel/mem_protect/stack_random
...
*** Booting Zephyr OS build zephyr-v2.6.0-1039-g523764b3fd75  ***
Running test suite stack_pointer_randomness
===================================================================
START - test_stack_pt_randomization
Test Stack pointer randomization
stack pointer changed 13 times out of 64 tests
PASS - test_stack_pt_randomization in 0.5 seconds
===================================================================
Test suite stack_pointer_randomness succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Impact**
It made CI fail consistently. 

**Logs and console output**
```
% west build -p always -b qemu_riscv32 -t run tests/kernel/mem_protect/stack_random
...
*** Booting Zephyr OS build zephyr-v2.6.0-1063-g0106d8f2a391  ***
Running test suite stack_pointer_randomness
===================================================================
START - test_stack_pt_randomization
Test Stack pointer randomization
stack pointer changed 0 times out of 64 tests
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_protect/stack_random/src/main.c:68: test_stack_pt_randomization: (sp_changed equal to 0)
Stack pointer is not randomized
FAIL - test_stack_pt_randomization in 0.6 seconds
===================================================================
Test suite stack_pointer_randomness failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: CI
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: between 84c5a46 and de13034

**Additional context**

fkokosinski, 2021-07-29:
Thanks for reporting this bug. I don't think anyone is currently working on this issue. I will look into it. Could you possibly reassign this to me?

katsuster, 2021-08-02:
@fkokosinski 
Sorry for late. I'm welcome. I can reproduce this problem but there is strange behavior.
If I added printk("%x\n", last_sp); at last of alternate_thread() then this test has passed...

```
8000623c
8000622c
8000625c
8000624c
8000623c
8000622c
8000625c
8000624c
8000623c
8000622c
8000625c
8000624c
8000623c
stack pointer changed 61 times out of 64 tests
 PASS - test_stack_pt_randomization in 0.15 seconds
===================================================================
Test suite stack_pointer_randomness succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
qemu-system-riscv32: terminating on signal 2
ninja: build stopped: interrupted by user.
```

katsuster, 2021-08-02:
I want to share current progress:

- By debugger analysis (NOT use printk, NOT change the binary): sp is changed for every thread
- This test is passed if ...
    - Children threads call printk()
    - A main thread calls k_yield() after k_thread_create()
    - Not use icount option of QEMU

So I guess it's problem of timing of thread switching in special situation.

```c
	/* Start thread */
	for (i = 0; i < THREAD_COUNT; i++) {
		k_thread_create(&alt_thread_data, alt_thread_stack_area,
				STACKSIZE, (k_thread_entry_t)alternate_thread,
				NULL, NULL, NULL, K_HIGHEST_THREAD_PRIO, 0,
				K_NO_WAIT);
		k_yield();    /* Add this line */
	}
```

```
# NG (use icount)

qemu-system-riscv32 -nographic -machine virt -bios none -m 256 -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline \
-icount shift=6,align=off,sleep=off \
-rtc clock=vm -kernel zephyr.elf

# OK (not use icount)

qemu-system-riscv32 -nographic -machine virt -bios none -m 256 -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline \
-rtc clock=vm -kernel zephyr.elf
```

katsuster, 2021-08-02:
Any comments and ideas are welcome.

fkokosinski, 2021-08-02:
>     * Not use icount option of QEMU

I've also found that manipulating the value of icount can force this test to pass. For example, changing the value of icount shift in [qemu_riscv32 board config](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/riscv/qemu_riscv32/qemu_riscv32_defconfig#L14) seems to do the trick.

Other than that, I've also noticed that changing the values of the following constants also makes this test pass:
* increasing `CONFIG_STACK_POINTER_RANDOM` to >64 [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/mem_protect/stack_random/prj.conf#L2)
* changing `STACKSIZE` [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/kernel/mem_protect/stack_random/src/main.c#L10) to either >=4096 (max value tested 2096000) or <2048 (min value tested 512)

I've also discovered that the commit that causes this test to fail is the following one:
```
78c222e56f23b2cfff86f49e8885d74f48110869 is the first bad commit
commit 78c222e56f23b2cfff86f49e8885d74f48110869
Author: Michał Narajowski <michal.narajowski@codecoup.pl>
Date:   Wed Jul 14 10:04:37 2021 +0200
    doc: release-notes-2.7: Return value for opcode callback
    Add release notes for Stable API Change in Mesh.
    Signed-off-by: Michał Narajowski <michal.narajowski@codecoup.pl>
 doc/releases/release-notes-2.7.rst | 2 ++
 1 file changed, 2 insertions(+)
 ```
 which seemed strange, as this commit only changes the release notes. But what this commit also does, is it increases the length of `BUILD_STRING` by one character. I've verified that I can make the test pass by forcing the `BUILD_STRING` to be one character shorter i.e. by:
 ```
 diff --git a/CMakeLists.txt b/CMakeLists.txt
index 2f8dcbc7ad..7214c0d98a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -106,7 +106,7 @@ endif()
 if(BUILD_VERSION)
   zephyr_compile_definitions(
-    BUILD_VERSION=${BUILD_VERSION}
+    BUILD_VERSION="zephyr-v2.6.0-100-g78c222e56f23" 
   )
 endif()
 ```
 
 The rodata section properties look like this after/before the change:
 ```
 Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [10] rodata            PROGBITS        80004928 0049b8 000e24 00   A  0   0  4 # fails
  [10] rodata            PROGBITS        80004928 0049b8 000e20 00   A  0   0  4 # passes
  ```

katsuster, 2021-08-06:
Hmm... It's very strange...

If I add this while loop after thread creation loop, this test is going to freeze.
So no context switching is happen.

```
--- a/tests/kernel/mem_protect/stack_random/src/main.c
+++ b/tests/kernel/mem_protect/stack_random/src/main.c
@@ -60,6 +60,9 @@ void test_stack_pt_randomization(void)
                                K_NO_WAIT);
        }

+       while (changed == 0) {
+               k_yield();
+       }

        printk("stack pointer changed %d times out of %d tests\n",
               changed, THREAD_COUNT);
```

But... this test is suddenly passed if I add one variable in global name space.

```
--- a/tests/kernel/mem_protect/stack_random/src/main.c
+++ b/tests/kernel/mem_protect/stack_random/src/main.c
@@ -13,6 +13,7 @@

 void *last_sp = (void *)0xFFFFFFFF;
 volatile unsigned int changed;
+volatile unsigned int foo;

 void alternate_thread(void)
 {
@@ -60,6 +61,10 @@ void test_stack_pt_randomization(void)
                                K_NO_WAIT);
        }

+       while (changed == 0) {
+               k_yield();
+       }
+       foo++;

        printk("stack pointer changed %d times out of %d tests\n",
               changed, THREAD_COUNT);
```

commit 19fc1ee7 (2021-08-30): subsys/random: change parameters for timer-based PRNG

PR #36996 disabled running mem_protect/stack_random test on qemu_riscv32
platform because of this test consistently failing on said platform.
This test starts new threads in equal time intervals, and because of
that we get repeating values after performing the modulus operation when
calculating the stack pointer address.

This can be solved by changing the value of the _RAND32_INC constant
that is used to increase the value returned by the timer-based PRNG.

This commit decreases the value of the mentioned constant from
1000000013U to 1000 [... cut]
closed 2021-08-30 by nashif