zephyrproject-rtos/zephyr #37007: Problem with out of tree driver

mcscholtz, 2021-07-16:
I am trying to write an out of tree driver and having problems. I tried looking at the documentation but it does not seem to cover this particular case since there is no documentation on how to do this: https://docs.zephyrproject.org/latest/samples/application_development/out_of_tree_driver/README.html

I think I am close to getting it to work, this is what my directory structure is:
├── CMakeLists.txt
├── drivers
│   ├── CMakeLists.txt
│   ├── eeprom
│   │   ├── CMakeLists.txt
│   │   ├── eeprom_m24xxx.c
│   │   ├── Kconfig
│   │   └── zephyr
│   │       └── module.yml
│   └── Kconfig
├── prj.conf
├── README.rst
├── sample.yaml
└── src
    └── main.c

I can compile the code but it seems like something is not linking properly and this is likely the problem, cmake gives the following warning:
```
No SOURCES given to Zephyr library: drivers__eeprom
Excluding target from build.
```
It looks like it compiles, because I see the following warnings:
```
[117/149] Building C object modules/eeprom/CMakeFiles/..__app__drivers__eeprom.dir/eeprom_m24xxx.c.obj
    ../drivers/eeprom/eeprom_m24xxx.c:603:39: warning: 'eeprom_m24xxx_api' defined but not used [-Wunused-const-variable=]
    603 | static const struct eeprom_driver_api eeprom_m24xxx_api = {
```
Using the device tree macros I can see the node is valid but when I try to get the binding it always returns NULL
```
#define EEPROM1_NODE DT_ALIAS(eeprom1)
#if !DT_HAS_COMPAT_STATUS_OKAY(st_m24xxx)
#error "EEPROM is not enabled"
#elif !DT_ON_BUS(EEPROM1_NODE, i2c)
#error "EEPROM is not on i2c bus"
#endif
const struct device *dev = device_get_binding(DT_LABEL(EEPROM1_NODE));
{
    printk("Failed to get EEPROM device binding\n");
    return NULL;
}
```
I always see that 'Failed to get EEPROM device binding' line on the console.

If I understand this correctly it looks like my driver is being compiled but the common interface for eeprom in the zephyr tree is not being compiled, any suggestions?

gmarull, 2021-07-19:
It may be useful to look at https://github.com/zephyrproject-rtos/example-application, it shows how to create oot drivers

mcscholtz, 2021-07-20:
@gmarull This is helpful, thank you. 

I will keep this issue open because documentation on the main website will make it easier to find this information.

gmarull, 2021-07-20:
@mcscholtz If you consider the issue solved, please, close the issue.

closed 2021-07-22 by mcscholtz