zephyrproject-rtos/zephyr #37038: stm32f4 - DMA tx interrupt doesn't enabled

xin10pan, 2021-07-19:
**Scenario:** Use nucleo-f429zi as master and enable I2S to transmit data
**Platform:** nucleo-f429zi
**sample:**  Use samples/drivers/i2s/litex/ and delete the code related to RX
`void main(void)
{
	void *rx_mem_block, *tx_mem_block;
	
	init();  //get I2S device

	/* start i2s tx driver */
    i2s_prepare_audio(host_i2s_tx_dev);  //call trigger function

	ret = i2s_trigger(host_i2s_tx_dev, I2S_DIR_TX, I2S_TRIGGER_START);
	if (ret != 0) {
		printk("i2s_trigger failed with %d error\n", ret);
		exit(-1);
	}
    printk("After i2s_trigger\n");

	while (true) {
		k_mem_slab_alloc(&i2s_tx_mem_slab, &tx_mem_block, K_NO_WAIT);
        memset(tx_mem_block, 0, AUDIO_FRAME_BUF_BYTES);
		i2s_write(host_i2s_tx_dev, tx_mem_block, AUDIO_FRAME_BUF_BYTES);
		k_mem_slab_free(&i2s_rx_mem_slab, &rx_mem_block);
	}
}`

**dts:**

```
     +&dma1 {
        status = "okay";
     };
    +
    +&i2s2 {
    +       pinctrl-0 = <&i2s2_ck_pb13 &i2s2_sd_pb15 &i2s2_ws_pb12>;
    +       status = "okay";
      }
```
**Result:** When transmit data via I2S, it seems that DMA tx interrupt is not enabled and it doesn't call dma_tx_callback. The configuration is showed above. I can't find out why.
By the way, there is not a real peripheral connected to nucleo-f429zi, is it a matter?
Looking forward to your reply!

xin10pan, 2021-07-22:
Yeah, I have configured I2S2 like this:

FRASTM, 2021-07-22:
well, Initially the stm32f429 DTS includes (from stm32f401.dts) the :

```
		i2s2: i2s@40003800 {
			compatible = "st,stm32-i2s";
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40003800 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00004000>;
			interrupts = <36 5>;
			dmas = <&dma1 4 0 0x400 0x3
				&dma1 3 0 0x400 0x3>;
			dma-names = "tx", "rx";
			status = "disabled";
			label = "I2S_2";
		};
```
Mapping the DMA request from **SPI2**: do you have SPI2 enable ?

FRASTM, 2021-07-22:
It seems that I2S2 extended (EXT_TX and EXT_RX ) pins are used for full duplex functions which is _not_ your configuration. So the DMA1 request is _not_ the ext one. Keeping the original dma request is right :
  dma1 stream 4 request (slot) 0 for Tx
  dma1 stream 3 request (slot) 0 for Rx

xin10pan, 2021-07-22:
Thanks for your reply!
I don't enable SPI2 and have also tried the original DMA request, it's the same result.
This is the generated dts file related to SPI, I2S and DMA is:
```
              dma1: dma@40026000 {
			compatible = "st,stm32-dma-v1";
			#dma-cells = < 0x4 >;
			reg = < 0x40026000 0x400 >;
			interrupts = < 0xb 0x0 0xc 0x0 0xd 0x0 0xe 0x0 0xf 0x0 0x10 0x0 0x11 0x0 0x2f 0x0 >;
			clocks = < &rcc 0x0 0x200000 >;
			status = "okay";
			label = "DMA_1";
			phandle = < 0x15 >;
		};
		dma2: dma@40026400 {
			compatible = "st,stm32-dma-v1";
			#dma-cells = < 0x4 >;
			reg = < 0x40026400 0x400 >;
			interrupts = < 0x38 0x0 0x39 0x0 0x3a 0x0 0x3b 0x0 0x3c 0x0 0x44 0x0 0x45 0x0 0x46 0x0 >;
			clocks = < &rcc 0x0 0x400000 >;
			st,mem2mem;
			status = "disabled";
			label = "DMA_2";
		};
		spi2: spi@40003800 {
			compatible = "st,stm32-spi";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40003800 0x400 >;
			clocks = < &rcc 0x2 0x4000 >;
			interrupts = < 0x24 0x5 >;
			status = "disabled";
			label = "SPI_2";
		};
		spi3: spi@40003c00 {
			compatible = "st,stm32-spi";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40003c00 0x400 >;
			clocks = < &rcc 0x2 0x8000 >;
			interrupts = < 0x33 0x5 >;
			status = "disabled";
			label = "SPI_3";
		};
		spi4: spi@40013400 {
			compatible = "st,stm32-spi";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40013400 0x400 >;
			clocks = < &rcc 0x3 0x2000 >;
			interrupts = < 0x54 0x5 >;
			status = "disabled";
			label = "SPI_4";
		};
		i2s2: i2s@40003800 {
			compatible = "st,stm32-i2s";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40003800 0x400 >;
			clocks = < &rcc 0x2 0x4000 >;
			interrupts = < 0x24 0x5 >;
			dmas = < &dma1 0x4 0x0 0x400 0x3 &dma1 0x3 0x0 0x400 0x3 >;
			dma-names = "tx", "rx";
			status = "okay";
			label = "I2S_2";
			pinctrl-0 = < &i2s2_ck_pb13 &i2s2_sd_pb15 &i2s2_ws_pb12 >;
		};
		i2s3: i2s@40003c00 {
			compatible = "st,stm32-i2s";
			#address-cells = < 0x1 >;
			#size-cells = < 0x0 >;
			reg = < 0x40003c00 0x400 >;
			clocks = < &rcc 0x2 0x8000 >;
			interrupts = < 0x33 0x5 >;
			dmas = < &dma1 0x5 0x0 0x400 0x3 &dma1 0x0 0x0 0x400 0x3 >;
			dma-names = "tx", "rx";
			status = "disabled";
			label = "I2S_3";
		};
```
As mentioned above,  I want to use Nucleo F429ZI to send media data, the files which  I have changed  is : ```
zephyr/samples/drivers/i2s/litex/prj.conf                    /* Open the configuration of I2S and DMA*/
zephyr/samples/drivers/i2s/litex/main.c                      /* I2S sample for TX*/
zephyr/boards/arm/nucleo_f429zi/nucleo_f429zi.dts  /* Enable I2S with pins and DMA*/
```
In addition, in the ST Nucleo F429ZI introduction website of zephyr (https://docs.zephyrproject.org/latest/boards/arm/nucleo_f429zi/doc/index.html), it doesn't mention that Nucleo F429ZI support I2S in "Supported Features", so does Nucleo F429ZI support I2S in zephyr?

FRASTM, 2021-07-22:
`./boards/arm/nucleo_f429zi/nucleo_f429zi.dts` or thanks to `./samples/drivers/i2s/litex/boards/nucleo_f429zi.overlay`
You should see the CONFIG_I2S and CONFIG_I2S_STM32

Also check the https://github.com/zephyrproject-rtos/zephyr/issues/34390

FRASTM, 2021-07-22:
In the Tx direction, the DMA interrupt occurs when the transfer is complete (TC flag). In case there is a mis-alignment between memory to periph (I2S), the DMA is waiting for more data to send.
The dma_tx_callback is given to the dma driver during the dma config by the peripheral (i2s) through the `stream->dma_cfg`

xin10pan, 2021-07-28:
Now, I use  the I2S in master mode and don't enable the PLL. I have checked the configuration of I2S and DMA, all are consistent with the original DTS&I2S_DMA_CHANNEL_INIT&i2s_stm32_config_2 except FIFO threshold. 
In the code below, "channel_config " would be "features"?
```
 .fifo_threshold = STM32_DMA_FEATURES_FIFO_THRESHOLD(		\
	 DT_DMAS_CELL_BY_NAME(DT_NODELABEL(i2s##index), dir, channel_config))
```

I have catch some logs of DMA configuration showed below:
```
dma_channel = 4
config->dma_slot = 0                        
config->channel_priority = 0                
config->channel_direction = 1               
config->dest_data_size = 2                  
config->source_data_size = 2          
DMA_InitStruct.periph_addr_adj = 2                                     
DMA_InitStruct.MemoryOrM2MDstAddress = 0x20001320          
DMA_InitStruct.PeriphOrM2MSrcAddress = 0x4000380c          
DMA_InitStruct.Priority = 0                                
DMA_InitStruct.Direction = 64                              
DMA_InitStruct.MemoryOrM2MDstIncMode = 1024                
DMA_InitStruct.PeriphOrM2MSrcIncMode = 0                   
DMA_InitStruct.Mode = 0                                    
DMA_InitStruct.PeriphOrM2MSrcDataSize = 2048               
DMA_InitStruct.MemoryOrM2MDstDataSize = 8192               
DMA_InitStruct.MemBurst = 0                                
DMA_InitStruct.PeriphBurst = 0                             
DMA_InitStruct.Channel = 0                                 
DMA_InitStruct.FIFOThreshold = 3                           
DMA_InitStruct.FIFOMode = 0                                
DMA_InitStruct.NbData = 384 
```
In addition, I have some other questions.
1.  I find that the interrupt of I2S is not triggered yet. What's the relationship between I2S interrupt and DMA?
2.  Are there any other configurations besides dts and i2s_config which users should be considered when use I2S?
3. Is there a sample of  nucleo-f429zi?
 
Looking forward to your reply, thank you!

FRASTM, 2021-07-28:
in the dma of type V1 (stm32) the 'channel_config' is  stream configuration and 'feature' is the FIFO threshold 
in the dma config: 
slot =0 is for the request on the channel 4 --> SPI2_TX
DMA_InitStruct.MemoryOrM2MDstAddress = 0x20001320 is the address of the buffer to send to the I2S2
DMA_InitStruct.PeriphOrM2MSrcAddress = 0x4000380c  is the address of the I2S2 data register

FRASTM, 2021-07-28:
1. 
With the DMA, the I2S interrupt does not occur, only the DMA Transmit (no receive in your appli) is triggered
2. on other config, to set,
3. You can have a look on the `tests/drivers/spi/spi_loopback`. This is an application running on the nucleo_f429zi. The SPI is similar to the I2S (to pass this test MISO & MOSI pins are connected by wire on the board)

xin10pan, 2021-07-28:
Thanks for your reply.
In i2s_ll_stm32.c, the original code to get fifo_threshold is
```
.fifo_threshold = STM32_DMA_FEATURES_FIFO_THRESHOLD(		\
	 DT_DMAS_CELL_BY_NAME(DT_NODELABEL(i2s##index), dir, **channel_config**))  **// channel_config should be modified 
                                                                                                                                               // to feature?**
```
Just now, I enable "I2S_STM32_USE_PLLI2S_ENABLE(=y)", then DMA interrupt is triggered. Is it a matter?

FRASTM, 2021-07-28:
yes, the structure should be :
 ```
.fifo_threshold = STM32_DMA_FEATURES_FIFO_THRESHOLD(
DT_DMAS_CELL_BY_NAME(DT_NODELABEL(i2s##index), dir, features))
```
Once the DMA irq occurs on the TX side ( DMA_InitStruct.NbData = 384 data transferred), is the  dma_tx_callback called ?

xin10pan, 2021-07-28:
Yes, dma_tx_callback is called. But why does the DMA interrupt relate to PLL?

FRASTM, 2021-07-28:
This is more the PLLI2SCLK to be enabled to activate the I2S, then I2S transfer can start with 
DMA

xin10pan, 2021-07-28:
Ok.  Does the default values of PLL parameters in zephyr/drivers/i2s/Kconfig.stm32 need to be modified?  Is there a HSI used when PLL is not enabled in nucleo-f429zi?

FRASTM, 2021-07-28:
yes, the PLLI2S clock source is either HSI or HSE after the divM
The PLLI2S has its own mulN, divR.

(refer to figure 16 of ref ManRM0386)

closed 2021-10-13 by github-actions[bot]