zephyrproject-rtos/zephyr #37056: Clarify device power states

gmarull, 2021-07-19:
**Is your enhancement proposal related to a problem? Please describe.**

As of today, there is no clear guidance on what device power states really mean on Zephyr. As a result of this problem, most (if not all) devices treat all "low power states" as a single state. In the end, this results in device drivers that do not behave according to any specification and so may result in unexpected behavior.

Let's start analyzing the current defined states:

```c
/** @brief Device power states. */
enum pm_device_state {
	/** Device is in active or regular state. */
	PM_DEVICE_STATE_ACTIVE,
	/**
	 * Device is in low power state.
	 *
	 * @note
	 *     Device context is preserved.
	 */
	PM_DEVICE_STATE_LOW_POWER,
	/**
	 * Device is suspended.
	 *
	 * @note
	 *     Device context may be lost.
	 */
	PM_DEVICE_STATE_SUSPEND,
	/**
	 * Device is suspended (forced).
	 *
	 * @note
	 *     Device context may be lost.
	 */
	PM_DEVICE_STATE_FORCE_SUSPEND,
	/**
	 * Device is turned off (power removed).
	 *
	 * @note
	 *     Device context is lost.
	 */
	PM_DEVICE_STATE_OFF,
	/** Device is being resumed. */
	PM_DEVICE_STATE_RESUMING,
	/** Device is being suspended. */
	PM_DEVICE_STATE_SUSPENDING,
};
```

- `PM_DEVICE_STATE_ACTIVE`: This state is quite clear, the device is _turned on_ and operational.
- `PM_DEVICE_STATE_LOW_POWER`: Here we start having problems. What is the difference between `PM_DEVICE_STATE_LOW_POWER` and `PM_DEVICE_STATE_SUSPEND`? Other than context preservation, there is no clear difference. Should a device implement one or the other? Or both? By looking at a few drivers, it's clear that many developers have gone the _safest_ way, just handle any states that sound like "low power" the same way, e.g.

https://github.com/zephyrproject-rtos/zephyr/blob/5b5396f7b6ff84f12553d9e16207df0cce7e19e7/drivers/i2c/i2c_nrfx_twim.c#L285-L296

It looks like system PM uses low power vs. suspend depending on the system transition:

https://github.com/zephyrproject-rtos/zephyr/blob/c2897c6b18121fa7733cbb9ad7e0207e4ba2054b/subsys/pm/power.c#L210-L235

This is however inconsistent with the runtime PM framework, where the suspend state is used.

- `PM_DEVICE_STATE_SUSPEND`: See comments for `PM_DEVICE_STATE_LOW_POWER`.
- `PM_DEVICE_STATE_FORCE_SUSPEND`: By looking at Zephyr code, it turns out that this state is not used other than in some "handle all" switch/case. However, it's never invoked by the system. Again, here it is not clear what this state means. How can a device be in a "force suspend" state?
- `PM_DEVICE_STATE_OFF`: This state is quite clear, means that device is turned off, i.e. power removed.
- `PM_DEVICE_STATE_RESUMING` and `PM_DEVICE_STATE_SUSPENDING` are transitional states.

**Describe the solution you'd like**

If we look at the ACPI specification, we have the following states:

| State  | Description                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   |
|--------|-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| D0     | This state is assumed to be the highest level of power consumption. The device is completely active and responsive, and is expected to remember all relevant context continuously.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            |
| D1     | The meaning of the D1 Device State is defined by each device class. Many device classes may not define D1. In general, D1 is expected to save less power and preserve more device context than D2.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            |
[... 19 lines cut ...]

- `D1` and `D2` have no clear mapping to Zephyr states
- `D3hot` may have two possible candidates, even though `PM_DEVICE_STATE_SUSPEND` seems the best candidates since it's the only low power state used by the runtime PM framework.
- `PM_DEVICE_STATE_FORCE_SUSPEND` has no place
- Transitional states have no representation (e.g. `PM_DEVICE_STATE_RESUMING`)

The proposed solution/s:

- Remove `PM_DEVICE_STATE_FORCE_SUSPEND` since it does not represent an actual state.
- Clarify the difference between `PM_DEVICE_STATE_SUSPEND` and `PM_DEVICE_STATE_LOW_POWER`. Possibly stay with just `PM_DEVICE_STATE_SUSPEND` and delete `PM_DEVICE_STATE_LOW_POWER` since the runtime framework does not use it.
- Decide if transitional states can be replaced with a sort of "transitioning" flag for all transitions.
- Decide if, for now, it is worth adding support for D1 and D2 type of states. If multiple low power states are supported, this likely means a sort of device PM policy will be required to decide which low power (D1, D2, D3x) state has to be used.
- Update documentation:
  - Clarify states meaning
  - Add details on ACPI mapping
  - Provide some device implementation guidance
  - Document allowed transitions 

**Describe alternatives you've considered**

N/A

**Additional context**

N/A

gmarull, 2021-07-19:
cc: @ceolin @nashif @erwango

ceolin, 2021-07-20:
Thanks for bringing it up
> As of today, there is no clear guidance on what device power states really mean on Zephyr. As a result of this problem, most (if not all) devices treat all "low power states" as a single state. In [...]

Linux PCI documentation defines these transitions as well, Wondering if they are defined in the ACPI documentation.

> 
> The following table provides a tentative mapping between ACPI and Zephyr states:
> ACPI State 	Zephyr state
> D0 	`PM_DEVICE_STATE_ACTIVE`
> D1 	?
> D2 	?
> D3hot 	`PM_DEVICE_STATE_SUSPEND` or `PM_DEVICE_STATE_LOW_POWER`

Isn't a better map to `PM_DEVICE_STATE_LOW_POWER` `D1` or `D2` due context preservation ?
 
> D3cold 	`PM_DEVICE_STATE_OFF`
>
> Some initial observations:
>
>     * `D1` and `D2` have no clear mapping to Zephyr states
>
>     * `D3hot` may have two possible candidates, even though [...]

Agree, don't see any purpose for have it.

> 
>     * Transitional states have no representation (e.g. `PM_DEVICE_STATE_RESUMING`)

I see transitional states only as internal states to be honest. Agree that this should no be exported, but I don't see a major problem in having them internally.

> 
> 
> The proposed solution/s:
> 
>     * Remove `PM_DEVICE_STATE_FORCE_SUSPEND` since it does not represent an actual state.
:+1:

> 
>     * Clarify the difference between `PM_DEVICE_STATE_SUSPEND` and `PM_DEVICE_STATE_LOW_POWER`. Possibly stay with just `PM_DEVICE_STATE_SUSPEND` and delete `PM_DEVICE_STATE_LOW_POWER` since the runtime framework does not use it.

I need to look more other specifications but my first thought is to keep it. Linux implements runtime PM in the same way we do, only suspending or resuming, but for system power management is available more states.

> 
>     * Decide if transitional states can be replaced with a sort of "transitioning" flag for all transitions.

That is a possibility, we just need to be careful to not make the implementation over-complicated. Transitional states are easy to read and give us enough context about from - to states.

> 
>     * Decide if, for now, it is worth adding support for D1 and D2 type of states. If multiple low power states are supported, this likely means a sort of device PM policy will be required to decide which low power (D1, D2, D3x) state has to be used.

Not sure as well. If I understood correctly neither Linux or Windows have it, they delegate this to the driver.

Btw, Linux does not define power states for devices. It is the driver responsibility put the device in the best power saving state based on the system state. Device Runtime PM is another thing and just define SUSPEND and RESUME.

> 
>     * Update documentation:
>       
>       * Clarify states meaning
>       * Add details on ACPI mapping
>       * Provide some device implementation guidance
>       * Document allowed transitions

:+1:
> 
> 
> **Describe alternatives you've considered**
> 
> N/A
> 
> **Additional context**
> 
> N/A

gmarull, 2021-07-20:
> Thanks for bringing it up
> 

> Linux PCI documentation defines these transitions as well, Wondering if they are defined in the ACPI documentation.
>
>
> Isn't a better map to `PM_DEVICE_STATE_LOW_POWER` `D1` or `D2` due context [...]

IMO if we are open to have D1 and D2 states we should support both, not only one. My feeling is that in most cases we'll have D0 and D3 and in just a few D1/D2. For this reason, naming `LOW_POWER` a state that will likely be used in specific cases doesn't sound like a good idea. I also wonder why the runtime PM uses `PM_DEVICE_STATE_SUSPEND` and not `PM_DEVICE_STATE_LOW_POWER`.

> Agree, don't see any purpose for have it.

:+1: 
> 
> > ```
> > * Transitional states have no representation (e.g. `PM_DEVICE_STATE_RESUMING`)
> > ```
> 
> I see transitional states only as internal states to be honest. Agree that this should no be exported, but I don't see a major problem in having them internally.
> 
> > The proposed solution/s:
> > ```
> > * Remove `PM_DEVICE_STATE_FORCE_SUSPEND` since it does not represent an actual state.
> > ```
> 
> 
> 
> > ```
> > * Clarify the difference between `PM_DEVICE_STATE_SUSPEND` and `PM_DEVICE_STATE_LOW_POWER`. Possibly stay with just `PM_DEVICE_STATE_SUSPEND` and delete `PM_DEVICE_STATE_LOW_POWER` since the runtime framework does not use it.
> > ```
> 
> I need to look more other specifications but my first thought is to keep it. Linux implements runtime PM in the same way we do, only suspending or resuming, but for system power management is available more states.

Linux has some differences with our solution. For example, the runtime PM part has a custom set of callbacks:

```c
struct dev_pm_ops {
	int (*suspend)(struct device *dev);
	int (*resume)(struct device *dev);
...
	int (*runtime_suspend)(struct device *dev);
	int (*runtime_resume)(struct device *dev);
	int (*runtime_idle)(struct device *dev);
};
```

Even though I assume a device can re-use regular suspend/resume callbacks.

> That is a possibility, we just need to be careful to not make the implementation over-complicated. Transitional states are easy to read and give us enough context about from - to states.

:+1: 

We also need to be careful with transitional states, for example, the `OFF` state doesn't have it. While it may be intentional, it needs to be documented to avoid confusion.

> Not sure as well. If I understood correctly neither Linux or Windows have it, they delegate this to the driver.

Yes, it seems to be up to the driver to support states. In the case of Windows, D0 and D3 seem to be the most common (active/suspended in Zephyr's language).

> 
> Btw, Linux does not define power states for devices. It is the driver responsibility put the device in the best power saving state based on the system state. Device Runtime PM is another thing and just define SUSPEND and RESUME.

Still on Linux there is a callback sequence guarantee, a critical feature that Zephyr is missing. As far as I can see, most device PM in Zephyr is nowadays done via `pm_device_state_set`, something that needs to be addressed soon, as it's clearly not the way to handle device PM. Anyone can call `pm_device_state_set` with any target state, then what can happen depends on driver design, this is why a lot of drivers have state change check logic, which leads to overcomplicated device PM implementations. #36125 (and #36759) tries to start addressing this problem, but there is still more work to be done.

Whether a device driver decides to have multiple suspended states is likely a driver decision. This also seems to be possible on Windows, but under certain conditions:

> Note that some devices are capable of several different low power modes within a single device power state. Such a device can use these modes if its driver can automatically switch the device from one mode to another without changing the device power state. As a general rule, however, if there is no user-perceptible difference between the modes, the device should use only the lowest power mode. If a low power mode, such as a low-speed mode, adversely affects performance or is not transparent to software other than the device driver, the hardware should not automatically use it. See the Device Class Power Management Reference Specification for details.

gmarull, 2021-09-10:
Some random thoughts, just to keep track of them:

- Rename `LOW_POWER` to `IDLE` (name makes things a bit more clear IMO, since `SUSPEND` also means low-power)
- Clarify how that links with runtime PM, where only `SUSPEND` is considered now.
- Drivers can implement `IDLE`, `SUSPEND` or both. Rules: `power_consumption(IDLE) > power_consumption(SUSPEND)`, `overhead(IDLE) < overhead(SUSPEND)`.
- System decides what to do when, e.g. suspending. For example, use `IDLE` if `SUSPEND` is not supported. Avoids drivers using the "fall-through" pattern, and aligns with previous point.

closed 2022-11-21 by gmarull