zephyrproject-rtos/zephyr #37057: PWM-blinky for Silabs MCU

LongNguyen1984, 2021-07-20:
**Is your feature request related to a problem? Please describe.**
I don't see any solution for Silabs MCU to run PWM examples.

**Describe the solution you'd like**
I think that someone can develop the code to configure PWM for Silabs MCU or can guild me to develop the code to use PWM of Silabs MCU

chrta, 2021-07-28:
PWM for Silabs MCUs was added in https://github.com/zephyrproject-rtos/zephyr/pull/33538. So you probably just have to adapt the device tree file and the Kconfig.defconfig.<socname> file of the soc. Please refer to the linked pull request as an example.

After that you should be able to run the blinky pwm example on your board.

LongNguyen1984, 2021-08-06:
In my case, I use Thunder Sense 2 Board. 
From the efr32mg_sltb004a.dts and efr32mg.dtsi files, they don't support PWM configuration. 
So it is impossible to build blinky_pwm, isn't it?

chrta, 2021-08-06:
The device tree files and the Kconfig file of the soc must be adapted for this, see my previous comment. But there is no need to write any C-Code for it, since the driver is already implemented, just not configured for this soc and board.

So it is possible by changing a few files, but not out-of-the-box.

LongNguyen1984, 2021-08-07:
I try to modified the files:  efr32mg.dtsi and efr32mg_sltb004a.dts 

/* My Include in efr32mg.dtsi*/
		timer0: timer@40018000 {
			compatible = "silabs,gecko-timers";
			reg = <0x40018000 0x400>;
			status = "disabled";
			label = "TIMER_0";

			pwm {
				compatible = "silabs,gecko-pwm";
				status = "disabled";
				label = "PWM_0";
				#pwm-cells = <2>;
			};
		};
/************************/
And the efr32mg_sltb004a.dts similar to the file in link #33538

But the compile will return: 
...
No SOURCES given to Zephyr library: drivers__pwm
...

LongNguyen1984, 2021-08-07:
After I modified the Kconfig.deconfig file:

config PWM_GECKO
	default y
	depends on PWM

Now I can compile but it not fading the LED. 
In the efr32mg_sltb004a.dts file, for describing pwm0:

&timer0 {
	status = "okay";

	pwm0: pwm {
		status = "okay";
		pin-location = <GECKO_LOCATION(26) GECKO_PORT_D GECKO_PIN(8)>;
		prescaler = <1024>;
	};
};

I intend to falling the LED0 of Thunderboard Sense2 at Pin 8 of GPIOD. 
But it is not work

chrta, 2021-08-09:
Please test https://github.com/zephyrproject-rtos/zephyr/pull/37534 I wired the PWM output to the green led, since only this pin is controllable from timer0 (not the red one).

LongNguyen1984, 2021-08-09:
Thanks Chrta, it works for the green led on Thunderboard Sense 2 now. 
One more question, how to select the location value in GECKO_LOCATION as in your setup

<GECKO_LOCATION(17) GECKO_PORT_D GECKO_PIN(9)>;

Where can I find the document to derive this value?

chrta, 2021-08-09:
In the datasheet of the soc https://www.silabs.com/documents/public/data-sheets/efr32mg12-datasheet.pdf in chapter `6.8 GPIO Functionality Table` on pg. 142ff. You can search for `TIM0_CC0` since timer0 is used here. The index in `GECKO_LOCATION()` is the number behind `TIM0_CC0 #` for the corresponding pin. This value is written into the location register of the peripheral.

Or you can have a look in the following chapter `6.9 Alternate Functionality Overview`

closed 2021-09-03 by cfriedt