zephyrproject-rtos/zephyr #37101: RFC: API change: Add SMbus support

ematt, 2021-07-21:
## Introduction

Extend I2C driver with SMbus support. 
This proposal is greatly inspired from the [I2C/SMBus subsystem from Linux](https://www.kernel.org/doc/html/latest/i2c/index.html)

### Problem description

SMbus block read cannot be performed with the current API. `i2c_api_full_io_t` can be used to perform SMbus operations, however SMbus block read is impossible because of the `Byte count` field.

### Proposed change

Add SMbus API by extending the existing I2C API.

1. There are two main differences between I2C and SBbus:
- SMbus has a speed limit of minimum 10KHz and maximum of 100KHz. I suggest to not implement any limitation in this regard, let the user be in charge of this.
- SMbus operations have a timeout of 35ms. Each SMbus operation must return ETIMEDOUT after this amount of time

2. Not all hardware can support all SMbus operations. A new function should be added for the user to check the driver functionality. 

3. New functions should be added for all SMbus operations: quick command, read/write byte/word, read/write block data

### Proposed change (Detailed)

Extend `i2c_driver_api` with the following functionality:
```
int i2c_api_smbus_scan_t(uint16_t address) - Checks if the device with a given address is present on bus
uint32_t i2c_api_get_functionality(const struct device *dev) - returns bit encoded functionality of the driver. See defines below

// SMbus specific commands
int i2c_api_smbus_quick_command_t(const struct device *dev)
int i2c_api_smbus_read_byte_t(const struct device *dev)
int i2c_api_smbus_write_byte_t(const struct device *dev, uint8_t value)
int i2c_api_smbus_read_byte_data_t(const struct device *dev, uint8_t command)
int i2c_api_smbus_write_byte_data_t(const struct device *dev, uint8_t command, uint8_t value)
int i2c_api_smbus_read_word_data_t(const struct device *dev, uint8_t command)
int i2c_api_smbus_write_word_data_t(const struct device *dev, uint8_t command, uint16_t value)
int i2c_api_smbus_read_block_data_t(const struct device *dev, uint8_t command, uint8_t *data, size_t data_len)
int i2c_api_smbus_write_block_data_t(const struct device *dev, uint8_t command, uint8_t data. size_t len)
```

Add the following defines to describe driver functionality:
```
I2C_FUNC_I2C - Plain i2c-level commands (Pure SMBus adapters typically can not do these)
I2C_FUNC_10BIT_ADDR - Handles the 10-bit address extensions
I2C_FUNC_SMBUS_QUICK - Handles the SMBus write_quick command
I2C_FUNC_SMBUS_READ_BYTE - Handles the SMBus read_byte command
I2C_FUNC_SMBUS_WRITE_BYTE - Handles the SMBus write_byte command
I2C_FUNC_SMBUS_READ_BYTE_DATA - Handles the SMBus read_byte_data command
I2C_FUNC_SMBUS_WRITE_BYTE_DATA - Handles the SMBus write_byte_data command
I2C_FUNC_SMBUS_READ_WORD_DATA - Handles the SMBus read_word_data command
I2C_FUNC_SMBUS_WRITE_WORD_DATA - Handles the SMBus write_byte_data command
I2C_FUNC_SMBUS_PROC_CALL - Handles the SMBus process_call command
I2C_FUNC_SMBUS_READ_BLOCK_DATA - Handles the SMBus read_block_data command
I2C_FUNC_SMBUS_WRITE_BLOCK_DATA - Handles the SMBus write_block_data command
```

Each of i2c_api_smbus_* should return ETIMEDOUT when the timeout duration has passed.

### Dependencies

All I2C drivers. 
At least `i2c_api_get_functionality` should be implemented in each driver.

### Concerns and Unresolved Questions

## Alternatives

Add a new flag for `i2c_msg` to instruct `i2c_api_full_io_t` to read the no. of bytes received at the first byte read.

ematt, 2021-07-21:
@henrikbrixandersen, here is the follow up of https://github.com/zephyrproject-rtos/zephyr/pull/36664

carlescufi, 2021-08-16:
@ematt Thank you for your proposal. I will include it in next week's API meeting.

carlescufi, 2021-08-31:
API meeting:

- @henrikbrixandersen raises the question of the 35ms timeout: our current I2C APIs do not support timeouts nor a cancel API, so this might require changing the I2C API itself
- @nashif points out that Embedded Controllers will probably benefit from SMBus support (@sjg20)
- Should SMBus be part of the I2C API or should it be a separate subsystem that uses the I2C API internally? @henrikbrixandersen to check the Linux implementation

closed 2023-03-22 by nashif