zephyrproject-rtos/zephyr #37103: ESP 8266 WIFI Shield Access Point cannot be enabled

moncef-bibani, 2021-07-21:
**Describe the bug**
i am trying to use the ESP 8266 WIFi Shield with the Board Nucleo F429Zi. I tested the connection to my WiFI network and it worked.  
my idea for my project is to run the ESP 8266 WIFI Shield as an Access Point with a WebServer to set up the WIFI Network Connection. After the Set Up the ESP WIFI should turn off. 

In my Code ( see below) , i set up the SSID as "ESPAP" without password for the Access point. The access point failed and is not enabled. I changed the parameters but it is still not working. Could someone of you help me to solve this Problem? Thank You!

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nucleo_f429zi myproject/esp_ap -- -DSHIELD=esp_8266
2. See CODE
**CODE**
/* Main */

void
main (void)
{
    struct net_if *default_net_iface = net_if_get_default();
    static struct wifi_connect_req_params wifi_conn_params;

    printf("Wifi Test\n");

    net_mgmt_init_event_callback(
        &wifi_management_callback,
        wifi_management_event_handler,
        WIFI_MANAGEMENT_EVENTS);

	net_mgmt_init_event_callback(
        &dhcp_management_callback,
        dhcp_management_event_handler,
		NET_EVENT_IPV4_ADDR_ADD);

    net_mgmt_add_event_callback(&wifi_management_callback);
	net_mgmt_add_event_callback(&dhcp_management_callback);

    printf("Network management setup complete, now wasting some time...\n");

    k_sleep(K_SECONDS(5));
    wifi_conn_params.ssid = "ESPAP";
    wifi_conn_params.ssid_length = strlen("ESPAP");
    wifi_conn_params.channel = WIFI_CHANNEL_ANY;
    wifi_conn_params.security = WIFI_SECURITY_TYPE_NONE;

    if (net_mgmt(NET_REQUEST_WIFI_AP_ENABLE, default_net_iface,
		     &wifi_conn_params, sizeof(struct wifi_connect_req_params))) {
		printf("AP mode failed\n");
		return;
	} else {
		printf("AP mode enabled\n");
    }
    k_sleep(K_SECONDS(60));
**Expected behavior**
I excepted that the ESP WiFi Shield acts as an Access Point. The WiFi SSID "ESPAP" should be displayed in the WiFi network without password.

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain  Zephyr SDK

nandojve, 2021-07-27:
Hi @cfriedt ,

I create the shield config that uses esp-at driver. However, I never tried to use it as a server.
I know that @mniestroj  is a major contributor for esp-at driver, maybe he can help us to understand current status.
@mniestroj do you know if is it possible use esp-at as server ? Is there any sample configuration ?

CC: @sylvioalves

mniestroj, 2021-07-27:
@moncef-bibani I tried using ESP8266 with ESP-AT v2.1 in Access Point mode. This worked with following shell command:
```
uart:~$ wifi ap enable ESPAP 5
AP mode enabled
```
after that I was seeing "ESPAP" WiFi network from my smartphone.

Keep in mind that various ESP-AT firmware versions might behave slightly different. I recommend ESP-AT v2.1, as it is probably tested the most. I am not using Access Point, but at least it is visible by smartphone correctly.

@nandojve Using esp-at as a server is another case. This is not supported (listen is not implemented) and there is an enhancement already reported in #37113.

nandojve, 2021-07-27:
Hi @moncef-bibani ,

This is the right command to use Arduino Serial ( usart6 ).
`west build -b nucleo_f429zi myproject/esp_ap -- -DSHIELD=esp_8266_arduino`

Did you tried ?
`west build -b nucleo_f429zi samples/net/wifi -- -DSHIELD=esp_8266_arduino`

Just to make sure everything is OK.

moncef-bibani, 2021-08-03:
Hi @nandojve @mniestroj ,
i tried the command with Arduino Serial. It´s still not working to enable the ap mode.

the wifi sample project from zephyr is worked fine and i could start the Access Point mode.
 Everything works fine in my code except the NET_REQUEST_WIFI_AP_ENABLE. I could connect to a wifi network with the net_mgmt procedure with "NET_REQUEST_WIFI_CONNECT" .
 I installed the firmware the AT V1.7 on the ESP-01S.

i don´t what the problem is, please help me to resolve it.

Thank you very much!

closed 2021-10-17 by github-actions[bot]

akrenz, 2023-10-27:
I had the same problem. Changing the parameter 

`wifi_conn_params.channel = WIFI_CHANNEL_ANY;`

to

`wifi_conn_params.channel = 5;`

solved it for me with ESP AT Firmware 2.1