zephyrproject-rtos/zephyr #37113: ESP8266 as a server

gebauerf, 2021-07-21:
**Is your enhancement proposal related to a problem? Please describe.**
I want to add WiFi functionality to my project. At the time i'm using hardware-networking with an ethernet-shield attached to my host MCU (nRF9160) to test and develop a webserver based on civetweb. Now i tried to include the ESP8266 as a slave for WiFi functionality. I'd like to access my webserver when the ESP is an AP-mode. First i evaluated the ESP with the wifi-sample from zephyr which worked fine for station-mode. After that, i integrated the ESP to my project. It seems that the network interface comes up as expected, but when the webserver gets initialized it can't listen to the defined port. When looking at the implementation of the driver, one can see that esp_listen and esp_accept is not implemented yet (**esp_offload.c**, line **33** & line **154**).
My log output is the following:

```
*** Booting Zephyr OS build v2.4.99-ncs2-2-gc835e166d004  ***
SPM: prepare to jump to Non-Secure image.
[00:00:00.202,667] <dbg> net_if.net_if_init: (main):
[00:00:00.208,526] <dbg> net_if.init_iface: (main): On iface 0x200190d0
[00:00:00.460,876] <inf> wifi_esp: Waiting for interface to come up
[00:00:00.889,038] <inf> wifi_esp: ESP Wi-Fi ready
[00:00:00.894,531] <dbg> net_if.net_if_up: (esp_workq): iface 0x200190d0
[00:00:00.902,160] <dbg> net_if.net_if_ipv6_calc_reachable_time: (main): min_reachable:15000 max_reachable:45000
[00:00:00.959,167] <dbg> net_if.net_if_post_init: (main):
[00:00:01.019,805] <inf> net_config: Initializing network
[00:00:01.025,939] <inf> net_config: Running dhcpv4 client...
[00:00:01.138,427] <dbg> lib_extensions.sscanf: [IMPLEMENTATION MISSING : sscanf]
[00:00:01.207,519] <dbg> wifi_esp_offload.esp_get:
[00:00:01.228,118] <err> mg_cry_internal_impl: set_ports_option @ 15073 in civetweb.c cannot listen to 8080: 35 (ENOTSUP)[00:00:01.254,882] <dbg> wifi_esp_offload.esp_recv: link_id 0, timeout 0, cb 0x0, data 0x0
[00:00:01.279,052] <err> mg_cry_internal_impl: mg_start2 @ 19913 in civetweb.c Failed to setup server ports Unable to start the server.
```

**Describe the solution you'd like**
Implementing the missing parts of the esp driver (listen, accept).

**Describe alternatives you've considered**
I saw that there is a great interest to use the ESP8266 as a server. See the issues [31376](https://github.com/zephyrproject-rtos/zephyr/issues/31376) or [37103](https://github.com/zephyrproject-rtos/zephyr/issues/37103).

**Additional context**
[image]

moncef-bibani, 2021-07-22:
Hello, i opened the issue #https://github.com/zephyrproject-rtos/zephyr/issues/37103 
how could you started the ESP mode AP. i see in your log that you used the esp driver directly in your project, how you did that ? 
i used this procedure to start the AP and it has not worked:  **_net_mgmt(NET_REQUEST_WIFI_AP_ENABLE, default_net_iface,
	     &wifi_conn_params, sizeof(struct wifi_connect_req_params))_** 

how did you integrate the driver esp.c and esp_offload.c in your project ? did you use in your west build command the esp_8266 as shield ?

gebauerf, 2021-07-22:
Hi,
i integrated the esp driver over the west.yml file. Add the expression **hal_espressif** to the file and afterwards use the _west update_ command. Instead of using the "shield" extension in the west build command, i integrated my own device description in a devicetree overlay file like this:
```
//Add support for ESP8266 WiFi-shield on arduino-header of the nRF9160DK
&uart3 {
  status = "okay";
  compatible = "nordic,nrf-uarte";
  tx-pin = <12>;
  rx-pin = <13>;
  current-speed = <115200>;

  esp8266 {
    status = "okay";
    compatible = "espressif,esp";
    label = "esp8266";
  };
};
```
For the Nucleo Board this may look a bit different. You should check the binding indexes for the Nucleo.

To initialize the ESP, add this configuration flags to your prj.conf file:
```
CONFIG_WIFI=y
CONFIG_WIFI_ESP=y
CONFIG_WIFI_ESP_AT_VERSION_1_7=y
CONFIG_WIFI_LOG_LEVEL_DBG=y
CONFIG_NET_IF_LOG_LEVEL_DBG=y
```
You may need some additional configurations like:
```
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_NET_NATIVE=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV4=y
```
Some of them depend on my webserver configuration, just try it out.

The esp driver uses the AT-commands to communicate with the chip. I loaded the AT-Firmware v1.7 on my ESP, available from the [Espressif Webpage](https://www.espressif.com/en/support/download/at). This procedure caused some troubles, i would recommend [this thread](https://github.com/espressif/ESP8266_NONOS_SDK/issues/168#issuecomment-418459533) when problems occur.
Hope this helps a little.

moncef-bibani, 2021-07-22:
Thank you very much for your help ! i got the configuration in the prj.conf , i am juste confusing how to use this driver in my project. shall i include the libraries esp.c & esp_offload.c in my project and call this function **_device_get_binding(esp_8266)_** ?? could you send an sample project of your main ? this will help me a lots. Thanks!

mniestroj, 2021-07-27:
@gebauerf Contribution of listen in esp-at driver is welcome! So far I do not plan to do that myself.

kartben, 2026-10-30:
Server (listen/accept) support for the esp-at driver is tracked in #51642, so closing this as a duplicate.

closed 2026-10-30 by kartben