zephyrproject-rtos/zephyr #37132: Assert on enabling Socket CAN

rbabub, 2021-07-22:
Assert while running socket can application seen during interface initialization.

ASSERTION FAIL [net_if_get_link_addr(iface)->addr != ((void *)0)] @ WEST_TOPDIR/zephyr/subsys/net/ip/net_if.c:3993

Flags added in prj.conf of socket can application,    CONFIG_ASSERT=y

**Expected behavior**
Socket CAN does not require interface link address to be assigned, above check is not applicable for socket can interfaces.
As this address is NULL for socket CAN interface it results in assertion.

**Logs and console output**
ASSERTION FAIL [net_if_get_link_addr(iface)->addr != ((void *)0)] @ WEST_TOPDIR/zephyr/subsys/net/ip/net_if.c:3993
[00:00:00.138,092] <err> os: Faulting instruction address (r15/pc): 0x6000f8bc
[00:00:00.146,820] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.155,334] <err> os: Current thread: 0x60081160 (unknown)
[00:00:00.162,841] <err> os: Halting system

Note:
This check was introduced part of commit id "1e4f268ea94066cfc60c80019cf515b2baafc167".

alexanderwachter, 2021-07-27:
@carlescufi we rather need a network stack expert here. Ping @rlubos

alexanderwachter, 2021-07-27:
For me it looks like we need something like  `| (IS_ENABLED(SOCKET_CAN) && is_socket_can(iface))` in the assert

rlubos, 2021-07-28:
> For me it looks like we need something like `| (IS_ENABLED(SOCKET_CAN) && is_socket_can(iface))` in the assert

Well if CAN interfaces indeed do not specify the LL address on the interface then indeed the above assertion check does not make sense (I'm assuming that CAN packet processing doesn't actually make use of this field across the stack). I wouldn't hovewer modify the assert condition itself as it's pretty clear right now, I'd rather encapsulate the entire assert with an if, i. e.
```
if (IS_ENABLED(SOCKET_CAN) && is_socket_can(iface) {
    /* Nothing to check. */
} else {
    /* assert here */
}
```
IMO it'd be easier to follow then.

alexanderwachter, 2021-07-28:
@rbabub can you make a PR with a fix?

rbabub, 2021-07-28:
ya sure, will check & submit the patch.

PR #37311 (2021-08-08): net: fix for assert on enabling socket CAN
Socket CAN does not require interface link address to be
assigned, check is not applicable for socket CAN
interfaces. As this address is NULL for socket CAN interface
it results in assertion.

Fixes #37132
closed 2021-08-08 by cfriedt