zephyrproject-rtos/zephyr #37160: Bootloader should provide the version of zephyr, mcuboot and a user defined version to the application

caco3, 2021-07-23:
We are using the MCUBoot based bootloader (`/bootloader/mcuboot`).

It seems there is no way for an application to get the following information from the bootloader:
- The used Zephyr version in the bootloader
- The used MCUBoot version
- A user defined bootloader version (like set in the application with the image tool, `imgtool sign --version $VERSION_FOR_IMGTOOL ...`)

We think that it is important that an application is able to read this information from the bootloader for troubleshooting or special handling of a newer bootloader version.

Did I miss such functionality or is it not existing? And if so, is there a reason for it to be missing?

As a workaround for our test system, I added such functionality by providing this information through the UART log:
```bash
# Add Git hash, MCUboot repo commit hash/tag and zephyrproject version to Startup Log Message
sed -i "/BOOT_LOG_INF(\"Starting bootloader/c\    BOOT_LOG_INF(\"Starting bootloader | \" __DATE__ \" \" __TIME__ \" | Version: $GIT_VERSION | MCUboot: $MCUBOOT_GIT_VERSION | Zephyrproject: $ZEPHYR_REVISION\");" mcuboot/boot/zephyr/main.c
```
How ever this is not be possible on the systems on the field.

I think the way to implement this is by defining fix locations in flash and store the mentioned information there.
The right place most likely would be at the end of the `boot_partition`, https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf52840dongle_nrf52840/fstab-debugger.dts#L18.
I have done this on other projects (not Zephyr based) and it worked well.

Laczen, 2021-07-26:
@caco3 this is a question for mcuboot. Mcuboot should decide to add a version and how an application can access it. It is also possible to reserve a little ram buffer to share this data with the application.

de-nordic, 2021-07-26:
@caco3 Can you move the issue here: https://github.com/mcu-tools/mcuboot ?

caco3, 2021-07-28:
@de-nordic Thanks, I created it in https://github.com/mcu-tools/mcuboot/issues/1079

closed 2021-07-28 by caco3