zephyrproject-rtos/zephyr #37175: nucleo-f756zg: rtos aware debugging not working

maksimdrachov, 2021-07-23:
**Describe the bug**

In [this](https://www.youtube.com/watch?v=jR5E5Kz9A-k) video it is shown how to debug a reel-board using gdb.

**To Reproduce**
The following commands are issued:
```
cd zephyr/samples/synchronization
west build -b nucleo_f756zg
west build --target guiconfig (checking: )
west debug
(gdb) c
```
Output looks fine
[image]

If I interrupt, it looks like this, seemingly no threads detected by gdb?
[image]

**Expected behavior**
However I'm expecting something like this (as shown in the video)
[image]

I've checked my Konfig, but the problem remains.
[image]

**Impact**
I can't really start developing with Zephyr.

**Environment (please complete the following information):**
 - OS: MacOS
 - Board: Nucleo-F756ZG
 - Commit SHA: f3d9808012

erwango, 2021-07-26:
@maksimdrachov Thread awareness support was only merge relatively recently in openocd (http://openocd.zylin.com/#/c/4988/).
Can you check you're using a version that embed this patch ?

maksimdrachov, 2021-07-26:
Hmm you're right

Upgraded openocd to the latest commit:
[image]

I've removed the previous install to make sure there was no chance of the older version being used. Used the following commands:
```
brew uninstall openocd
brew install openocd --HEAD
```

Serial output seems to be working as expected:
[image]

However the same issue remains:
[image]

Any other suggestions?
Should I take this to openocd?
Are there any tools you could recommend that work with STM32 and Zephyr?

erwango, 2021-07-26:
@maksimdrachov Ok, thanks for having had a try. I'll have a check on my side as well.
As an alternative, you should be able to use pyocd.
Also, you could use Segger, but please note that it currently dos not work for some targets, so I'd not advise it right now.

erwango, 2021-07-26:
@maksimdrachov One additional piece of info:
In the video you mention, pycocd is the debugger in use.

For openocd, one additional config can be required (to be added in <board>/support/openocd.cfg)
```
# Enable Zephyr thread awareness.
$_TARGETNAME configure -rtos Zephyr
```
But I've not been able to test yet.

maksimdrachov, 2021-07-26:
Ok, I'll try to test 👍

maksimdrachov, 2021-07-26:
It works! Big thanks!

The proposed (working) solution can be seen in the screenshot below:
[image]

erwango, 2021-07-26:
ok, great. We need need to set this by default. I'll raise an issue for that.

erwango, 2021-07-26:
@maksimdrachov I'm closing current issue.

closed 2021-07-26 by erwango