zephyrproject-rtos/zephyr #37186: entropy: Bluetooth derived entropy device

JordanYates, 2021-07-25:
**Is your enhancement proposal related to a problem? Please describe.**

Some devices have no access to dedicated entropy hardware, but are running the Bluetooth host against some external controller. Not having an entropy device limits the options for `RNG_GENERATOR_CHOICE` to `TIMER_RANDOM_GENERATOR`,  which is obviously unacceptable in production applications (The Kconfig option itself specifies it is only for testing).

This results in applications that are unable to reasonably use `sys_rand32_get` and `sys_rand_get`.

One example of the above situation is the nRF5340, where the application core has no access to the RNG peripheral on the network core.

**Describe the solution you'd like**

The Bluetooth HCI command `HCI_LE_Rand` (0x2018) can be used to query 8 bytes of random data, and can therefore be used as a pseudo entropy device. 

Note that the performance of this is probably poor enough (especially if the HCI is over UART/SPI instead of rpmsg) to default `RNG_GENERATOR_CHOICE` to `XOROSHIRO_RANDOM_GENERATOR` when using the Bluetooth HCI command as the entropy source.

**Additional context**

All existing entropy devices are described in devicetree and chosen via `zephyr,entropy`. This would therefore require a node which doesn't strictly correspond with a physical device. Examples of devices like this do already exist in tree (see `segger,rtt-uart.yaml`).
```
	entropy: bt_hci_entropy {
		compatible = "zephyr,bt-hci-entropy";
		label = "bt_hci_entropy";
		status = "okay";
	};
```

Currently the random subsystem is initialized at `PRE_KERNEL_2` `CONFIG_KERNEL_INIT_PRIORITY_DEFAULT`, as the entropy devices are all initialized at `PRE_KERNEL_1` `CONFIG_KERNEL_INIT_PRIORITY_DEVICE`. This setup will not work with a Bluetooth entropy device as it is not available until `bt_enable` is called, which can be as late as the application main loop.

A potential solution to the above is to add a new symbol `RNG_GENERATOR_DEFER`, which would defer calling `entropy_get_entropy` until the first request for random data.

tejlmand, 2022-02-17:
> One example of the above situation is the nRF5340, where the application core has no access to the RNG peripheral on the network core.

This is an extremely bad example, cause the application has access to cryptocell which provides RNG.

frkv, 2022-02-17:
Multiple cores may have multiple ways of providing entropy, but the process of asking for and transferring the entropy from one core to another may impede on the security. 

I can't talk for all devices (and therefore I can't answer this RFC on a generic level) but I can talk for the nRF5340. You should not utilize the entropy-source from the network core in app-core of this device. This is not how this device was designed. The nRF5340 app-core should use its local peripheral for entropy, namely the CryptoCell HW peripheral. Providing a way to do this using the entropy source in the network core using IPC has security implications and implications on Root of Trust on the device. 

This is especially important when considering the TrustZone security extensions and the SPU that is available on this device that is part of the overall security of using the CryptoCell HW peripheral and the Chain-of-trust that this can establish.

tejlmand, 2022-02-17:
> Great, how does one go about using this in mainline Zephyr?

I just checked up on this, and it seems the rng driver for cryptocell has still not been upstreamed (sorry for this confusion, I thought this had been upstreamed already)
This means the driver is only available as part of [nRF Connect SDK](https://github.com/nrfconnect/sdk-nrf)

Then I'm afraid there is really not any good solution to this problem for time being, except the suboptimal netcore strategy 😞 .
Not sure if @frkv is able to provide more details on this.