zephyrproject-rtos/zephyr #37188: Get an error  of "Illegal load of EXC_RETURN into PC" when print log in IO interrupt callback

ThisSummerRace, 2021-07-25:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
I print log both in the UART callback function and  IO interrupt callback function , unless remove the log in  IO interrupt callback, it will get the error of "Illegal load of EXC_RETURN into PC" sometimes.
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

carlescufi, 2021-07-27:
@ThisSummerRace you'll need to be a bit more specific here. What are you using exactly for logging? What board can you reproduce this with? Which sample?

ThisSummerRace, 2021-08-01:
> @ThisSummerRace you'll need to be a bit more specific here. What are you using exactly for logging? What board can you reproduce this with? Which sample?

1、I use the  RTT for logging
2、I do the sample with homekit,not a standard sample on zephyr source code.

cfriedt, 2021-09-23:
@ThisSummerRace - please reopen the issue if you can provide something reproducible with minimal changes to an upstream sample or test case. We currently do not have enough information to verify that this is a bug let alone triage it.

E.g. if you were to provide a `.patch` file for some upstream test or sample application and provide something like.
```
west build -b qemu_cortex_a53 -t run samples/foo/bar
```

Also, please fill in fields such as the toolchain, OS, and commit sha

closed 2021-09-23 by cfriedt