zephyrproject-rtos/zephyr #37189: Bug "Key 'name', 'cmake-ext' and 'kconfig-ext' were not defined" when build a zephyr application

nqduy35, 2021-07-25:
**Describe the bug**
After finishing the zephyr installation following the guide proposed by Zephyr, I cannot build any program (even hello_world with native_posix) and this problem always produce. 

**To Reproduce**
Steps to reproduce the behavior:
1. After following the guide : https://docs.zephyrproject.org/latest/getting_started/installation_linux.html 
2. cd ~/zephyrproject/zephyr/samples/hello_world
3. cmake build -b native_posix .
4. See error

**Logs and console output**

> -- west build: generating a build system
> Including boilerplate (Zephyr base): /home/abc/zephyrproject/zephyr/cmake/app/boilerplate.cmake
> CMake Deprecation Warning at /home/abc/zephyrproject/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
>   The OLD behavior for policy CMP0079 will be removed from a future version
>   of CMake.
> 
>   The cmake-policies(7) manual explains that the OLD behaviors of all
>   policies are deprecated and that a policy should be set to OLD only under
>   specific short-term circumstances.  Projects should be ported to the NEW
>   behavior and not rely on setting a policy to OLD.
> Call Stack (most recent call first):
>   /home/abc/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
>   /home/abc/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
>   CMakeLists.txt:4 (find_package)
> 
> 
> -- Application: /home/abc/zephyrproject/puber
> -- Zephyr version: 2.4.0 (/home/abc/zephyrproject/zephyr)
> -- Found Python3: /usr/bin/python3 (found suitable exact version "3.7.3") found components: Interpreter
> -- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
> CMake Error at /home/abc/zephyrproject/zephyr/cmake/zephyr_module.cmake:48 (message):
>   validation.invalid
> 
>    --- All found errors ---
> 
>   ["Key 'name' was not defined.  Path: ''", "Key 'cmake-ext' was not defined.
>   Path: '/build'", "Key 'kconfig-ext' was not defined.  Path: '/build'"]
> 
>   ERROR: Malformed "build" section in file:
>   /home/abc/zephyrproject/modules/lib/canopennode/zephyr/module.yml
> 
>   <SchemaError: error code 2: Schema validation failed:
> 
>    - Key 'name' was not defined. Path: ''.
>    - Key 'cmake-ext' was not defined. Path: '/build'.
>    - Key 'kconfig-ext' was not defined. Path: '/build'.: Path: '/'>

**Environment:**
 - OS: Debian 9 on Virtual Box
 - Toolchain: Zephyr SDK 0.12.3
 - Zephyr v2.4.0

carlescufi, 2021-07-27:
This might be an incompatibility between the latest west and Zephyr 2.4.0? @mbolivar-nordic

mbolivar-nordic, 2021-07-28:
I can't reproduce. I'm guessing you're mixing old and new versions of zephyr and module repositories, or have forgotten to run `west update`.

@nqduy35 I also saw your bug report https://github.com/zephyrproject-rtos/west/issues/512. I couldn't reproduce that either. I think you may have something wrong with your environment.

Looking at this in the error output:

```
["Key 'name' was not defined. Path: ''", "Key 'cmake-ext' was not defined.
Path: '/build'", "Key 'kconfig-ext' was not defined. Path: '/build'"]

ERROR: Malformed "build" section in file:
/home/abc/zephyrproject/modules/lib/canopennode/zephyr/module.yml
```

This isn't a west issue. The `name`, `cmake-ext`, and `kconfig-ext` keys are related to the module.yml file format, which is not read by west. That format is read by the zephyr build system directly.

In the bug report, you wrote:

> cmake build -b native_posix .

I'm guessing you meant `west build -b native_posix .` instead, right?

That works fine for me on zephyr v2.4.0 with west 0.11.0 after running `west update`.

You can also try making sure all your workspace repositories are completely clean by running `west forall -c 'git clean -ffdx'`, **BUT** be careful to commit all your work first so you don't lose anything.

mbolivar-nordic, 2021-07-28:
Downgrading from bug to question until we can reproduce, given the above.

nqduy35, 2021-07-28:
I followed your suggestions (`west update` and `west forall -c 'git clean -ffdx'`) and also downgraded the west from v0.11.0 to v0.8.0 (the west version compatible with zephyr v2.4.0) and everything works fine now. Thank @mbolivar-nordic and @carlescufi for your supports.

closed 2021-07-28 by nqduy35