zephyrproject-rtos/zephyr #37191: nrf5340: Support +3dBm TX power

JordanYates, 2021-07-26:
**Is your enhancement proposal related to a problem? Please describe.**

The nRF5340 radio supports up to +3dBm TX power, but it is not currently possible to enable this in Zephyr.
The dependancies do not allow values higher than 0dBm to be selected.

https://github.com/zephyrproject-rtos/zephyr/blob/7d07a941faf49d4f8d5b4fc35a9acfde71b1e3f3/subsys/bluetooth/controller/Kconfig#L193-L197

**Describe the solution you'd like**

Ability to select `BT_CTLR_TX_PWR_PLUS_3` when building the Bluetooth controller.

**Additional context**

Enabling this is not quite as simple as updating the Kconfig symbol dependancies, as the nRF5340 needs register writes to enable +3dBm (`VREGRADIO.VREQH`).

joaodullius, 2021-07-30:
@JordanYates Just do double-check, you are referring to the Zephyr Bluetooth Controller implementation and not the Nordic SoftDevice Controller, right?

JordanYates, 2021-07-30:
Yes, the Zephyr controller.
I don't know whether the Softdevice supports +3dBm or not.

cvinayak, 2021-07-30:
> Enabling this is not quite as simple as updating the Kconfig symbol dependancies, as the nRF5340 needs register writes to enable +3dBm (VREGRADIO.VREQH).

@JordanYates As I have to investigated the software requirements to enable +3dBm, you are probably better here to suggest changes in the radio.c file used by the Zephyr controller, i.e. may be in `radio_tx_power_set()`.

PR #44538 (2022-04-08): Bluetooth: Controller: Add +3dBm Tx power support in nRF53 Series
Add implementation in Controller to set radio high voltage
to enable support for +3dBm Tx Power in nRF53 Series SoCs.

Fixes #37191.