zephyrproject-rtos/zephyr #37193: mcumgr: Probably incorrect error handling with udp backend

emob-nordic, 2021-07-26:
In case there is no free space for mcumgr buffer, received data are probably
inproperly dropped [smp_udp.c:114-127](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/mcumgr/smp_udp.c#L114:L127).

For more information refer to:
https://github.com/zephyrproject-rtos/zephyr/issues/37105
https://github.com/zephyrproject-rtos/zephyr/pull/37081

carlescufi, 2021-11-09:
@emob-nordic are you looking into this?

emob-nordic, 2021-11-09:
@carlescufi I'm not. I was told in the past to write this issue so someone competent could address it later.

carlescufi, 2021-11-30:
@aunsbjerg could you please take a look at this issue?

aunsbjerg, 2021-11-30:
@carlescufi I will try, but am a bit pressed for time, so it will be a few days before I can do anything

de-nordic, 2022-01-31:
I'll take over. I have idea how to follow with this, although it requires change in mcumgr/lib.

nordicjm, 2023-04-04:
@de-nordic the file has since moved and changed, what parts are wrong in the UDP handling?

nordicjm, 2023-06-06:
It is assumed that this is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/44536 whereby if there are insufficient buffers to receive the command, no error response is sent.

closed 2023-06-06 by nordicjm