zephyrproject-rtos/zephyr #37202: esp32c3 build error

lgl88911, 2021-07-26:
**Describe the bug**

The following error occurred when building esp32c3

> -- Found toolchain: espressif (/home/frank/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf)
> CMake Error at /mnt/e/tomato/zephyrproject/zephyr/cmake/compiler/gcc/generic.cmake:8 (message):
>   Zephyr was unable to find the toolchain.  Is the environment misconfigured?
> 
>   User-configuration:
> 
>   ZEPHYR_TOOLCHAIN_VARIANT: espressif
> 
>   Internal variables:
> 
>   CROSS_COMPILE:
>   /home/frank/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/xtensa-esp32-elf-
> 
> 
>   TOOLCHAIN_HOME:
>   /home/frank/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf
> 
> 
> Call Stack (most recent call first):
>   /mnt/e/tomato/zephyrproject/zephyr/cmake/generic_toolchain.cmake:42 (include)
>   /mnt/e/tomato/zephyrproject/zephyr/cmake/app/boilerplate.cmake:570 (include)
>   /mnt/e/tomato/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
>   /mnt/e/tomato/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
>   CMakeLists.txt:5 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!

**To Reproduce**
Steps to reproduce the behavior:
1. Setup environment according to https://docs.zephyrproject.org/latest/boards/riscv/esp32c3_devkitm/doc/index.html2. cmake -DBOARD=board\_xyz
2. west build -b esp32c3_devkitm samples/hello_world
3. See error

**Additional context**
Modify cmake/toolchain/espressif/generic.cmake can solve this problem. This is only a temporary solution. The correct approach should be to select the target name according to the toolchain.
```
set(CROSS_COMPILE_TARGET riscv32-esp-elf)
set(SYSROOT_TARGET       riscv32-esp-elf)
```

sylvioalves, 2021-07-27:
@lgl88911 This https://github.com/zephyrproject-rtos/zephyr/pull/36818 fixes that. Thanks!

closed 2021-08-09 by cfriedt

lgl88911, 2021-09-05:
@sylvioalves Thanks, Verify pass at v2.6.0-rc1-301-gd9c666a5abf8