zephyrproject-rtos/zephyr #37205: openocd: Configure thread awareness by default

erwango, 2021-07-26:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr thread awareness support was merged sometime ago in oponocd (http://openocd.zylin.com/#/c/4988/)

To enjoy this feature, 2 configuration steps are required:
- Kconfig: `CONFIG_DEBUG_THREAD_INFO=y`
- openocd: `$_TARGETNAME configure -rtos Zephyr`

Since the openocd part is not documented, users may find issues to enable it (cfhttps://github.com/zephyrproject-rtos/zephyr/issues/37175)

**Describe the solution you'd like**
Add the openocd thread awareness support to all openocd enabled boards by default.
Then users will only have to deal with the documented Kconfig part.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2021-07-26:
@MaureenHelm, @nandojve, @galak. FYI

trond-snekvik, 2021-07-26:
> openocd: $_TARGETNAME configure -rtos Zephyr

This can also be done for JLink if the JLink GDB Server version is 7.22 or higher