zephyrproject-rtos/zephyr #37206: counter: stm32: Missing implementation of set_top_value

erwango, 2021-07-26:
**Is your enhancement proposal related to a problem? Please describe.**
Current implementation of counter api in stm32 rtc driver is a no-op:
```
static int rtc_stm32_set_top_value(const struct device *dev,
				   const struct counter_top_cfg *cfg)
{
	const struct counter_config_info *info = dev->config;

	if ((cfg->ticks != info->max_top_value) ||
		!(cfg->flags & COUNTER_TOP_CFG_DONT_RESET)) {
		return -ENOTSUP;
	} else {
		return 0;
	}
```

**Describe the solution you'd like**
Implement effective support for this feature with optional `COUNTER_TOP_CFG_DONT_RESET` if supported by HW.

FRASTM, 2021-07-27:
In the `counter_set_top_value`  api, there is a callback to be called when the top_value is reached: **counter_top_callback_t**
The `rtc_stm32_data` must be extended to include this callback from the  `rtc_stm32_isr`

FRASTM, 2021-08-12:
After Disabling the WritePortection on all the RTC registers, the following sequence is required to configure or change the wakeup timer (WUT) auto-reload value.
1. disable the wakeup timer.
2. Poll WUTWF until it is set in RTC_ISR to make sure the access to wakeup auto-reload counter and to WUCKSEL[2:0] bits is allowed. It takes 1 to 2 RTCCLK clock cycles (due to clock synchronization).
3. Program the wakeup auto-reload value and the wakeup clock selection.
4. Set WUTE in RTC_CR to enable the timer again.
The wakeup timer restarts down-counting.

The RTC wakeup (WUT) of the stm32 counts DOWN  from its load value max 0xFFFF to 0.
The RESET / DONT_RESET flag of the `counter_top_cfg`, change the value of the starting WUT counter :
- starts from the new value set (RESET mode)
- adds the current counter value to the new value set  (DONT_RESET mode). 
In any case this value could be maximized to 0xFFFF.

FRASTM, 2021-08-12:
Depending on the stm32 soc device, the RTC WakeUp interrupt has its own IRQ entry and EXTI line.
Consequently the DTS must be adapted to accept more than the ALARM interrupt entry and EXTI line.

Some devices, like the stm32L552 or stm32L562 have a common interrupt EXTI line for all the RTC functions.

FRASTM, 2021-08-12:
The zephyr API, explicitly mentions that " Top value can only be changed when there is no active channel alarm."
This is detected by the valid alarm_callback function.

Because the stm32 WUT is counting down, the  `-ETIME` return code is not identified:
    "_if COUNTER_TOP_CFG_DONT_RESET was set and new top value is smaller than current counter value (counter counting up)."_ It means , the new top value won't be reached.
In case of down counting, a proposal here is to return -ETIME if COUNTER_TOP_CFG_RESET_WHEN_LATE is NOT set and the new top value is **greater** than the current counter value.
In case of of down counting, if COUNTER_TOP_CFG_RESET_WHEN_LATE is set and  and the new top value is **greater** than the current counter value, then the counter is stopped, new value is loaded and counter is re-started.
In case of of down counting, if the new top value is **smaller** than the current counter value, then the counter is stopped, new value is loaded and counter is re-started.

FRASTM, 2021-08-18:
The zephyr API, explicitly mentions that " Top value can only be changed when there is no active channel alarm."
This is detected by the valid alarm_callback function.

When setting the new top value, the current position of the counter is compared.
if new top value to set > current counter position then resetting the counter depends on the flag COUNTER_TOP_CFG_DONT_RESET.
if new top value to set < current counter position then resetting the counter depends on the flag COUNTER_TOP_CFG_RESET_WHEN_LATE. if not set, the new top value cannot be set, returning an -ETIME error.

erwango, 2021-09-07:
Closing for now as too complex vs gain.

closed 2021-09-07 by erwango