zephyrproject-rtos/zephyr #37222: k_queue data corruption, override user data after reserved heading word

ruehlchris, 2021-07-27:
**Describe the bug**
In my application there are two queues used in two threads (consumer)
and a single thread (producer), feed two queues in interval of 60 seconds.
When I dequeue the message and copy the data after the reserved word to
my process buffer, it happens that the first word of my data becomes <0x00><0x00>.
The k_queue_append() command used to add the data to the queue, my guess is
that the bug is in the part of the code where " behind-the scenes memory allocation"
had been done.

**To Reproduce**
reproduces with:
code sample producer thread:
`char *memptr = k_malloc(data_length+K_QUEUE_HEADERSIZE);  /* add 2 bytes for k_queue header*/`
`    if (memptr) {`
`        /* copy data behind the header 2 byte offset */`
`        memcpy(memptr+K_QUEUE_HEADERSIZE,data,data_length);`
`        k_queue_append(queue,(void *)memptr);`
`    }`

code sample consumer thread:
`char *data = (char*) k_queue_get((struct k_queue *)queue,K_MSEC(300));`
`        if (data!=NULL) {`
`            LOG_INF("Send multicast mesh request queue(%p)",queue);`
`            coap_send_request(COAP_METHOD_PUT,`
`                              (const struct sockaddr *)&multicast_local_addr,`
`                              path, data+K_QUEUE_HEADERSIZE, sz, NULL);`
`            for(int i=0; i<sz; ++i) {`
`              printk("(%x)",data[i]);`
`            }`
`            printk("\n");`
`           k_free((void*)data);`
`        }`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Corrupted data, process logic will fail.

**Logs and console output**
hex print of the (char*) of the dequeued pointer:
overridden word[2]
(0)(0)**(0)(0)**(31)(39)(4d)(59)(30)(34)(32)(38)(f4)(2)(1c)(27)(8)(0)(0)(0)(0)(0)(0)(0)(45)(da)(4a)(0)                                                                                  
good response
(0)(0)(30)(37)(31)(39)(4d)(59)(30)(34)(32)(39)(15)(bb)(6)(6a)(1f)(6)(0)(5)(32)(f)(0)(51)(1d)(4f)(1)(a0)

**Environment (please complete the following information):**
 - OS: Linux Debian 10
 - Toolchain NCS, nrf v1.6.0 , Zephyr SDK detached: a62ea8fa29

ruehlchris, 2021-08-09:
Hi there,

I had a review of the k_queue implementation and found the reason for problem.
k_queue_append()
 calls internally
static int32_t queue_insert(struct k_queue *queue, void *prev, void *data,  bool alloc, bool is_append)
 with alloc=false, and is_append=true
which then calls
sys_sfnode_init(data, 0x0);

data is cast to "sys_sfnode_t *node" within sys_sfnode_init
and then the flag is stored at next_and_flags which is defined unative_t. 

#ifdef __LP64__
typedef uint64_t unative_t;
#else
typedef uint32_t unative_t;
#endif

struct _sfnode {
	unative_t next_and_flags;
};
typedef struct _sfnode sys_sfnode_t;

Documentation is not in sync with this and should be updated, for 32bit platforms a header for 4 bytes or two words
and for 64bit platforms 8 bytes must reserved as header of the memory pointer hand over to k_queue_append().

My fix is to change the definition to 4 bytes

#define K_QUEUE_HEADERSIZE 4

and solved the issue.

Regards
Chris

andyross, 2021-08-11:
The documentation is subtle but seems not incorrect to me.  e.g. for k_queue_append: "A queue data item must be aligned on a word boundary, and the first word of the item is reserved for the kernel's use."

It's true that it relies on the reader understanding that "word" is an architecture-dependent type, and it would be nicer to be more specific.  Similar ambiguity exists in the k_fifo and k_lifo utilities that are implemented as k_queue wrappers.

I think we can close this as a bug; not that I'd NAK a documentation fixup to make it clearer.  Please reopen if I've misunderstood.

closed 2021-08-11 by andyross