zephyrproject-rtos/zephyr #37223: Include Android's CHRE as a module

yperess, 2021-07-27:
## Origin

AOSP: https://android.googlesource.com/platform/system/chre/+/HEAD/

## Purpose

CHRE is Android's EC event loop and management of peripheral information. It allows separation of the [frameworks](https://android.googlesource.com/platform/system/chre/+/HEAD/pal/include/chre/pal) from the feature work (the consumers of the raw data). Having this as a module would allow Zephyr apps to write and include nanoapps compatible with CHRE to consume the raw peripheral data. This would avoid a lot of boiler plate code needed to communicate with these peripherals.

## Mode of integration

This should be a module name `chre`

## Pull Request

None yet, this was experimentally done as a subsystem for a single CHRE source file (to see what it would take to build it) at https://chromium-review.googlesource.com/c/chromiumos/platform/ec/+/2991393

## Description

CHRE provides a platform independent solution for handling and managing peripherals. Currently, Android devices are required to implement the PAL (platform abstraction layer) APIs in order to get CHRE up and running. At which point their existing nanoapps (which can be seen as features consuming the data from the peripherals) can be used. Note that nanoapps can be loaded statically OR dynamically in CHRE. Having CHRE available in Zephyr as a module would enable us to implement these frameworks such as the sensor framework using devicetree; at which point, consumers of the data would be able to avoid a lot of the boilerplate and complexity of the work. For the initial work, we'd like to focus on the sensor framework and some of these benefits  include:
* Having to manage multiple requests at different data rates.
* Managing timestamp spreading when sensor FIFO is used.
* Handling sensor orientation

On top of these benefits, we'd be able to supply the Zephyr source tree with common nanoapps that consume data from these frameworks. Again, in the case of sensors these might include:
* Impact detection.
* Online (live) calibration
* On body detection
* Lid angle calculation

CHRE is a well established framework for managing peripheral data. If integrated well, it could drastically reduce the friction of using these peripherals.

See Issue #36963

# Dependencies

None

## Revision

`b1574433c580e7f7f8c0e8adc39055f63a2d8238`

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``Apache-2.0``

yperess, 2021-08-02:
Ping, any thoughts on this?

microbuilder, 2021-08-18:
@yperess Thanks for the presentation at the TSC 18/Aug/21 ... if you can post a link to the presentation here it would help evaluate this.

0Grit, 2021-08-18:
I like this. It is essentially a standardization and platform abstraction for all the embedded event subscribers and receivers I have needed to write throughout the years. 

1. I look forward to seeing some data regarding codesize and resource usage.
2. I'd like to learn more about the sensorhub concept

yperess, 2021-08-18:
[CHRE-on-Zephyr TSC.pdf](https://github.com/zephyrproject-rtos/zephyr/files/7008626/CHRE-on-Zephyr.TSC.pdf)

microbuilder, 2021-08-18:
You can find this following the link at the top of the issue, but some documentation on CHRE for the curious: https://android.googlesource.com/platform/system/chre/+/HEAD/README.md

yperess, 2021-08-19:
> I like this. It is essentially a standardization and platform abstraction for all the embedded event subscribers and receivers I have needed to write throughout the years.
>
> 1. I look forward to [...]

I spent some time gathering initial statistics:
1. When building the core CHRE we're adding ~1600 lines of code to the build.
2. The core sensor framework (reusable part) we're adding another ~1500 lines of code + whatever code we write to implement the PAL (this is hard to estimate).
3. The core audio framework is ~500 lines + the code to implement the PAL
4. The core GNSS framework is ~750 lines + the code to implement the PAL
5. The core WiFi framework is ~750 lines + the code to implement the PAL
6. The core WWAN framework is ~100 lines + the code to implement the PAL

I think that the key thing to keep in mind is that what we're trying to do is make things simple for the app developer. The developer should be using this if they need the features. In this case (when the feature is required) under the current model, the developer would need to implement those features themselves, this is what we're trying to solve.

As per code size, we (the Zephyr community) would obviously do our best to keep the footprint small. This is the benefit to having this all in a common location, any improvement goes out to everyone.

This would look roughly like:
```
CONFIG_CHRE=y
CONFIG_SENSOR_FRAMEWORK=y
CONFIG_SENSOR_FRAMEWORK_TIMESTAMP_SPREADING=y
CONFIG_SENSOR_FRAMEWORK_<feature_name>=y
```

I would argue that most apps can afford this size cost and the rare instances that can't will likely want to implement everything themselves anyway and use the raw drivers directly. Those developers might benefit from being able to see how Zephyr implements various features.

0Grit, 2021-08-20:
Personally I've got no issue with resource consumption and it's optional regardless.

Most devs would end up recreating it in application specific form otherwise. As mentioned on the call I have had need for something standardized like this for years and think it could provide immense value.

yperess, 2021-08-20:
What's the process on this? Is there a vote to make CHRE an official module?

yperess, 2021-08-31:
Ping, is there anything else that should be discussed? It seems like there are no objections to this work.

yperess, 2021-09-18:
Can we create the fork?

yashi, 2021-09-18:
Hi @yperess, I'm interested in CHRE on Zephyr.   How did the presentation go?  I read the [TSC minutes](https://docs.google.com/document/d/1qOSz9EdSLMuCR8wNdlvq4kCjqoxWyiKktAtLrsV5mUg/edit#heading=h.bgujgrc0i2z5) and see TSC label removed but not sure about the current situation.

I don't see Zephyr in the [CHRE repo](https://android.googlesource.com/platform/system/chre/+/HEAD/platform) yet.  Are you planning to merge zephyr port to the upstream? or

> Can we create the fork?

Are you planning to fork the upstream?

yperess, 2021-09-18:
@yashi I believe that the next step is that an admin from Zephyr needs to fork https://android.googlesource.com/platform/system/chre/+/refs/heads/master into https://github.com/zephyrproject-rtos/chre.

We then create a `zephyr` branch and add the `zephyr/module.yml` file and merge the appropriate changes into Zephyr to build it.

So really we're just waiting on the fork to be able to do any actual work.

yashi, 2021-09-18:
@yperess, Sorry for my ignorant but does the upstream already have working Zephyr port?  I checked the [repo](https://android.googlesource.com/platform/system/chre/+/refs/heads/master ) but did't see it.  I mean, if the upstream already have working Zephyr port, we don't need to fork it under Zephyr project (I know we have many cloned repos under).  We can create a PR with glue code (if needed) and a change to `west.yml`.  I'm not a member of TSC so ...

If there already is working code, can I try it here on my board?

yperess, 2021-09-19:
> @yperess, Sorry for my ignorant but does the upstream already have working Zephyr port?  I checked the [repo](https://android.googlesource.com/platform/system/chre/+/refs/heads/master ) but did't [...]
Yes, I suppose we could simply add it directly as a new remote.
> If there already is working code, can I try it here on my board?
I'll try to put something together over the next 2 days.

yperess, 2021-09-21:
Seems I can't just pull it in, I've tried adding it as a module directly to `west.yaml` and while the sync happens correctly it seems that without a zephyr branch and a `zephyr/module.yaml` file I'm not able to enable the config. See https://github.com/zephyrproject-rtos/nanopb/blob/zephyr/zephyr/module.yml as an example for another module.

So I think we really need an admin to create https://github.com/zephyrproject-rtos/chre/

microbuilder, 2021-09-21:
@galak What's the process to formally create a module repo under `zephyrproject-rtos`?

yashi, 2021-09-22:
@yperess, what's your plan? Do you plan to merge everything Zephyr needs, including `zephyr/module.yaml`, to the CHRE upstream, or you are planning to merge glue code to Zephyr?  Either way, we can create a CHRE repo under your github account and work on it.
- If former, when it's ready, we can ask the CHRE upstream to merge.  For Zephyr, we just need a PR with a `west.yaml` change.
- If latter, create Zephyr repo along with CHRE under your account, work on it both.  And ask both upstream for merge.

yperess, 2021-09-22:
> @yperess, what's your plan? Do you plan to merge everything Zephyr needs, including `zephyr/module.yaml`, to the CHRE upstream, or you are planning to merge glue code to Zephyr?  Either way, we can create a CHRE repo under your github account and work on it.
> - If the former, when it's ready, we can ask the CHRE upstream to merge.  For Zephyr, we just need a PR with a `west.yaml` change.

Yes, this one 😁
I just had a chat with someone on the CHRE team and they're happy to include the module file. I'll be sending a change tonight to AOSP. I'll be sending a corresponding PR to zephyr tomorrow.

yperess, 2022-01-05:
Ping, this seems to have been approved at the TSC meeting way way back. I'm getting some bandwidth to finish it up. Can we please get the fork done? @nashif would you be able to help get the fork created or @ mention someone else that can? I heard from my org that Intel is very interested in using this so it might become a bit of a higher priority.

nashif, 2022-01-12:
done https://github.com/zephyrproject-rtos/chre

closed 2022-01-12 by nashif