zephyrproject-rtos/zephyr #37225: subsys: mgmt: hawkbit

ycsin, 2021-07-27:
**Describe the bug**
Found something in the `hawkbit` that might be improved while I'm debugging why the `hawkbit` sample isn't working for me:
1. The [`hawkbit_init`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L610) function is assuming the "storage" label to be with `DT_CHOSEN_ZEPHYR_FLASH_CONTROLLER_LABEL` (usually `flash0`?) whereas it can  can be with the external serial flash so this should follows where the "storage" is defined instead of a constant.
2. Same goes for the [`nvs_init(DT_CHOSEN_ZEPHYR_FLASH_CONTROLLER_LABEL)`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L620)
3. There's an extra space before the "not" [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L630) that should be removed.
4. For some reason it marked the image as OK [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L638), but the image still isn't OK. Which led to a reboot [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L1235) when `hawkbit_autohandler();` is invoked.
5. It rebooted before the log messages get flushed, so maybe have to delay before [reboot](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L1235) or do something in the reboot function to flush the logs.

**To Reproduce**
Build `hawkbit` sample with "storage" in external flash.

**Expected behavior**
`hawkbit` sample working.

**Impact**
Can't get `hawkbit` sample to work.

**Environment (please complete the following information):**
 - OS: Windows 10
 - gnuarmemb
 - v2.6.0-rc2-107-gebe282b02d9f

**Logs**
```
*** Booting Zephyr OS build v2.6.0-rc2-107-gebe282b02d9f  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
I: Bootloader chainload address offset: 0x18000
I: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build v2.6.0-rc2-107-gebe282b02d9f  ***
[00:00:02.010,000] <inf> main: Hawkbit sample app started
[00:00:04.602,000] <inf> modem_gsm: Waiting for modem to boot up
[00:00:15.308,000] <inf> modem_gsm: Modem RDY
[00:00:16.385,000] <inf> modem_gsm: PPP channel 2 connected to GSM_1
[00:00:16.691,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:16.749,000] <inf> modem_gsm: Model: EC21
[00:00:16.808,000] <inf> modem_gsm: Revision: EC21EFAR06A01M4G
[00:00:16.868,000] <inf> modem_gsm: IMEI: 867962047311785
[00:00:17.052,000] <inf> modem_gsm: Network registered, home network.
[00:00:17.061,000] <inf> modem_gsm: Attached to packet service!
[00:00:17.080,000] <inf> net_ppp: Initializing PPP to use GSM_1
[00:00:17.084,000] <inf> modem_gsm: AT channel 1 connected to GSM_2
[00:00:17.157,000] <inf> main: hawkbit_init
[00:00:18.275,000] <inf> fs_nvs: 3 Sectors of 4096 bytes
[00:00:18.275,000] <inf> fs_nvs: alloc wra: 0, fe8
[00:00:18.275,000] <inf> fs_nvs: data wra: 0, 0
[00:00:18.276,000] <inf> hawkbit: Image is not confirmed OK
[00:00:18.276,000] <inf> hawkbit: Marked image as OK
[00:00:21.953,000] <inf> hawkbit: Erased second slot
[00:00:21.953,000] <inf> main: Starting hawkbit polling mode
[00:00:22.953,000] <err> hawkbit: The current image is not confirmed
[00:00:22.953,000] <err> hawkbit: Image is unconfirmed
[00:00:22.953,000] <err> hawkbit: Rebooting to previous confirmed image.
```

## Suggested fix
1 & 2: Refer to the [`nvs` sample](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/nvs/src/main.c), determine the location of "storage" partition directly from devicetree:
```c
...
#define STORAGE_NODE DT_NODE_BY_FIXED_PARTITION_LABEL(storage)
#define FLASH_NODE DT_MTD_FROM_FIXED_PARTITION(STORAGE_NODE)
...
int hawkbit_init(void)
{
	bool image_ok;
	int ret = 0, rc = 0;
	struct flash_pages_info info;
	int32_t action_id;
	const struct device *flash_dev;

	flash_dev = DEVICE_DT_GET(FLASH_NODE);

	fs.offset = FLASH_AREA_OFFSET(storage);
	rc = flash_get_page_info_by_offs(flash_dev, fs.offset, &info);
	if (rc) {
		LOG_ERR("Unable to get storage page info");
		return -EIO;
	}

	fs.sector_size = info.size;
	fs.sector_count = 3U;

	rc = nvs_init(&fs, flash_dev->name);
...
```

3: Remove the space:
```c
LOG_INF("Image is %s confirmed OK", image_ok ? "" : "not");
```

4. Perform additional check if the image is actually "OK"?

ycsin, 2021-07-27:
@ceolin @gmarull @Navin-Sankar Any clue/comments?

ycsin, 2021-07-27:
Updated, please read again.

Navin-Sankar, 2021-07-27:
@ycsin I agree point 1 & 2. It's good to get the correct flash device from the devicetree storage partition node.
3. remove unwanted space between " not".

cfriedt, 2021-07-27:
@sylvioalves - is this related to the PR you had just made? Just wondering if there is some commonality.

sylvioalves, 2021-07-27:
> @sylvioalves - is this related to the PR you had just made? Just wondering if there is some commonality.

No, it is not related.

ycsin, 2021-07-28:
Updated

cfriedt, 2021-07-28:
@ycsin - would you be open to just making a PR for this?

ycsin, 2021-07-29:
@cfriedt already opened 4 PRs for different issue but @Navin-Sankar requested them to be made into 1 PR instead [here](https://github.com/zephyrproject-rtos/zephyr/pull/37278#issuecomment-888245601). I will do it if its really necessary after I finish my other work.

closed 2021-09-14 by cfriedt