zephyrproject-rtos/zephyr #37228: Bluetooth SMP does not complete pairing

thedjnK, 2021-07-27:
**Describe the bug**
We have an application designed for the nRF5340 SoC which uses Bluetooth, and SMP (mcumgr) which we allow access to over Bluetooth with paired (note: not bonded) devices with a fixed passkey, this is enabled with the authenticated SMP Kconfig enabled. It seems that on the nRF5340 there is an issue with SMP (security manager protocol) whereby a remote device (iPhone) connects, pairs with the device and the connection becomes encrypted, but the pairing success never gets propagated to the application layer and after 30 seconds, the device is kicked off with an unspecified pairing failed message (this would be in the spec whereby a remote device does not accept a pairing, i.e. a timeout). Adding debug to SMP yields:
```
bt_smp_encrypt_change function is called, at the "if (!smp->local_dist && !smp->remote_dist) {" line, smp->local_dist is 1, smp->remote_dist is 3
smp_check_complete function is called, at the "if (!smp->local_dist && !smp->remote_dist) {" line, smp->local_dist is 0, smp->remote_dist is 3
smp_master_identfunction is called, at the "if (!smp->local_dist && !smp->remote_dist) {" line, smp->local_dist is 0, smp->remote_dist is 3
smp_ident_addr_infofunction is called, at the "if (!smp->local_dist && !smp->remote_dist) {" line, smp->local_dist is 0, smp->remote_dist is 1
smp_pairing_complete is called after about 30 seconds, status is 0x8, code flows into the pairing failed, status 0x8 is BT_SECURITY_ERR_UNSPECIFIED
```
This might be nRF5340 specific as we have another build for an nRF52840 which does seem to be working OK.

**To Reproduce**
This might be application setting specific but we have a main project with a hci_rpmsg subproject (this is from nRF connect version 1.6) and they are setup to allow pairing, fixed passkey, advertising extensions, DLE, PHY changes, etc. but not bonding and SMP is configured for authenticated access

**Expected behavior**
Pairing to work

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU Arm Embedded Toolchain 10-2020-q4-major 10.2.1
 - Commit SHA or Version used: The version that nRF connect SDK v1.6 ships with

thedjnK, 2021-07-27:
Tested an nRF52840 and it does not exhibit this issue, the key exchange bits go to 0 and it calls the pairing complete callback.

carlescufi, 2021-08-02:
If it happens only on nRF5340 it is possibly due to the `hci_rpmsg` configuration.

joerchan, 2021-08-02:
@thedjnK Good job including the relevant information.

The issue happens when doing legacy pairing without bonding. The reason why it works for nrf52840 is that it would be using secure connections pairing. I recommend doing the same for nrf5340. You need to enable CONFIG_BT_TINYCRYPT_ECC=y, it appears a default is wrong for nrf5340.

PR #37370 (2021-08-03): Bluetooth: host: Fix key distribution not calling pairing complete
Fix key distribution not calling pairing complete callback when
distributing encryption information when not bonding.
Encryption information is only distributed in legacy pairing so issue
was only seen during legacy pairing.

Fixes: #37228
closed 2021-08-03 by carlescufi