zephyrproject-rtos/zephyr #37236: ESP32 will not start when CONFIG_ASSERT=y is enabled

maweicoltsmart, 2021-07-27:
board： ESP32 
branch：main
add this line CONFIG_ASSERT=y to esp32_deconfig

west build -b esp32 samples\boards\esp32\wifi_station

download fireware and run, The program will have the following error。
`
I (12) boot: ESP-IDF c21d252a21 2nd stage bootloader
I (13) boot: compile time 19:27:25
I (13) boot: chip revision: 1
I (15) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (22) boot.esp32: SPI Speed      : 40MHz
I (26) boot.esp32: SPI Mode       : DIO
I (31) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00002000 00006000
I (59) boot:  1 phy_init         RF data          01 01 00008000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00100000
I (74) boot: End of partition table
I (78) boot_comm: chip revision: 1, min. application chip revision: 0
I (85) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0bd60h ( 48480) map
I (112) esp_image: segment 1: paddr=0001bd88 vaddr=3ffb0000 size=01f84h (  8068) load
I (116) esp_image: segment 2: paddr=0001dd14 vaddr=3ffb1f84 size=00180h (   384) load
I (118) esp_image: segment 3: paddr=0001de9c vaddr=40080000 size=00400h (  1024) load
I (127) esp_image: segment 4: paddr=0001e2a4 vaddr=40080400 size=01d74h (  7540) load
I (138) esp_image: segment 5: paddr=00020020 vaddr=400d0020 size=3d880h (252032) map
I (239) esp_image: segment 6: paddr=0005d8a8 vaddr=40082174 size=146b8h ( 83640) load
I (286) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
I (612) wifi:wifi driver task: 3ffb2270, prio:2, stack:3584, core=0
I (613) wifi:wifi firmware version: 7e2c914
I (613) wifi:wifi certification version: v7.0
I (615) wifi:config NVS flash: disabled
I (618) wifi:config nano formating: disabled
I (623) wifi:Init data frame dynamic rx buffer num: 32
I (627) wifi:Init management frame dynamic rx buffer num: 32
I (633) wifi:Init management short buffer num: 32
I (637) wifi:Init dynamic tx buffer num: 32
I (641) wifi:Init static rx buffer size: 1600
I (645) wifi:Init static rx buffer num: 10
I (649) wifi:Init dynamic rx buffer num: 32
I (654) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (786) wifi:mode : softAP (ac:67:b2:c6:6e:15)
I (787) wifi:Total power save buffer number: 16
I (787) wifi:Init max length of beacon: 752/752
I (788) wifi:Init max length of beacon: 752/752
ASSERTION FAIL [arch_irq_unlocked(key) || _kernel.cpus[0].current->base.thread_state & (((1UL << (0))) | ((1UL << (3))))] @ ../kernel/include/kswap.h:85
        Context switching while holding lock!
@ WEST_TOPDIR/zephyr/lib/os/assert.c:45
[00:00:00.812,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.812,000] <err> os: Current thread: 0x3ffb2270 (unknown)
[00:00:00.827,000] <err> os: Halting system
`

uLipe, 2021-07-27:
@maweicoltsmart , thanks for reporting that, we already found the cause and a fix for this, we will push it soon.

uLipe, 2021-07-28:
This PR: https://github.com/zephyrproject-rtos/hal_espressif/pull/43

Added to espressif_hal will fix the issue, once Zephyr gets updated to use up to date hal, the fix will be available.

@maweicoltsmart

maweicoltsmart, 2021-07-29:
Updated the code of hal, the problem is solved, thank you!

closed 2021-07-29 by maweicoltsmart