zephyrproject-rtos/zephyr #37239: Initial Xen public headers adding

firscity, 2021-07-27:
## Origin

Xen Project
Link: [Xen git](https://xenbits.xen.org/gitweb/?p=xen.git;a=summary)
Branch: RELEASE-4.15.0

## Purpose

These headers describe interface to Xen hypercalls, structures, defines etc. They are needed for Xen-related features development. 

## Mode of integration

I would like to place headers to main include directory (`include/xen/public`), they will be used by Xen drivers (PV console, event channels, Xenbus etc.).

## Pull Request

Link to PR: https://github.com/zephyrproject-rtos/zephyr/pull/36691

## Description

As I mentioned, these headers are needed for adding Xen features support to Zephyr project (they provide generic interface for drivers).
In future, during Xen drivers development, the list of required Xen headers may be increased.

# Dependencies

Since it is just public headers, that will be included to Zephyr source tree, no direct dependencies to external component will be created.

## Revision

Modified headers from Xen RELEASE-4.15.0, present in mentioned PR.

## License

``MIT``

carlescufi, 2021-08-18:
The TSC has voted to have the headers in-tree, as per the proposal in #36691, and as opposed to having a separate repository containing them.
I will now take the matter to the board.

firscity, 2021-09-02:
Dear @carlescufi, as I can find in contributing guideline, governing board has two weeks to review matter. Can you please tell the current status or the way how I can check it by myself?

nashif, 2021-09-08:
@firscity  This will have to wait for main branch to reopen in 2 weeks, it will not be part of 2.7.

firscity, 2021-09-08:
Dear @nashif, okay, thank you for updates. Please tag me here or in PR when time will come - I will rebase my changes.