zephyrproject-rtos/zephyr #37241: Failed to compile http_server under ARM Cortex m0 architecture

maweicoltsmart, 2021-07-27:
branch：main
soc arch：arm cortex m0

Compiling samples\net\civetweb\http_server under an arm cortex m0 MCU shows the following error:

```
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `mg_atomic_inc':
D:/ZephyrProject/code/zephyrproject/modules/lib/civetweb/src/civetweb.c:1228: undefined reference to `__sync_add_and_fetch_4'
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/ZephyrProject/code/zephyrproject/modules/lib/civetweb/src/civetweb.c:1228: undefined reference to `__sync_add_and_fetch_4'
collect2.exe: error: ld returned 1 exit status
```

cfriedt, 2021-07-27:
@maweicoltsmart - can you please provide more reproducibility information; BOARD, etc

maweicoltsmart, 2021-07-28:
> @maweicoltsmart - can you please provide more reproducibility information; BOARD, etc

I discovered this problem when using an MCU based on the ARM Cortex m0 architecture that was not in this project. Preliminary guess is whether MCUs based on ARM Cortex m0 have this problem, so I did an experiment with board nucleo_f030r8 and found that it also has this problem.

Because it is just to verify whether MCUs based on ARM Cortex m0 have this problem, I made the following changes to board nucleo_f030r8:

```
diff --git a/boards/arm/nucleo_f030r8/Kconfig.defconfig b/boards/arm/nucleo_f030r8/Kconfig.defconfig
index 7b5331693f..fe5cdcb10e 100644
--- a/boards/arm/nucleo_f030r8/Kconfig.defconfig
+++ b/boards/arm/nucleo_f030r8/Kconfig.defconfig
@@ -8,4 +8,18 @@ if BOARD_NUCLEO_F030R8
 config BOARD
        default "nucleo_f030r8"

+# ENC28J60 is L2 chip slave on SPI
+config SPI
+       default y
+
+config NET_L2_ETHERNET
+       default y
+
+# ENC28J60 Ethernet Device
+config ETH_ENC28J60
+       default y
+
+config ETH_ENC28J60_0
+       default y
+
 endif # BOARD_NUCLEO_F030R8
```

```
diff --git a/boards/arm/nucleo_f030r8/nucleo_f030r8.dts b/boards/arm/nucleo_f030r8/nucleo_f030r8.dts
index d6520ecd70..0f76a95079 100644
--- a/boards/arm/nucleo_f030r8/nucleo_f030r8.dts
+++ b/boards/arm/nucleo_f030r8/nucleo_f030r8.dts
@@ -106,3 +106,20 @@
 &iwdg {
        status = "okay";
 };
+
+mikrobus_spi: &spi1 {};
+
+
+&mikrobus_spi {
+       status = "okay";
+       /* LPCXpresso55xxx boards all use SSEL1. */
+       eth_click: eth_click@0 {
+               compatible = "microchip,enc28j60";
+               reg = <0x0>;
+               local-mac-address = [00 00 00 01 02 03];
+               /* Errata B7/1 specifies min 8Mhz, 20MHz max according to RM */
+               spi-max-frequency = <10000000>;
+               int-gpios = <&gpiob 7 GPIO_ACTIVE_LOW>; /* INT */
+               label = "ETH_CLICK";
+       };
+};
```

```
diff --git a/boards/arm/nucleo_f030r8/nucleo_f030r8_defconfig b/boards/arm/nucleo_f030r8/nucleo_f030r8_defconfig
index 8be75cf6ab..98138c0e43 100644
--- a/boards/arm/nucleo_f030r8/nucleo_f030r8_defconfig
+++ b/boards/arm/nucleo_f030r8/nucleo_f030r8_defconfig
@@ -28,3 +28,6 @@ CONFIG_GPIO=y

 # Enable Clocks
 CONFIG_CLOCK_CONTROL=y
+CONFIG_NETWORKING=y
[... 16 lines cut ...]
        };

        soc {
                flash-controller@40022000 {
                        flash0: flash@8000000 {
-                               reg = <0x08000000 DT_SIZE_K(64)>;
+                               reg = <0x08000000 DT_SIZE_K(640)>;
                        };
                };
```

Next we compile this program. 
`west build -b nucleo_f030r8 samples\net\civetweb\http_server`
Then the following compilation error appears.
```
[143/150] Linking C executable zephyr\zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
cmd.exe /C "cd . && C:\gnu_arm_embedded\bin\arm-none-eabi-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=D:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/config/libsubsys__net__lib__config.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/serial/libdrivers__serial.a  modules/civetweb/lib..__modules__lib__civetweb.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v6-m/nofp"  -LD:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m0  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cmd.exe /C "cd /D D:\ZephyrProject\code\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: modules/civetweb/lib..__modules__lib__civetweb.a(civetweb.c.obj): in function `mg_atomic_inc':
D:/ZephyrProject/code/zephyrproject/modules/lib/civetweb/src/civetweb.c:1228: undefined reference to `__sync_add_and_fetch_4'
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: D:/ZephyrProject/code/zephyrproject/modules/lib/civetweb/src/civetweb.c:1228: undefined reference to `__sync_add_and_fetch_4'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\ZephyrProject\code\zephyrproject\zephyr\build'
```

cfriedt, 2021-07-28:
@maweicoltsmart - what version of the gnu arm embedded toolchain are you using?
```
arm-none-eabi-gcc -v
```

Does this work for you?

```
```

cfriedt, 2021-07-29:
@galak, is `__sync_add_and_fetch_4()` something that zephyr or the toolchain or sdk should provide? 

According to SO, it is not.
https://stackoverflow.com/a/58394755

In that case, I feel like it might just be a bug in [civetweb](https://github.com/zephyrproject-rtos/civetweb).

cfriedt, 2021-07-29:
The undefined symbol error is no longer present with the fix provided in zephyrproject-rtos/civetweb#8

```
Users/cfriedt/zephyr-sdk-0.13.0-rc3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_prebuilt.elf section `datas' will not fit in region `SRAM'
/Users/cfriedt/zephyr-sdk-0.13.0-rc3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: section .intList VMA [0000000020004000,0000000020004057] overlaps section datas VMA [0000000020000000,000000002000438b]
/Users/cfriedt/zephyr-sdk-0.13.0-rc3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `SRAM' overflowed by 60343 bytes
collect2: error: ld returned 1 exit status
```

cfriedt, 2021-09-18:
tested again on main after tagging v2.7.0-rc2

closed 2021-12-03 by github-actions[bot]