zephyrproject-rtos/zephyr #37253: west flash is failed with openocd  for ST-Nucleo-L476RG

digidhamu, 2021-07-27:
**Describe the bug**
1west flash1 is failed with openocd for ST-Nucleo-L476RG

**To Reproduce**
1. `west build -p auto -b nucleo_l476rg samples/basic/blinky`
2. `west flash`
3. See error

**Expected behaviour**
Blinky program should be flashed and led should should blink

**Impact**
Not allowed to flash and test a starter project to get started

**Logs and console output**
```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
Traceback (most recent call last):
  File "/Users/dhamukrish/.virtualenvs/zephyenv/bin/west", line 8, in <module>
    sys.exit(main())
  File "/Users/dhamukrish/.virtualenvs/zephyenv/lib/python3.9/site-packages/west/app/main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "/Users/dhamukrish/.virtualenvs/zephyenv/lib/python3.9/site-packages/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/Users/dhamukrish/.virtualenvs/zephyenv/lib/python3.9/site-packages/west/app/main.py", line 338, in run_command
    self.run_extension(args.command, argv)
  File "/Users/dhamukrish/.virtualenvs/zephyenv/lib/python3.9/site-packages/west/app/main.py", line 408, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/Users/dhamukrish/.virtualenvs/zephyenv/lib/python3.9/site-packages/west/commands.py", line 132, in run
    self.do_run(args, unknown)
  File "/Users/dhamukrish/Documents/digidhamu/k8s.do.digidhamu.com/zephyr/zephyrproject/zephyr/scripts/west_commands/flash.py", line 29, in do_run
    do_run_common(self, my_args, runner_args)
  File "/Users/dhamukrish/Documents/digidhamu/k8s.do.digidhamu.com/zephyr/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 228, in do_run_common
    runner.run(command_name)
  File "/Users/dhamukrish/Documents/digidhamu/k8s.do.digidhamu.com/zephyr/zephyrproject/zephyr/scripts/west_commands/runners/core.py", line 509, in run
    self.do_run(command, **kwargs)
  File "/Users/dhamukrish/Documents/digidhamu/k8s.do.digidhamu.com/zephyr/zephyrproject/zephyr/scripts/west_commands/runners/openocd.py", line 114, in do_run
    self.require(self.openocd_cmd[0])
  File "/Users/dhamukrish/Documents/digidhamu/k8s.do.digidhamu.com/zephyr/zephyrproject/zephyr/scripts/west_commands/runners/core.py", line 546, in require
    ret = shutil.which(program)
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/shutil.py", line 1390, in which
    if os.path.dirname(cmd):
  File "/usr/local/Cellar/python@3.9/3.9.6/Frameworks/Python.framework/Versions/3.9/lib/python3.9/posixpath.py", line 152, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

**Environment**
 - OS: macOS 10.15.7
 - Toolchain gnuarmemb
 - Version: zephyr-v2.6.0

FRASTM, 2021-07-28:
did you update the west recently ?

erwango, 2021-07-28:
@digidhamu I can't provide much hints on Zephyr Mac OS installs, but I'd say you might face an issue in openocd installation.
Is it correctly installed ?

digidhamu, 2021-07-28:
> @digidhamu I can't provide much hints on Zephyr Mac OS installs, but I'd say you might face an issue in openocd installation.
> Is it correctly installed ?

> did you update the west recently ?
> My West version: v0.11.0 can successfully flash the nucleo_l476rg
> 
> $ west flash
> 
> ```
> -- west flash: rebuilding
> ninja: no work to do.
> -- west flash: using runner openocd
> -- runners.openocd: Flashing file: ./zephyr/build/zephyr/zephyr.hex
> Open On-Chip Debugger 0.10.0+dev-01508-gf79c90268-dirty (2021-03-26-16:13)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : clock speed 500 kHz
> Info : STLINK V2J33M25 (API v2) VID:PID 0483:374B
> Info : Target voltage: 3.250952
> Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : starting gdb server for stm32l4x.cpu on 3333
> Info : Listening on port 3333 for gdb connections
>     TargetName         Type       Endian TapName            State       
> --  ------------------ ---------- ------ ------------------ ------------
>  0* stm32l4x.cpu       hla_target little stm32l4x.cpu       running
> 
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08003f9c msp: 0x20002000
> Info : device idcode = 0x10076415 (STM32L47/L48xx - Rev: 4)
> Info : flash size = 1024kbytes
> Info : flash mode : dual-bank
> Info : Padding image section 0 at 0x080037f4 with 4 bytes (bank write end alignment)
> Warn : Adding extra erase range, 0x080037f8 .. 0x080037ff
> auto erase enabled
> wrote 14328 bytes from file ./build/zephyr/zephyr.hex in 1.031002s (13.571 KiB/s)
> 
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x08001428 msp: 0x200006e0
> verified 14324 bytes in 0.583010s (23.993 KiB/s)
> 
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> Info : Unable to match requested speed 500 kHz, using 480 kHz
> shutdown command invoked
> ```

@FRASTM 
Yes. It is latest version of West v0.11.0

erwango, 2021-07-28:
@digidhamu Ping on the openocd installation question (that you have to do manually)

digidhamu, 2021-07-28:
> @digidhamu Ping on the openocd installation question (that you have to do manually)

I did manually as below. Is there any config we need to do?
```
Warning: open-ocd 0.11.0 is already installed and up-to-date.
To reinstall 0.11.0, run:
  brew reinstall open-ocd
```

erwango, 2021-07-28:
> > @digidhamu Ping on the openocd installation question (that you have to do manually)
> 
> I did manually as below. Is there any config we need to do?
> 
> ```
> Warning: open-ocd 0.11.0 is already installed and up-to-date.
> To reinstall 0.11.0, run:
>   brew reinstall open-ocd
> ```

ok. that's a good start.
I'm not exactly sure. Maybe @galak could help here ?

digidhamu, 2021-07-28:
I could able to fix the issue and followed below steps

1. `brew uninstall openocd`
2. `rm -rf zypher/build`
3. `brew install openocd`
4. `west build -p auto -b nucleo_l476rg samples/basic/blinky`
5. `west flash`
6. `west debug`

So, I am closing this issue. Thanks for your  great support here.

closed 2021-07-28 by digidhamu