zephyrproject-rtos/zephyr #37264: tests-ci : can: isotp: implemmentation test failed

hakehuang, 2021-07-28:
**Describe the bug**
implemmentation test is failed on zephyr-v2.6.0-1424-g14f09a3b00d4 on mimxrt1170_evk_cm7

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1170_evk_cm7  --sub-test can.isotp
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
W: Bitrate error: 24
*** Booting Zephyr OS build zephyr-v2.6.0-1424-g14f09a3b00d4  ***
Running test suite isotp
===================================================================
START - test_bind_unbind
PASS - test_bind_unbind in 0.793 seconds
===================================================================
START - test_send_receive_net_sf
PASS - test_send_receive_net_sf in 0.2 seconds
===================================================================
START - test_send_receive_net_blocks
E: Sequence number missmatch
W: Unhandled error/status (status 0x00000001, result = 0x00000000
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:149: receive_test_data_net: (remaining_len >= 0 is false)
recv error: -4
FAIL - test_send_receive_net_blocks in 0.25 seconds
===================================================================
START - test_send_receive_net_single_blocks
ASSERTION FAIL [!sys_dnode_is_linked(&to->node)] @ WEST_TOPDIR/zephyr/kernel/timeout.c:93
E: r0/a1:  0x00000004  r1/a2:  0x0000005d  r2/a3:  0x00800000
E: r3/a4:  0x30004a3d r12/ip:  0xa0000000 r14/lr:  0x3000c72b
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x3000d846
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80000580 (sysworkq)
E: Halting system

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-1424-g14f09a3b00d4

hakehuang, 2021-07-28:
@dleach02 , @MaureenHelm

CristXu, 2021-08-12:
https://github.com/zephyrproject-rtos/zephyr/pull/37627

PR #37627 (2021-08-12): arm: mcux (RT1170 EVK) fix isotp test failure
Config the FlexCAN3 with a maximum 80MHz root clock to fix

Fixes #37264
closed 2021-08-12 by MaureenHelm