zephyrproject-rtos/zephyr #37267: tests-ci : kernel: threads: apis test Timeout

hakehuang, 2021-07-28:
**Describe the bug**
apis test is Timeout on zephyr-v2.6.0-1424-g14f09a3b00d4 on lpcxpresso55s28

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso55s28  --sub-test kernel.threads
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-1424-g14f09a3b00d4  ***
Running test suite threads_lifecycle
===================================================================
START - test_thread_runtime_stats_get
PASS - test_thread_runtime_stats_get in 0.1 seconds
===================================================================
START - test_k_thread_stack_space_get_user
PASS - test_k_thread_stack_space_get_user in 0.1 seconds
===================================================================
START - test_threads_spawn_params
PASS - test_threads_spawn_params in 0.101 seconds
===================================================================
START - test_threads_spawn_priority
PASS - test_threads_spawn_priority in 0.101 seconds
===================================================================
START - test_threads_spawn_delay
PASS - test_threads_spawn_delay in 0.201 seconds
===================================================================
START - test_threads_spawn_forever
PASS - test_threads_spawn_forever in 0.1 seconds
===================================================================
START - test_thread_start_user
PASS - test_thread_start_user in 0.101 seconds
===================================================================
START - test_thread_start
PASS - test_thread_start in 0.1 seconds
===================================================================
START - test_threads_suspend_resume_cooperative
PASS - test_threads_suspend_resume_cooperative in 0.201 seconds
===================================================================
START - test_threads_suspend_resume_preemptible
PASS - test_threads_suspend_resume_preemptible in 0.201 seconds
===================================================================
START - test_threads_priority_set
PASS - test_threads_priority_set in 0.1 seconds
===================================================================
START - test_threads_abort_self
PASS - test_threads_abort_self in 0.101 seconds
===================================================================
START - test_threads_abort_others
PASS - test_threads_abort_others in 1.152 seconds
===================================================================
START - test_threads_abort_repeat
PASS - test_threads_abort_repeat in 0.201 seconds
===================================================================
START - test_delayed_thread_abort
PASS - test_delayed_thread_abort in 0.51 seconds
===================================================================
START - test_essential_thread_operation
PASS - test_essential_thread_operation in 0.1 seconds
===================================================================
START - test_essential_thread_abort
PASS - test_essential_thread_abort in 0.1 seconds
===================================================================
START - test_systhreads_main
PASS - test_systhreads_main in 0.1 seconds
===================================================================
START - test_systhreads_idle
PASS - test_systhreads_idle in 0.101 seconds
===================================================================
START - test_customdata_get_set_coop
PASS - test_customdata_get_set_coop in 0.501 seconds
===================================================================
START - test_customdata_get_set_preempt
PASS - test_customdata_get_set_preempt in 0.501 seconds
===================================================================
START - test_k_thread_foreach
TEST: Newly added thread found
TEST: tid: 0x30000720, prio: 0
PASS - test_k_thread_foreach in 0.7 seconds
===================================================================
START - test_k_thread_foreach_unlocked
TEST_UNLOCKED: Newly added thread found
TEST_UNLOCKED: tid: 0x30000720, prio: 0
TEST_UNLOCKED: Newly added thread in callback found
TEST_UNLOCKED: tid: 0x30000ac0, prio: 0
TEST_UNLOCKED: Newly added thread found
TEST_UNLOCKED: tid: 0x30000720, prio: 0
PASS - test_k_thread_foreach_unlocked in 0.24 seconds
===================================================================
START - test_k_thread_foreach_null_cb
ASSERTION FAIL [user_cb != ((void *)0)] @ WEST_TOPDIR/zephyr/kernel/thread.c:55
user_cb can not be NULL
E: r0/a1:  0x00000004  r1/a2:  0x00000037  r2/a3:  0x80000000
E: r3/a4:  0x00000000 r12/ip:  0xa0000000 r14/lr:  0x10007a1b
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x1000e7ea
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x30000c90 (test_k_thread_foreach_null_cb)
PASS - test_k_thread_foreach_null_cb in 0.37 seconds
===================================================================
START - test_k_thread_foreach_unlocked_null_cb
ASSERTION FAIL [user_cb != ((void *)0)] @ WEST_TOPDIR/zephyr/kernel/thread.c:83
user_cb can not be NULL
E: r0/a1:  0x00000004  r1/a2:  0x00000053  r2/a3:  0x80000000
E: r3/a4:  0x00000000 r12/ip:  0xa0000000 r14/lr:  0x10007ae5
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x1000e7ea
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x30000c90 (test_k_thread_foreach_unlocked_)
PASS - test_k_thread_foreach_unlocked_null_cb in 0.38 seconds
===================================================================
START - test_k_thread_state_str
PASS - test_k_thread_state_str in 0.1 seconds
===================================================================
START - test_thread_name_get_set
PASS - test_thread_name_get_set in 0.1 seconds
===================================================================
START - test_thread_name_user_get_set
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-1424-g14f09a3b00d4

hakehuang, 2021-07-28:
@dleach02 @MaureenHelm

dleach02, 2021-08-13:
@ioannisg I've debugged this down to an issue with trying to retrieve the ESF in the fault handler. The lpc55s28 has TRUSTED_EXECUTION_SECURE enabled by default. This appears to change the behavior here where the exc_return value passed to get_esf() has a value of 0xFFFFFFFD where it causes the code flow to go through:

```
	if ((exc_return & EXC_RETURN_EXCEPTION_SECURE_Secure) == 0U) {
		/* Secure Firmware shall only handle Secure Exceptions.
		 * This is a fatal error.
		 */
		return NULL;
	}

```
The NULL value causes an assert. I'm not sure what the general strategy should be for the platform where there isn't TFM but, by default, we set CONFIG_TRUSTED_EXECUTION_SECURE for the board? Any suggestion or guidance here?

PR #37725 (2021-08-17): boards: arm: lpc55s28: Remove Trustzone-M flags
The LPC55S28 does not have the Trustzone-M support.

Fixes #36768
Fixes #37266
Fixes #37267
commit f567e029 (2021-08-17): boards: arm: lpc55s28: Remove Trustzone-M flags

The LPC55S28 does not have the Trustzone-M support.

Fixes #36768
Fixes #37266
Fixes #37267
closed 2021-08-17 by carlescufi