zephyrproject-rtos/zephyr #37293: Native POSIX MAC addresses are not random and are duplicate between multiple instances

bob2oneil, 2021-07-28:
**Describe the bug**
I am building a network of multiple instances of Zephyr, each instance represents a simulated radio with a wired and wireless interface.

I am using the following kernel configurations to generate 2 virtual interfaces and I want truly random MAC addresses, or random enough so that there are no collisions between multiple running instances on a single workstation. 

```
# Random MAC is required for more than 1 POSIX interfaces
CONFIG_ETH_NATIVE_POSIX_RANDOM_MAC=y

# POSIX two interfaces, wired and wireless
CONFIG_ETH_NATIVE_POSIX_INTERFACE_COUNT=2
```

But for multiple instances, the MAC addresses will match between instances, so I get the same MAC for the zeth0 interface as 02:00:5e:00:53:3b and for zeth1 as 02:00:5e:00:53:cc.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b native_posix_64 -- -DOVERLAY_CONFIG=prj[x].conf*
2. The custom kernel overlay file prj[x].conf simply contains a different prefix for the POSIX drive name (e.g. CONFIG_ETH_NATIVE_POSIX_DRV_NAME="zeth1.").
3. Run applications, and observe the MAC addresses assigned to the zeth[x].0 and zeth[x].1 interfaces. They will match in my experience.
4. See error as my application logs below.

**Expected behavior**
Unique MAC addresses for each virtual interface.

**Impact**
Showstopper relative to networking between multiple nodes using Linux bridging between devices.

**Logs and console output**
```
exec: Wired interface 'zeth1.0', MTU 1600, MAC Address : 02:00:5e:00:53:3b
Raw socket on wired interface 'zeth1.0; is open with socket 0
HSS UDP socket on wired interface 'zeth1.0; is open with socket 1
LSS UDP socket on wired interface 'zeth1.0; is open with socket 2
ifaceAddIpv4Addr: Adding IPv4 Address to Ethernet interface 0x5593cc81d4a0 as 138.0.0.1
ifaceSetNetmask: Setting Ethernet interface 0x5593cc81d4a0 netmask as 255.255.255.0
exec: Wireless interface 'zeth1.1', MTU 9014, MAC Address : 02:00:5e:00:53:cc

exec: Wired interface 'zeth2.0', MTU 1600, MAC Address : 02:00:5e:00:53:3b
Raw socket on wired interface 'zeth2.0; is open with socket 0
HSS UDP socket on wired interface 'zeth2.0; is open with socket 1
LSS UDP socket on wired interface 'zeth2.0; is open with socket 2
ifaceAddIpv4Addr: Adding IPv4 Address to Ethernet interface 0x558ab5cc94a0 as 138.0.0.2
ifaceSetNetmask: Setting Ethernet interface 0x558ab5cc94a0 netmask as 255.255.255.0
exec: Wireless interface 'zeth2.1', MTU 9014, MAC Address : 02:00:5e:00:53:cc
Raw socket on wireless interface 'zeth2.1; is open with socket 3

exec: Wired interface 'zeth3.0', MTU 1600, MAC Address : 02:00:5e:00:53:3b
Raw socket on wired interface 'zeth3.0; is open with socket 0
HSS UDP socket on wired interface 'zeth3.0; is open with socket 1
LSS UDP socket on wired interface 'zeth3.0; is open with socket 2
ifaceAddIpv4Addr: Adding IPv4 Address to Ethernet interface 0x555fb35394a0 as 138.0.0.3
ifaceSetNetmask: Setting Ethernet interface 0x555fb35394a0 netmask as 255.255.255.0
exec: Wireless interface 'zeth3.1', MTU 9014, MAC Address : 02:00:5e:00:53:cc
Raw socket on wireless interface 'zeth3.1; is open with socket 3
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Zephyr SDK Build Version zephyr-v2.6.0-1448-g4566d0ec580c)

**Additional context**

From source file eth_native_posix.c, it would seem the gen_random_mac() routine
generates the same MAC addresses for multiple instances of the zephyr.exe binary.

```
#if IS_ENABLED(CONFIG_ETH_NATIVE_POSIX_RANDOM_MAC)
	/* 00-00-5E-00-53-xx Documentation RFC 7042 */
	gen_random_mac(ctx->mac_addr, 0x00, 0x00, 0x5E);

	ctx->mac_addr[3] = 0x00;
	ctx->mac_addr[4] = 0x53;
```

bob2oneil, 2021-07-29:
A workaround of Native POSIX is to simply make modifications to the MAC address itself runtime.  Changes made show up under "net iface".  More testing is required to verify that the application changes do what is necessary to reassign the MAC attributes.

nashif, 2021-08-03:
@aescolar  FYI

aescolar, 2021-08-04:
Short answer:
call native_posix as `zephyr.exe -seed=$RANDOM`

Long answer: native_posix is meant by design to produce by default the exact same results each time you execute it. This is due to its nature as a test/debug platform. And that includes the random generator being seeded with the same number by default. If you want it to not produce the same result you need to seed it with something else.

This is a design choice, not a bug.

bob2oneil, 2021-08-04:
Thanks for the explanation, that makes sense.  Glad to see there is a command line argument for doing this.

closed 2021-08-04 by bob2oneil