zephyrproject-rtos/zephyr #37294: RTT logs not found with default west debug invocation on jlink runner

trueb2, 2021-07-29:
**Describe the bug**
When the large structures are statically defined, the  RTT logs are not found with a standard west debug invocation while using the jlink runner.

They are found if I manually run additional west commands `monitor exec SetRTTAddr 0x00000000200ABCDE` (or automate the jlink runner to run the command).

**To Reproduce**
Steps to reproduce the behavior:
1. Build an application with a very large statically allocated buffer
     1. CONFIG_USE_SEGGER_RTT=y
     2. CONFIG_RTT_CONSOLE=y
     3. nrf52840dk_nrf52840
2. Run `west debug` in one terminal
3. Run `JlinkRTTClient` in another terminal
4. `mon reset 0` and `continue` in the GDB session
5. Observe no logs in RTT client
6. `monitor exec SetRTTAddr 0x00000000200ABCDE`, `mon reset 0`, and `continue` in the GDB session
7. Observe logs in RTT client

Confirmed against `blinky` with settings and large heap as follows

```
CONFIG_GPIO=y
CONFIG_USE_SEGGER_RTT=y
CONFIG_RTT_CONSOLE=y
CONFIG_UART_CONSOLE=n
CONFIG_SERIAL=n
```

```
..

K_HEAP_DEFINE(payload_heap, 220000);

void main(void)
{

...

	while (1) {
		void*  buf = k_heap_alloc(&payload_heap, 1024, K_NO_WAIT);
		gpio_pin_set(dev, PIN, (int)led_is_on);
		led_is_on = !led_is_on;
		k_msleep(SLEEP_TIME_MS);
		k_heap_free(&payload_heap, buf);
	}
}
```

Built with
`west build -b nrf52840dk_nrf52840 zephyr/samples/basic/blinky`

**Expected behavior**
Logs always show for default RTT settings without manual intervention or with an additional flag on runner.

**Impact**
An annoyance for an experienced developer, a show stopper for an inexperienced developer.

**Logs and console output**

Logs after applying patch to apply SetRTTAddr
```
trueb2@jbmm firmware % west --verbose debug
ZEPHYR_BASE=/workspace/firmware/zephyr (origin: configfile)
-- west debug: rebuilding
cmake version 3.19.7 is OK; minimum version is 3.13.1
Running CMake: /opt/homebrew/bin/cmake --build /workspac/workspace/firmware/build
ninja: no work to do.
-- west debug: using runner jlink
runners.jlink: JLINKARM_GetDLLVersion()=75001
-- runners.jlink: JLink version: 7.50a
[... 143 lines cut ...]
**Additional context**
This change fixes the problem by locating the _SEGGER_RTT symbol address in the default build directory's zephyr.map. 

It is a change to the jlink runner under the debug case.

```
trueb2@jbmm zephyr % git diff scripts                                                                                                            (git)-[tags/zephyr-v2.6.0] 
diff --git a/scripts/west_commands/runners/jlink.py b/scripts/west_commands/runners/jlink.py
index 0a473b9f73..570f795231 100644
--- a/scripts/west_commands/runners/jlink.py
+++ b/scripts/west_commands/runners/jlink.py
@@ -229,6 +229,11 @@ class JLinkBinaryRunner(ZephyrBinaryRunner):
                                '-ex', 'load']
                 if self.reset_after_load:
                     client_cmd += ['-ex', 'monitor reset']
+                os.system("rg _SEGGER_RTT build/zephyr/zephyr.map | rg 0x00000000 | awk '{ print $1 }' > build/SetRTTAddr.txt")
+                f = os.open('build/SetRTTAddr.txt', os.O_RDONLY)
+                addr = os.read(f, len('0x0000000020000000')).decode('utf-8')
+                client_cmd += ['-ex', f'monitor exec SetRTTAddr {addr}']
+                os.close(f)
             if not self.gdb_host:
                 self.require(self.gdbserver)
                 self.print_gdbserver_message()
trueb2@jbmm zephyr %                                                                                                                             (git)-[tags/zephyr-v2.6.0] 
```

carlescufi, 2021-09-22:
@trueb2 thanks for the extensive report, but I cannot reproduce this locally on an nRF2840 DK and J-Link 7.543:

```
diff --git a/samples/basic/blinky/prj.conf b/samples/basic/blinky/prj.conf
index 91c3c15b37..e7aa2a09d4 100644
--- a/samples/basic/blinky/prj.conf
+++ b/samples/basic/blinky/prj.conf
@@ -1 +1,5 @@
 CONFIG_GPIO=y
+CONFIG_USE_SEGGER_RTT=y
+CONFIG_RTT_CONSOLE=y
+CONFIG_UART_CONSOLE=n
+CONFIG_SERIAL=n
diff --git a/samples/basic/blinky/src/main.c b/samples/basic/blinky/src/main.c
index e2fbfd3dc5..1874c2aeed 100644
--- a/samples/basic/blinky/src/main.c
+++ b/samples/basic/blinky/src/main.c
@@ -27,6 +27,8 @@
 #define FLAGS  0
 #endif

+K_HEAP_DEFINE(payload_heap, 220000);
+
 void main(void)
 {
        const struct device *dev;
@@ -44,8 +46,10 @@ void main(void)
        }

        while (1) {
+               void*  buf = k_heap_alloc(&payload_heap, 1024, K_NO_WAIT);
                gpio_pin_set(dev, PIN, (int)led_is_on);
                led_is_on = !led_is_on;
                k_msleep(SLEEP_TIME_MS);
+               k_heap_free(&payload_heap, buf);
        }
 }
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..3453f62ca3 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,3 @@
-# nothing here
+CONFIG_USE_SEGGER_RTT=y
+CONFIG_RTT_CONSOLE=y
+CONFIG_UART_CONSOLE=n
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 6c5c8a27dc..f3df1cfd94 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -4,10 +4,15 @@
  * SPDX-License-Identifier: Apache-2.0
  */

+
 #include <zephyr.h>
 #include <sys/printk.h>

+K_HEAP_DEFINE(payload_heap, 220000);
+
 void main(void)
 {
+       void*  buf = k_heap_alloc(&payload_heap, 1024, K_NO_WAIT);
        printk("Hello World! %s\n", CONFIG_BOARD);
+       k_heap_free(&payload_heap, buf);
 }
```

And then: 

```
SEGGER J-Link RTT Logger
Compiled Jul  8 2021 18:23:17
(c) 2016-2017 SEGGER Microcontroller GmbH, www.segger.com
         Solutions for real time microcontroller applications

Default logfile path: /home/carles/.config/SEGGER

------------------------------------------------------------

Device name. Default: NRF52 > Target interface. > Interface speed [kHz]. Default: 4000 kHz > RTT Control Block address. Default: auto-detection > RTT Channel name or index. Default: channel 1 > Output file. Default: RTT_<ChannelName>_<Time>.log >
------------------------------------------------------------
Connected to:
  J-Link OB-SAM3U128-V2-NordicSemi compiled Feb  2 2021 16:47:20
  S/N: 683258040

Searching for RTT Control Block...OK. 3 up-channels found.
RTT Channel description:
  Index: 0
  Name:  Terminal
  Size:  1024 bytes.

Output file: /dev/stdout

Getting RTT data from target. Press any key to quit.
------------------------------------------------------------

*** Booting Zephyr OS build v2.7.0-rc2-72-g7cf0a5c4a188  ***
Transfer rate: 0 KiByte/s Bytes written: 61 Byte   ^C
Shutting down...
```

carlescufi, 2021-09-22:
@trueb2 your example hints a the boot banner always being displayed (the line starting with `*** Booting Zephyr OS`) but `LOG()` statements sometimes not showing. This might be then due to some configuration of the logging subsystem, and not a problem with the RTT transport, but I cannot reproduce this based on the info you have posted either.

carlescufi, 2021-09-23:
@trueb2 can you confirm if this is an issue only with `LOG()` statements but `printk()` works fine?

carlescufi, 2021-10-05:
Closing since there is no activity and cannot be reproduced. Please reopen if this is still an issue.

abxy, 2022-03-23:
I think this should be re-opened. I have encountered this issue too.

Like @carlescufi I failed to reproduce it using the above steps because the `_SEGGER_RTT` symbol was not located at  high enough address. I managed to reproduce it by allocating a large buffer with a large alignment, causing it to sort before `_SEGGER_RTT`  within the .bss section. I believe this method should be reliable across different environments. 

Patch to reproduce:

```
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..5541d6f21b 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,5 @@
 # nothing here
+CONFIG_USE_SEGGER_RTT=y
+CONFIG_RTT_CONSOLE=y
+CONFIG_UART_CONSOLE=n
+CONFIG_SERIAL=n
diff --git a/samples/hello_world/src/main.c b/samples/hello_world/src/main.c
index 987728bbf2..08e9adae1c 100644
--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -4,9 +4,16 @@
  * SPDX-License-Identifier: Apache-2.0
  */
 
-#include <zephyr.h>
+ #include <zephyr.h>
+ #include <sys/printk.h>
+
+__aligned(64) char very_large_buffer[220000];
 
 void main(void)
 {
-       printk("Hello World! %s\n", CONFIG_BOARD);
+       very_large_buffer[0] = 1;
+       while (1) {
+               printk("Hello World! %s\n", CONFIG_BOARD);
+               k_sleep(K_SECONDS(1));
+       }
 }
```

Build and launch debugger:
```
west build -b nrf52840dk_nrf52840
west debug
```

In a different shell, view RTT log using e.g. telnet. 
```
telnet localhost 19021
```

In the GDB shell print the address of `_SEGGER_RTT`, it should be rather high, closer to `0x20035e20`  than to  `0x20000000` as seen below. Then resume execution and notice that nothing is printed to the RTT log.

```
(gdb) print &_SEGGER_RTT
$1 = (SEGGER_RTT_CB *) 0x20035e20 <_SEGGER_RTT>
(gdb) continue
Continuing.
```

Halt execution in GDB again and set the RTT address. The buffered log should print in the telnet window and new new lines should print every second. 

```
(gdb) monitor exec SetRTTAddr 0x20035e20
Executed SetRTTAddr 0x20035e20
(gdb) c
Continuing.
```

abxy, 2022-03-24:
This issue https://github.com/zephyrproject-rtos/zephyr/issues/40433 gave me an idea for what I think would be a good solution.

If I update the runner args in board.cmake to use nRF52840_xxAA instead of nrf52 west debug for my sample "just works".

```
board_runner_args(jlink "--device=nRF52840_xxAA" "--speed=4000")
```

carlescufi, 2022-03-24:
> This issue #40433 gave me an idea for what I think would be a good solution.
> 
> If I update the runner args in board.cmake to use nRF52840_xxAA instead of nrf52 west debug for my sample "just works".
> 
> ```
> board_runner_args(jlink "--device=nRF52840_xxAA" "--speed=4000")
> ```

That is curious, do you know why this is the case @abxy? @anangl and @simtind does this ring a bell?
Because that is exactly what fixed the issue [here](https://github.com/zephyrproject-rtos/zephyr/issues/40433#issuecomment-998950074)

simtind, 2022-03-24:
As with a lot of older software that supported nrf52832 early on, nrf52 is likely a shorthand for nrf52832_xxaa in the jlink device selection. Rtt buffer probably ends up in upper ram regions not present in 52832, and so jlink never finds it, and cannot start communication.

West should provide the full device identifier to jlink whenever possible.

carlescufi, 2022-03-24:
> West should provide the full device identifier to jlink whenever possible.

@simtind from this list, right?

https://github.com/zephyrproject-rtos/zephyr/pull/44181#issuecomment-1077725249

PR #44181 (2022-03-25): boards: nrf: Fix jlink invocation strings
The jlink executable expects a fully qualified IC name, and not just the
family.

Fixes #37294
commit 1568b665 (2022-03-25): boards: nrf: Fix jlink invocation strings

The jlink executable expects a fully qualified IC name, and not just the
family. Apply to all Nordic boards.

Fixes #37294
closed 2022-03-25 by carlescufi