zephyrproject-rtos/zephyr #37305: Bluetooth Direction Finding Support of "AoA RX 1us"

saleh-unikie, 2021-07-29:
**Describe the bug**
By running the "bluetooth direction finding example" in "AoA RX mode" as you can find in this link:
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/direction_finding_connectionless_rx

Two different slot duration can be chosen. 2 micro second (us) and 1 micro second (us). By default it is set to 2 us and works fine, but when it has been set to 1 us, it returns "EINVAL" error. To change the default value (2us) to 1us, this line should be changed to 0x1:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_rx/src/main.c#L247

The error is generated when it  tries to validate the arguments at this line:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/direction.c#L224

The above line return false when consequently is expanded to the below code, using the defined macros:
`((df_ant_info.switch_sample_rates) & BIT((BT_HCI_LE_1US_AOA_RX)))`  

The `BT_HCI_LE_1US_AOA_RX` itself is defined as `BIT(2)` as you can find here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/bluetooth/hci.h#L1614

So this "bit shift operation" has executed two times and is the source of the bug.

**To Reproduce**
Navigate to the "direction_finding_connectionless_rx" project and build it.
1. cd zephyr/samples/bluetooth/direction_finding_connectionless_rx/
2. Change the slot duration from 0x2 to 0x1: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_rx/src/main.c#L247
2. west build -b nrf52833dk_nrf52833
3. west flash
In addition to the receiver, a transmitter is needed too. Navigate to "direction_finding_connectionless_tx" project, build and flash it into another device.
1. cd zephyr/samples/bluetooth/direction_finding_connectionless_tx/
2. west build -b nrf52833dk_nrf52833
3. west flash

By opening a terminal to the receiver device, you will see the error as described in "Logs and console output" section.

**Expected behavior**
Enabling of receiving CTE should return without any error.

**Impact**
It is not possible to use the CTE based direction finding with 1us slots. Both 1us and 2us slots are defined and described by the Bluetooth SIG.

**Logs and console output**
Open a terminal connection to the receiver device, by rebooting the device, you will see this log which shows the program has started successfully:
`*** Booting Zephyr OS build v2.6.0-rc1-ncs1-3-g0944459b5b62  ***
Starting Connectionless Locator Demo
Bluetooth initialization...success
Scan callbacks register...success.
Periodic Advertising callbacks register...success.
Start scanning...success
`

Now turn on the transmitter, which send Bluetooth advertisement data to the receiver, at this time you will see the error:
`...
Waiting for periodic sync...
success. Periodic sync established.
Enable receiving of CTE...
failed (err -22)
Scan disable...Success.
...`

**Environment (please complete the following information):**
- OS: Linux
- Toolchain: nRF Connect SDK
- v2.6.0

**Additional context**
Add any other context about the problem here.

saleh-unikie, 2021-07-29:
#37308 A pull request is made upon the reported bug

ppryga, 2021-07-29:
@saleh-unikie Thank you for reporting the bug and provide PR with fix.

closed 2021-07-30 by cfriedt