zephyrproject-rtos/zephyr #37306: fix bogus commit message

cfriedt, 2021-07-29:
#37151 was merged with a bogus commit message including misrepresentation of the author. It's likely this was just due to incorrect usage of git.

To resolve this, please create a new PR with 2 commits:
1. revert 160c6a8d68b5a56aca5d6b682b7c26d96a97da04
1. same contents as the commit above but a message that follows the [commit guidelines](https://docs.zephyrproject.org/latest/contribute/index.html#commit-guidelines)

cfriedt, 2021-07-29:
Thanks for finding this @henrikbrixandersen , @nvlsianpu

scottwcpg, 2021-07-29:
@cfriedt 
I believe this happened during PR conflict resolution don't know which one.
I tried to follow your requirement to create two PR's but failed as usual.
git revert 160c6a8
Added message stating it was a git usage mistake probably during merge conflict resolution of previous PR's
Added the file that used to be in the revered commit
Created a new commit 
Sync'd with tree using git pull --rebase upstream main
This resulted in a conflict with the new file.  Fixing the conflict and finishing the rebase resulted in a single revert commit
I have not pushed this as I don't believe this is what you want.
What is the correct sequence of git commands to do what you want?

cfriedt, 2021-07-29:
@scottwcpg - it appears that someone has already deleted commit 160c6a8d68b5a56aca5d6b682b7c26d96a97da04 from git history, so you will need to make a new PR. I'm going to close this issue for now.

closed 2021-07-29 by cfriedt