zephyrproject-rtos/zephyr #37309: ARC: add MPU v6 (and others) support

ruuddw, 2021-07-29:
**Problem & Solution**
Currently ARC MPU v2 and v4 are supported. v4 supports the SecureShield programming model. The v2 programming model is compatible with later versions of the ARC MPU, but doesn't support all features (e.g. more regions, smaller minimal region size, additional memory attributes).
Support for recent MPU versions should be added.

**Additional context**
The trigger for this is a request for ARC MPU v6 support (and userspace) for ARC v2 HS. Highest priority is support for the 32 regions (banked programming model). But also support for the cacheability attributes is requested.

jacoba-at, 2021-08-06:
Would this enhancement include support for MPU in HSDK, or should this be a separate request?
Also I noticed that embarc_osp added support for MPU V3 for HSDK.
Should that be also added as part of the MPU support for HSDK?

ruuddw, 2021-08-09:
The request already mentions 'recent versions'. V3 should work with current V2 driver, just not supporting the minimum region size of 32 bytes.
ARC development boards that include the hardware are typically supported if new features are added. Thanks for mentioning HSDK specifically, v2 indeed includes the MPU. It makes sense to add a configuration with MPU to Zephyr.
What's your timeline? We currently don't have this enhancement listed for 2.7.

abrodkin, 2021-08-12:
@jacoba-at I'm a bit confused so your comments are much appreciated.
Initially @ruuddw says:
> Highest priority is support for the 32 regions (banked programming model).
> But also support for the cacheability attributes is requested.

And both above items ("32 regions" & "cacheability attributes") are features of the MPUv6.
But you were then mentioned MPUv3 (the one we have on HSDK-4xD).

So could you please confirm which features are needed & which MPU version do you really have or will have in your HW so that we may work on the right ones ;)

IRISZZW, 2021-08-13:
for convenience, ARC MPU features table:
|                      | MPU V2 | MPU V3 | MPU V4 | MPU V6 | MPU V7 |
|----------------------|--------|--------|--------|--------|--------|
| max region number    |   16   |   16   |   16   |   32   |   32   |
| min region size      |   2K   |   32   |   32   |   32   |   32   |
| secure               |        |        |    √   |        |        |
| non-cacheable region |        |        |        |    √   |    √   |
| serializable region  |        |        |        |    √   |    √   |
| non-volatile region  |        |        |        |        |    √   |

jacoba-at, 2021-08-13:
@abrodkin Our SoC is based on HS48 and we use MPU V6 with 32 regions. This is our top priority.
We use HSDK-48 evaluation board as a reference and we will continue to use it for SW development that is not tightly coupled with our SoC, but quite similar. We will do so since our SoC is not available yet and we develop low level SW for it and test in a simulated environment.
So we would like to be able to run userspace applications on HSDK-48, but the priority is for the V6 features.

abrodkin, 2021-08-16:
@jacoba-at ok, got it. Stay tuned and hopefully we'll get something for you to try soonish ;)

jacoba-at, 2021-08-30:
@abrodkin I tested the code and was able to set 32 MPU regions on our SoC. You can close this.
Thanks for getting it in!

abrodkin, 2021-08-30:
Fixed with https://github.com/zephyrproject-rtos/zephyr/pull/37871

closed 2021-08-30 by abrodkin