zephyrproject-rtos/zephyr #37319: West 0.11.0 fails in Zephyr doc build under other manifest repo & renamed Zephyr fork

gregshue, 2021-07-29:
**Describe the bug**
West 0.11.0 failures cause my documentation build of Zephyr v2.6.0 to fail.  There are three distinct differences in `my- workspace`:
1. The manifest repository is "my-application", mounted as `my-workspace/my-application`
2. The Zephyr code fork is in the repository "my-zephyr", mounted as `my-workspace/zephyr`
3. I am building the v2.6.0 documentation from `my-workspace/zephyr/doc/`.

The relevant fragment of the failure is:

> ...
> Running Sphinx v3.3.1
> Warning: Could not extract kernel version
> 
> Exception occurred:
>   File "/home/zephyr/.local/lib/python3.8/site-packages/west/manifest.py", line 1419, in get_projects
>     raise ValueError(unknown, uncloned)
> ValueError: (['zephyr'], [])
>  

I found two issues in manifest.py and locally fixed them:

* `1388: mpath: Optional[Path] = Path(mp.path).resolve()` creates a path to a new subdirectory, where it should create the path to where the existing manifest repository is mounted.

  When I locally change this line to be `mpath: Optional[Path] = Path(mp.abspath).resolve()` it now points to the right location for my setup.
  
* `1401: rpath = Path(pid).resolve()` creates a path to a new subdirectory, where it should create the path to where the existing project is mounted.

  When I locally I change this line to be `rpath = Path(self.topdir).joinpath(pid).resolve()` it now seems to resolve properly.

When both of the above changes are in place, the documentation generation of `my-workspace/zephyr/doc/` completes without issue.

**To Reproduce**
Steps to reproduce the behavior:
* See above description.

**Expected behavior**
I expect `west` to properly find and report repositories no matter how they are described in the manifest.

**Impact**
Showstopper.  Document generation failures are now blocking everyone on my team. We use the Zephyr documentation generation system to also generate documentation for `my-application`, needed to support developers and for release artifact generation.  If the problem were in Zephyr itself, I would patch my fork of Zephyr. Since `west` is outside of `my-application`, the fix is not easily distributed everywhere it is needed.  

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.2
 - Commit SHA or Version used: v2.6.0

**Additional context**
Add any other context about the problem here.

gmarull, 2021-07-30:
I think the scenario described here is not supported by Zephyr documentation, and I don't see how it can be supported. Applications can have their own docs and link to Zephyr (using intersphinx if Sphinx is also used for application). I'd suggest to look at how nRF SDK builds the Zephyr docs alongside other docsets: https://github.com/nrfconnect/sdk-nrf.

gregshue, 2021-07-30:
@gmarull 
The scenario described here has nothing to do with any application docs.  I found this trying to build only Zephyr documentation from the `zephyr/docs/` folder of my workspace tree using the `zephyr/docs/Makefile` command.  I just happen to have my manifest in a different repository, and added a prefix to the name of my fork of Zephyr. 

What part of this scenario is not supported by `west`?

mbolivar-nordic, 2021-07-30:
Thanks for the report. The west get_projects() method is indeed doing the wrong thing, but that's a west issue, not a zephyr issue.

I've opened https://github.com/zephyrproject-rtos/west/issues/523 to track the west issue and https://github.com/zephyrproject-rtos/west/pull/524 to fix it.

Let's discuss the west portions in either that issue or pull request, not here in the zephyr repository.

mbolivar-nordic, 2021-07-30:
Here is a separate comment about the docs build issue.

For background: The ValueError you reported is from doc/_extensions/zephyr/link-roles.py's setup() function.

This function uses Manifest.get_projects() to try to find the zephyr repository URL as it is configured in the current workspace by looking up the 'zephyr' project. That lookup will find any project named 'zephyr' in the workspace's manifest, and use its URL when forming URLs for `:zephyr_file:` or `:zephyr_raw:`.

It also isn't passing `allow_paths=False` to `get_projects`, which it probably should be. Unless it's calling get_projects with $PWD set to the parent directory of the zephyr repository, and the intention that the zephyr repository is always named 'zephyr', which seems ... like a bad assumption.

> What part of this scenario is not supported by west?

I would say the right question here is "what part of this scenario is not supported by link-roles.py".

And the answer is:

>  I just happen to have [...] added a prefix to the name of my fork of Zephyr.

For reasons that I hope are clear given the above background.

mbolivar-nordic, 2021-07-30:
> It also isn't passing `allow_paths=False` to `get_projects`, which it probably should be.

https://github.com/zephyrproject-rtos/zephyr/pull/37343 makes this change.

gregshue, 2021-07-30:
> It also isn't passing allow_paths=False to get_projects, which it probably should be. 

It seems to me it probably should **NOT** pass allow-paths=False. It is unreasonable to force every fork of Zephyr to also be named "zephyr" (especially since we may need multiple forks in the same SCM organization/project).  It is tolerable to have any Zephyr fork mounted at `zephyr/`.

Does this mean that Zephyr documentation generation (and eventually requirements tracing) is dependent on `west`?

mbolivar-nordic, 2021-07-30:
> It seems to me it probably should NOT pass allow-paths=False. It is unreasonable to force every fork of Zephyr to also be named "zephyr" (especially since we may need multiple forks in the same SCM organization/project). It is tolerable to have any Zephyr fork mounted at zephyr/.

Even if that's what you want, the current code would still be wrong, due to the relative path / current working directory issues I have described.

> Does this mean that Zephyr documentation generation (and eventually requirements tracing) is dependent on west?

No. West is optional, even for the docs build.

mbolivar-nordic, 2021-07-30:
@gregshue please close this issue if you are satisfied with the explanation

closed 2021-07-30 by gregshue