zephyrproject-rtos/zephyr #37324: subsys/mgmt/hawkbit: Unable to finish download

ycsin, 2021-07-30:
**Describe the bug**
I'm currently trying the `hawkbit` sample for the OTA on my custom board. For some reason the download process is extremely long and the gap between incoming data is increasingly long. The first handful chunks of data is normal, after that the next chunk would arrive at ~30s later, then ~1 min later, then about 2 mins later and eventually it would fail after downloading about 20kB or after ~17 minutes.

After trying a few things, I found that the download is slow only if it tries to write the downloaded buffer into the flash [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L807). If I comment that line out, the download is actually pretty fast and typically managed to download everything (251kB) under 40 seconds.

To determine the time [`flash_img_buffered_write()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L807) took to finish, I tried to calculate the elapsed time using `k_uptime_get()` before and after the function call, and the log system tells me that it took 0 seconds (rounded). And now I I'm not sure how to debug this and get it working.

From my current understanding, the `hawkbit` will do a `http GET` request to the download link using the `http_client_req` then the connection/socket should be opened and the image chunks will be pushed continuously from the server without requiring ack from the client (I could be wrong here). The `hawkbit` client will simply write the received data into the flash until the `HTTP_DATA_FINAL` is received. I don't know how a function that literally returned immediately can affect this download process.

**The setup that I use:**

- STM32G0B1RE custom board
- Quectel EC21 LTE modem (`gsm_ppp` driver)
- `image-0` partition is 416 kB in the internal flash
- `image-1` partition is 416 kB in the external `spi-nor` flash

**To Reproduce**
Steps to reproduce the behavior:
1. Build and flash `hawkbit` sample

**Expected behavior**
I expect the download to finish under 40 seconds.

**Impact**
Unable to use `hawkbit`

**Logs and console output**
Recapturing, coming soon.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNUARMEMB
 - v2.6.0-rc2-107-gebe282b02d9f
   - I applied this [commit](https://github.com/zephyrproject-rtos/zephyr/commit/e45c07cd60d05256c4d33d52a0f164b9908dd6b4#diff-51dbf7bd95ecafa1efbee72465fe2ba520297cbc23472b83e6c793c1eb74d384) manually.

ycsin, 2021-07-30:
@jukkar Any comments on the `http_client` part?
@Navin-Sankar @burumaj  Any idea?

ycsin, 2021-07-30:
There was a time where it downloaded 20+% of the image before there's a hard fault, and doesn't reach that far again after a reset, quite random.

ycsin, 2021-08-05:
To verify the write speed of the `spi-nor`, I wrote a simple program to write a total of 251 kB of 512 bytes dummy string into the `spi-nor` flash's `image-1` and here is the result:

Write that requires a page erase: **43ms**
Write that doesn't require a page erase: **8ms**

**Log:**

```
[00:00:11.298,000] <inf> main: ==================================================
[00:00:11.298,000] <inf> main: Write test ended for AT25DF081A
[00:00:11.298,000] <inf> main: ==================================================
[00:00:11.298,000] <inf> main: Total elapsed time: 6 s (6291 ms)
[00:00:11.298,000] <inf> main: Write speed: 41 kb/s
[00:00:11.298,000] <inf> main: ==================================================
[00:00:11.298,000] <inf> main: Reading back to verify written data
[00:00:13.151,000] <inf> main: ==================================================
[00:00:13.151,000] <inf> main: Data matches
[00:00:13.151,000] <inf> main: ==================================================
[00:00:13.151,000] <inf> main: Total elapsed time: 1 s (1853 ms)
[00:00:13.151,000] <inf> main: Read speed: 251 kb/s
[00:00:13.151,000] <inf> main: ==================================================
```

**Speed test program**
[main.c.txt](https://github.com/zephyrproject-rtos/zephyr/files/6936643/main.c.txt)

**Full log**
[spi-nor_speed-test.txt](https://github.com/zephyrproject-rtos/zephyr/files/6936638/spi-nor_speed-test.txt)

ycsin, 2021-08-05:
@pabigot , just wondering if you have any clue about this?

ycsin, 2021-08-05:
I increased the `CONFIG_IMG_BLOCK_BUF_SIZE` to 4096, which is the sector size of my external serial flash, then now it is able to download the whole firmware.

de-nordic, 2021-08-05:
@ycsin You are right there is a bug: the target image for upload is taken from the DTS, as the partition defined with "image-1" label, but the write size for the device is determined for the flash selected by chosen "zephyr,flash", which may be different device, as it probably happens here.
The BUILD_ASSERT, at the top of flash_img.c, fails to catch that the `CONFIG_IMG_BLOCK_BUF_SIZE` has not been set properly, because check is done against incorrect flash, as explained above.

ycsin, 2021-08-05:
@de-nordic after this I encountered another bug where if I flash an image to the primary slot over the updated image, it will result in a hard fault inside the bootloader. I debugged and traced this just now, it seems to happen in the swap scratch function. So for some reason the bootloader is thinking that there's an upgrade and then hard fault. I've left my workplace, will update this comment later.

I think there's another bug: after it booted up the updated image, which is unconfirmed, if I reset the device, the bootloader will boot right into the primary instead of doing a swap.

ycsin, 2021-08-11:
> I increased the `CONFIG_IMG_BLOCK_BUF_SIZE` to 4096, which is the sector size of my external serial flash, then now it is able to download the whole firmware.

I encountered the issue again today even after this change and removing the flash-write part didn't help, I tried the `big_http_download` again and it is still able to download the firmware in full, but the speed dropped to 3kBps somehow.

Downloading the file using my laptop is still very fast, still not sure where the issue lies. I guess I might have to work around this using partial downloads.

ycsin, 2021-08-11:
@Navin-Sankar Any comments?

d3zd3z, 2021-08-12:
That setting the download block size to match the page size eliminates the problem suggests to me that we are running into a situation where instead of just writing a page in pieces, we are erasing it and rewrite it for each of the smaller blocks.

Whether a page can be partially written does depend on the flash device in question. Most NOR flashes are able to write to parts of a page just without problem. Other devices do require the whole page to be written once between erases. If that is the case, I'm not sure there is a very good solution other than to buffer an entire page.

I didn't readily find the information about the STM part in question, but most of these allow a minimum write size of 8 or 16 bytes, so it should be possible to write to different 16-byte parts of a page without needing to erase the page.

ycsin, 2021-08-13:
@d3zd3z after a few days of testing, I'm guessing that the cause of this issue might be networking/socket related, instead of the `flash_img_buffered_write` function itself. The `spi-nor` flash that I'm using should be able to write parts of a page without problem, the STM flash minimum write size should be 8 bytes IIRC.

I do encounter a bootloader-related issue as mentioned in this [comment](https://github.com/zephyrproject-rtos/zephyr/issues/37324#issuecomment-893376518), I think I should create a separate issue for it.

ycsin, 2021-08-17:
After further testing I found that the download is slow when something is blocking during the reception of the data, such as `flash_img_buffered_write`, or if the log is set to immediate mode.

Increasing `CONFIG_IMG_BLOCK_BUF_SIZE` causes the write to happen less frequently and therefore improves the download speed, but this seems to be a workaround instead of a fix.

d3zd3z, 2021-08-20:
> After further testing I found that the download is slow when something is blocking during the reception of the data, such as flash_img_buffered_write, or if the log is set to immediate mode.

This makes me think of something. I haven't studied the STM driver all that much, but I know that some flash drivers make the flash device itself inaccessible during the flash operation (the flash goes away while being programmed). They generally run code out of RAM to do this, and run with interrupts masked. If well written (and the hardware supports it), they can interrupt the flash operation, make the code reappear, and handle the interrupt. Again I don't know the behavior here.

But, is it possible to figure out if the flash_img_buffered_write() might be blocking for a while and causing packets to be dropped? Dropped packets would slow down the transfer significantly, waiting for retransmits from the server.

rlubos, 2021-09-13:
@ycsin From the networking point of view, it could be the case, that due to increased CPU consumption the application thread is not able to consume the data fast enough, so we fill up all of the RX buffers. If the low-level network driver does not specify a timeout for the allocation, or the timeout is too small, it'd drop the packet, which would result in retransmission at TCP level and decreased performance.

We had a similar issue with one of the Ethernet drivers not long ago (see https://github.com/zephyrproject-rtos/zephyr/issues/36891#issuecomment-881503651) - the application did not catch up with the incoming traffic, which resulted in a packet drop. Adding a timeout for the `net_pkt_rx_alloc_with_buffer()` function in the Ethernet driver solved the problem.

Now, I'm not very familiar with the PPP implementation, but there seems to be a similar case with the [ppp driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/net/ppp.c#L103). Probalby worth checking if it's related?

ycsin, 2021-09-13:
> @ycsin From the networking point of view, it could be the case, that due to increased CPU consumption the application thread is not able to consume the data fast enough, so we fill up all of the RX buffers. If the low-level network driver does not specify a timeout for the allocation, or the timeout is too small, it'd drop the packet, which would result in retransmission at TCP level and decreased performance.
> 
> We had a similar issue with one of the Ethernet drivers not long ago (see [#36891 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/36891#issuecomment-881503651)) - the application did not catch up with the incoming traffic, which resulted in a packet drop. Adding a timeout for the `net_pkt_rx_alloc_with_buffer()` function in the Ethernet driver solved the problem.
> 
> Now, I'm not very familiar with the PPP implementation, but there seems to be a similar case with the [ppp driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/net/ppp.c#L103). Probalby worth checking if it's related?

Thanks heaps for the pointer. I'm not familiar with PPP nor networking, but this does sound like something I'm facing, I will try to test that in the ppp driver.

ycsin, 2021-09-14:
I think there are a few ways that can solve this issue, from the networking perspective:

1. As mentioned by @rlubos, increase the delay in ppp driver so that it waits longer for allocation.

and from the application perspective, in this case the hawkbit/stream_flash:

1. Format the slot-1 before starting to download so that the application will only have to write and do not have to erase, which reduce the CPU time spent in the application.
2. Increase the `CONFIG_IMG_BLOCK_BUF_SIZE` so that the application writes the flash less often, which also reduce the CPU time spent in the application, at the cost of RAM.

I will try the PPP fix as soon as I can. For the hawkbit part, I guess I will make a PR to erase the slot-1 prior to download at some point.

But all these seems to be some workarounds, I think the real issue is that both the PPP and the Hawkbit/application are using too much CPU time for the entire operation. I wonder if the data transmission from modem to memory in PPP and the data transmission from memory to SPI flash can be offloaded to DMA?

nvlsianpu, 2021-09-14:
Maybe worth to consider moving calls to `flash_img_buffer_write` to other thread with lower priority than image collection thread, or to the worqueue. (this maight be done in the flash steram module as well).As image you are collecting is in external flash this should does the job.

ycsin, 2021-10-04:
@rlubos some update on this, I've moved to a new version of my custom board and that uses AT45 flash with a page of 512B, the previous one (AT25) was 4KiB, this makes the format to happen quite frequently and caused the upgrade to fail, even with the [ppp]([url](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/net/ppp.c#L106)) delay at 200 msec. What is the trade off if I continue to increase this delay?

I tried to enable SPI DMA and sometimes I get this [error](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/net/ppp.c#L108).

ycsin, 2021-10-05:
Some more updates:
@nvlsianpu I tried to offload `flash_img_buffer_write` part ([L850 - L865](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/hawkbit/hawkbit.c#L849)) to a work, but it doesn't seem to help.

I think the CPU is simply too occupied by the networking stuff and UART interrupts due to firmware transfer: at least 7kiB per second, up to ~15 KiB per second from my observation. This is when I comment out the `flash_img_buffer_write` part. And it caused the log subsys to lag when the modem is transferring data.

rlubos, 2021-10-05:
> What is the trade off if I continue to increase this delay?

I think it depends on the driver/modem really, if packets keep coming and they are not consumed due to lack of RX buffer, the modem will eventually drop them. It could be the case that the respective driver/modem has buffering capabilities on it's own, but they're not unlimited.

So if we're not talking about sporadic CPU business causing delay, but a general CPU overload we should rather think of limiting the data flow at TCP level. Typically, you can limit the amount of data sent by the server by decreasing the Recieve Window size sent in the ACK messge. Unfortunately, from what I've seen, TCP2 implementation does not implement the feature, and sends a fixed value of IPv6 MTU all the time, effectively giving no limits to the server. I suggest to open an enhancement issue for the feature, as IMO it's quite valuable for constrained devices, and it seems to be a regressinon compared to the previous TCP implementation (where Recieve window handling was implemented).

ycsin, 2021-10-05:
> IPv6 MTU all the time

Is this feature limited to IPv6 only? or including IPv4?

rlubos, 2021-10-05:
> > IPv6 MTU all the time
> 
> Is this feature limited to IPv6 only? or including IPv4?

Well, I don't want to say for the author, but it seems that IPv6 MTU was just an arbitrary value chosen for the Window Size. It should not matter whether IPv4 or IPv6 is used at the IP layer.

ycsin, 2021-10-13:
I workaround this by enabling SPI async, I wonder if #39275 would be able to fix this for SOCs that doesnt support SPI async by limiting the download speed?

rlubos, 2021-10-14:
> I workaround this by enabling SPI async, I wonder if #39275 would be able to fix this for SOCs that doesnt support SPI async by limiting the download speed to what the CPU can gracefully process?

After testing that PR a bit it turned out that recv window handling will require a bit more effort to implement than proposed in the PR. So this is still an open topic.

ycsin, 2021-11-18:
I have tried to use hawkBit on a new board that uses the `esp_at` driver. And it seems to suffer the problem even though I already enabled SPI_ASYNC.

It stucked at 66% by default, if I increase the `CONFIG_WIFI_ESP_AT_MDM_RING_BUF_SIZE` to 2048 then it stucks at 90%+, so I guess it's the data overloading the CPU again.

I think the `esp_at` driver uses some form of offloaded networking instead of the native TCP2?

@rlubos @mniestroj any idea?

mniestroj, 2021-11-22:
> I think the esp_at driver uses some form of offloaded networking instead of the native TCP2?

Correct. TCP stack is implemented on ESP chip.

@ycsin Do you use UART hardware flow control?

ycsin, 2021-11-22:
> @ycsin Do you use UART hardware flow control?

@mniestroj  thanks for the reply.
No, I do not use hw flow control:

```mosel
/* WiFi */
&usart1 {
	pinctrl-0 = <&usart1_tx_pc4 &usart1_rx_pc5>;
	current-speed = <115200>;
	status = "okay";
	esp8266 {
		compatible = "espressif,esp-at";
		label = "esp8266";
		power-gpios = <&gpioc 6 GPIO_ACTIVE_HIGH>;
		reset-gpios = <&gpioc 10 GPIO_ACTIVE_HIGH>;
		status = "okay";
	};
};
```

mniestroj, 2021-11-23:
@ycsin Having HW flow control is quite critical, as it will suspend sending/receiving data stream when the other end is busy (e.g. processing previous packet). Without that, part of the frame can be easily dropped and there is no easy way to recover from that.

ycsin, 2022-03-01:
I believe this issue is now solved by #43018 and/or #39275. I'm able to download a 300+kB firmware and write that into flash using SPI interrupt API. Previously this is only possible using SPI async APIs.

closed 2022-03-01 by ycsin