zephyrproject-rtos/zephyr #37332: Increased power consumption for STM32WB55 with enabled PM and Bluetooth

r2r0, 2021-07-30:
**Describe the bug**
Power consumption is relitvely high in case when application is compiled with (enabled) options: CONFIG_PM, CONFIG_PM_POLICY_RESIDENCY, CONFIG_BT and CONFIG_BT_CENTRAL.
We observe this behavior on our device with our application but it can be easily reproduced on Nucleo devboard. 

**To Reproduce**
Use Nucleo WB55RG with blinky example.
Use following configurations:
1. As in original Zephyr sample
2. Only PM
CONFIG_GPIO=y
CONFIG_PM=y
CONFIG_PM_POLICY_RESIDENCY=y 
3. Only BT
CONFIG_GPIO=y
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
4. PM and BT
CONFIG_GPIO=y
CONFIG_PM=y
CONFIG_PM_POLICY_RESIDENCY=y
CONFIG_BT=y
CONFIG_BT_CENTRAL=y

I got following measurements (with LED disconnected and ST-Link jumpers removed) :
1. 2.2 mA
2. 0.3 mA
3. 4.9 mA
4. 6.9 mA
 
Before taking measurements I executed each application version with connected LED and in case 4 LED is not blinking.

**Expected behavior**
In case when PM residency policy is enabled and BT is configured but not used in application power consumption should be comparable with case when BT is not enabled in Zephyr.

**Impact**
showstopper

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.4
 - Commit SHA: 9143f4fd8c3588fe0d76ee447ab16f579f124919

**Additional context**
 - FUS FW version: 1.2.0
 - BLE FW version: 1.12.0

erwango, 2021-07-30:
@r2r0 Thanks for raising this PR.
For information, STM32CubeWB version available in Zephyr is 1.9.0.
As a consequence, STM32WB HCI lib is aligned on this version and so only tests with BLE FW version 1.9.0 have been performed. We're working on this and will update to 1.12.0 in this release, but this may takes some more weeks.

r2r0, 2021-08-02:
After downgrading BLE stack to 1.9.0.2 mesurements still shows mentioned problem:
1. 2.4 mA
2. 0.3 mA
3. 2.2 mA
4. 5.1 mA

FRASTM, 2021-08-02:
When setting the SHA1 9143f4fd8c3588fe0d76ee447ab16f579f124919, the sample does not blink on case4, (it blinks with case 3).
With the master on latest (997bbcb000c35f1d0de90668bb824bcce594daed),
case 3 and 4 give almost same values in average(2.9mA), when measuring power with STM32CubePowerMonitor.

FRASTM, 2021-08-03:
Case # 3 : BT but no PM on nucleo_wb55rg
[image]

FRASTM, 2021-08-03:
case #4 : BT and PM on nucleo_wb55rg
[image]

FRASTM, 2021-08-10:
Do you retrieve better average with BT and PM ?

r2r0, 2021-08-12:
Yes, with the head of Zephyr (as today) I got quite reasonable average current consumption (0.15mA).
Thanks a lot for your support.
BTW> Do you know what was the exact reason for such behavior?

FRASTM, 2021-08-18:
When the PM is on, the kernel tick is based on the stm32 lowPower Timer LPTIM1. The max sleeping period is then 2000ms and there is no more wakeup in between (about each 500ms like in case # 3)

erwango, 2021-08-25:
> Yes, with the head of Zephyr (as today) I got quite reasonable average current consumption (0.15mA).
Thanks a lot for your support.

@r2r0, don't hesitate to close the issue if satisfied.