zephyrproject-rtos/zephyr #37346: STM32WL LoRa increased the current in "suspend_to_idle" state

honestech74, 2021-07-30:
**Describe the bug**
When enabled the LoRa in STM32WL SoC, it draws more current than 70uA

**To Reproduce**
1. add the following configs to `samples/boards/stm32/power_mgmt/blinky`
```
CONFIG_SPI=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
```
2. build and flash to `nucleo_wl55jc` board
3. monitor the current at `JP1` on the Nucleo-WL55 board

**Expected behavior**
It should not draw more current than old configs when SoC is in suspend state.
That is, when LED is off, the current should be about `4uA` at test point(JP1).

**Monitored result(LED off)**
- When disabled LoRa, it draws about `4uA`
- When enabled LoRa, it draws about `75uA` in average, but there are triangle waves with `7.5ms` cycle. At peak, it's about `420uA`.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: 0.12.4
 - Zephyr v2.6.99

FRASTM, 2021-08-02:
@honestech74    _Zephyr v2.6.99_ does not exist yet ?? could it be then 1.6.99 which is very very old ...

Can you confirm that you have the `power.c` for the ./soc/arm/st_stm32/**stm32wl**
--> the PR   https://github.com/zephyrproject-rtos/zephyr/pull/36574 has introduced the power management for the stm32wl soc serie.

honestech74, 2021-08-02:
https://github.com/zephyrproject-rtos/zephyr/issues/37352#issuecomment-891067961

honestech74, 2021-08-02:
I have tested LoRa with HAL STOP2 mode(not using `CONFIG_PM`), it shows the same wave in the current measurement.
It seems like there are some enabled clocks or somethings even if the radio is sleeping and MCU is stopped.

nashif, 2021-08-03:
@Mani-Sadhasivam  can you please take a look.

Mani-Sadhasivam, 2021-08-04:
@honestech74 LoRa module (an `IP` incase of STM32WL) is put into sleep mode when it is not used for Tx/Rx. That means, only the current drawn by the Radio antenna and the module itself will be minimal. But for STM32WL, there are peripherals feeding the LoRa module like clock, SPI etc... If we could identify the power consuming sources, then we can try to put them to sleep by using `CONFIG_PM` in lora standalone driver for STM32WL.

That also requires `CONFIG_PM` support in relevant drivers also I believe. Maybe @erwango can provide more info.

erwango, 2021-08-23:
Moved isue from bug to feature request as power management is not fully available yet on all stm32 series nor devices

erwango, 2021-09-30:
@aurel32, @fabiobaltieri I know you had a look at low power on WL recently.
Is that part of the issues you've seen ?

aurel32, 2021-09-30:
Yes, it seems to be similar to the issue I encountered, which is fixed with PR #38758.

erwango, 2021-09-30:
> Yes, it seems to be similar to the issue I encountered, which is fixed with PR #38758.

Thanks @aurel32.
@honestech74, would you mind checking a recent main branch which includes #38758

honestech74, 2021-09-30:
@erwango I've double checked it, but unfortunately, it has not been reduced. I am seeing the same situation before update zephyr to latest.
@aurel32 @fabiobaltieri If you are getting another current when following the reproduce, let me show your result.

aurel32, 2021-09-30:
I am getting an average current of about 3µA, but this is not on this board, but instead on a Seeed Lora E5 module, so there might be some differences. Just looking quickly at the config difference between this board and what I have, have you tried modifying the configuration to include:
```
CONFIG_PM=y
CONFIG_SERIAL=n
CONFIG_CONSOLE=n
CONFIG_UART_CONSOLE=n
```

aurel32, 2021-09-30:
After removed `CONFIG_SPI=y`, I am getting 11uA. I am not sure now if LoRa would work without it or not, but will check later. @aurel32 @fabiobaltieri Thank you for the work.

Unfortunately CONFIG_SPI=y is needed to communicate with the LORA module.

fabiobaltieri, 2021-09-30:
Hey I did some measurement on my setup but unfortunately I have some other stuff on my board (also Seeed based) that are making a bit of noise and can't quite isolate the WL. That said I wasn't able to reproduce the problem and get a spike at that frequency. A quick thought, could it be that something is running and restarting on the second core?

About CONFIG_SPI raising the sleep current: not sure if it's the same for this one, but the SX1272 was coming out of reset in standby, so it needed an SPI command to go to sleep. May be the case here as well.

honestech74, 2021-09-30:
Oh, `CONFIG_SPI=y`, I am still getting about 77uA and peaks. I think it's related to this config.
@aurel32 Are you sure if you're getting low current(3uA) with LoRa working?

fabiobaltieri, 2021-09-30:
> Oh, `CONFIG_SPI=y`, I am still getting about 77uA and peaks. I think it's related to this config. @aurel32 Are you sure if you're getting low current(3uA) with LoRa working?

Sorry just to clarify, you are getting those spikes just with the plain `samples/boards/stm32/power_mgmt/blinky` + `CONFIG_SPI=y` and no LORA options?

honestech74, 2021-09-30:
|Sorry just to clarify, you are getting those spikes just with the plain samples/boards/stm32/power_mgmt/blinky + CONFIG_SPI=y and no LORA options?

> Sorry just to clarify, you are getting those spikes just with the plain `samples/boards/stm32/power_mgmt/blinky` + `CONFIG_SPI=y` and no LORA options?
Here are my prj.conf for `samples/boards/stm32/power_mgmt/blinky` without any changes.
```
CONFIG_PM=y
CONFIG_PM_DEVICE=y
CONFIG_PM_DEVICE_RUNTIME=y
CONFIG_SPI=y
CONFIG_LORA=y
CONFIG_LORA_SX12XX=y
CONFIG_SERIAL=n
CONFIG_CONSOLE=n
CONFIG_UART_CONSOLE=n
```

honestech74, 2021-09-30:
@aurel32 @fabiobaltieri For sure, can you get low power with the `samples/drivers/lora/send` with `CONFIG_PM=y`? If yes, let me show the scope like PPK.

aurel32, 2021-09-30:
First of all, just to make things clear, I do not have a Nucleo-WL55 board. I use a SeeedStudio LoRa E5 module without board, soldered with wires to the JTAG connector:
[image]

This module has a STM32WLE5JC SoC which only has a single Cortex-M4 core. The Nucleo-WL55 board has a STM32WL55JC SoC, with a Cortex-M4 coupled with a Cortex-M0. **This might makes a difference.**

I also do not have a way to measure the power consumption with high time resolution, I only get an average over a few seconds. Therefore I don't know if there are spikes or not, but given I am confident that the 3µA average is accurate to +/1µA and includes spikes if any (I have cross-checked multiple methods).

> @aurel32 @fabiobaltieri For sure, can you get low power with the `samples/drivers/lora/send` with `CONFIG_PM=y`? If yes, let me show the scope like PPK.

I have tested that with two modifications: 1) sending data every 15 seconds instead of 1 second and 2) use of my own "board" definition. I can confirm the power consumption goes to 3µA in between two LoRa transmission. All that said I only reach this value when I power cycle the device after programming it (maybe reset is enough). It seems that once the ARM SWD interface has been enabled, it consumes some power.

aurel32, 2021-09-30:
> 2) use of my own "board" definition

I have compared quickly the two DTS trees. I have found that the nucleo_wl55jc board definition default to a HSI clock with a PLL. On my side I am using the MSI running at 48MHz and without PLL.

When using the PLL instead, I see an increase of the power consumption to ~14µA, but that's far from your measurements. I guess it's due to the time for the PLL to start and stabilize each time the MCU wakes up. Anyway if you want to try that change, here are the relevant parts from my DTS file:

```
&clk_msi {
        status = "okay";
        msi-range = <11>;
};

&rcc {
        clocks = <&clk_msi>;
        clock-frequency = <DT_FREQ_M(48)>;
        cpu1-prescaler = <1>;
        ahb3-prescaler = <1>;
        apb1-prescaler = <1>;
        apb2-prescaler = <1>;
};
```

honestech74, 2021-10-01:
@aurel32 Thank you for the double checking.
I can definitely say that it doesn't work for nucleo_wl55jc board anyway. So there might be an issue on STM32WL55 SoC or something like HAL.
@fabiobaltieri I think that zephyr doesn't support Cortex-M0+ of STM32WL55, so there isn't any clock for it.

fabiobaltieri, 2021-10-02:
Tried to take some more measurement on my board (I have the Seed-E5 module, an nRF52832 a MAX17055, a LiPo charger and a mediocre LDO to deal with here...), I don't know how but while fiddling with wiring the PPK in, I managed to get my unit in a similar state as well, only difference being that the pulses were at ~5.6ms, but otherwise similar shape (and higher base consumption but that's prob the rest of my board). Would send you a screenshot but I can't get there anymore, I suspect that the supply bounced and the CPU crashed.

Can you try to take some measurement around the board trying to correlate the pulses? Specifically on the DCDC inductor (L13 on the Nucleo board): wondering if that pulse is the DCDC kicking in for some reason. If you could probe it on VLXSMPS or something... That may explain why it happens when the Radio get initialized.

About the other core: yeah Zephyr does not support it yet, was thinking it maybe it was starting up anyway and either crashing executing random opcodes or maybe running something that has been programmed with the Cube IDE. I don't know how the core starts either, but on the H7 there's an option to have the second core start automatically so I've figured let's make sure that's not the case here.

closed 2021-10-11 by honestech74

thiagohd, 2022-11-11:
> 

Can you share your solution please?

thiagohd, 2022-11-11:
> I could get 6uA with `tx-enable-gpios` and `rx-enable-gpios` disconnected when the radio is sleeping. For this, I've added simple gpio stuff to `samples/drivers/lora/send` with `CONFIG_PM`. 

@honestech74 Can you share your solution please?

honestech74, 2022-11-11:
@thiagohd You can use the [Pin Control](https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#pinctrl-guide) from zephyr v3.1.0. 

In the past, I set the gpios to `GPIO_DISCONNECTED` like `tx-enable-gpios` and `rx-enable-gpios` on `nucleo_wl55jc` board. Like this:
```
gpio_pin_set(tx_enable_gpio, GPIO_DISCONNECTED);
```
You might need the additional GPIOs due to your board and can add the low power stuff to your PM. Please take a look at `zephyr/soc/arm/st_stm32/stm32wl/power.c`