zephyrproject-rtos/zephyr #37352: STM32WL ADC doesn't work with power management

honestech74, 2021-07-31:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

FRASTM, 2021-08-02:
There is only a [AN5568](https://www.st.com/content/ccc/resource/technical/document/application_note/group1/89/e0/fa/82/05/f6/41/b2/DM00739353/files/DM00739353.pdf/jcr:content/translations/en.DM00739353.pdf) . Could the situation apply in this sample ?
"_The STM32WL devices embed a multiple 12-bit/2.50 Msps ADC. This very fast and accurate converter can jeopardize the battery lifetime if left powered-up. As the ADC consumption is roughly proportional to the acquisition frequency (around 200 μA/Msps), from a consumption standpoint, the application can choose between the following solutions:
– Perform the acquisition at low speed to limit the maximum current.
– Perform the acquisition at the maximum speed to switch back to ultra-low-power mode quickly.
When the acquisition is performed slowly, the ADC consumption itself can go down to few tens of a μA, drastically limiting the maximum current. This may be mandatory when the power source provides a limited current. If the CPU has no other task to perform during that time, this slow acquisition increases the time spent by the system in Run, Sleep, LPRun or LPSleep mode, versus the time spent in Stop or Standby ultra-low-power mode."_
Otherwise, even if the stm32wl is close to the stm32wb serie, the sequence to switch to HSI before entering Stop Mode is not mentioned for the stm32wlx. When testing the samples/drivers/adc on nucleo_wb55rg with CONFIG_PM is correct.

FRASTM, 2021-08-02:
Can you confirm that you have the `power.c` for the ./soc/arm/st_stm32/**stm32wl**
--> the PR   https://github.com/zephyrproject-rtos/zephyr/pull/36574 has introduced the power management for the stm32wl soc serie.

ABOSTM, 2021-08-02:
Zephyr v2.6.99 doesn't exist. Do you mean 1.6.99 ?
I tested on my side and it is working well:
board: nucleo_wl55jc 
sha1 : 17d2e9d084b2f02f8d39cd38eecdd7c0b5af59a2
I follow your **To Reproduce** steps

FRASTM, 2021-08-02:
@honestech74 Can you confirm that you have the power.c for the ./soc/arm/st_stm32/stm32wl
--> the PR #36574 has introduced the power management for the stm32wl soc serie.

honestech74, 2021-08-02:
> Zephyr v2.6.99 doesn't exist. Do you mean 1.6.99 ?

@ABOSTM Sorry for the confusing version. I am in master branch and you can check the version from this file.
https://github.com/zephyrproject-rtos/zephyr/blob/main/VERSION#L1-L5
Also commit SHA1: [`57690f5`](https://github.com/zephyrproject-rtos/zephyr/commit/57690f566e806c4f3cb956ce3594b6907e229fb4)

> Can you confirm that you have the power.c for the ./soc/arm/st_stm32/stm32wl

@FRASTM Sure, I have it and it works well for `boards/stm32/power_mgmt/blinky`

honestech74, 2021-08-02:
> Zephyr v2.6.99 doesn't exist. Do you mean 1.6.99 ?
I tested on my side and it is working well:
board: nucleo_wl55jc
sha1 : 17d2e9d
I follow your To Reproduce steps

@ABOSTM I have double checked on your commit sha, but couldn't get running and same in last commit of main branch.
If you followed **Reproduce** and got running, please check your SHA. I guess you are in another branch or miss `CONFIG_PM`?
I have another issue on I2C with `CONFIG_PM`.

ABOSTM, 2021-08-02:
@honestech74 ,
Let me change a bit the status of my tests:
After flashing the board (**west flash** with openOCD), it is working fine, even after pressing reset button, it works
But if I unplug and replug the board (ST-link usb powered), then it fails, **ADC reading: xxx** Displayed  only once in console.

honestech74, 2021-08-02:
@ABOSTM 
You are right, I can see it's running when flash using `stm32cubeprogrammer` with openocd, but stopped after unplug and replug.
Even if it's running, getting 1.2mA power consumption and there are high peaks every a second(there is `k_sleep(1000)` in sample) Is it reasonable?
[image]

I'd like to use `J-Link` so tested with it until now. It's stopped after show one ADC log.
I never saw this difference between programmers, I can't trust the result when using `stm32cubeprogrammer`.

honestech74, 2021-08-03:
@ABOSTM Can you test it using J-Link? It seems like there is something wrong in ADC driver. I've tested ADC without `CONFIG_PM` but stopped SoC using LL library and init the clock after wake up. But it's not running after wake up.

honestech74, 2021-08-07:
I could get ADC running with STOP0/1/2 mode with HAL library. I could stop SoC using LL_PWR_* functions and wake up the counter(RTC). But there are some operations to make it working. Before entering STOP mode, I have disabled ADC. When wake up, enabled ADC. I am not sure whether ther is any specification on ST datasheet and reference manual.

FRASTM, 2021-08-12:
It is mentioned that the content of the ADC registers is retained in STOP0/1 but not in STOP 2, so to be re-initialised.

FRASTM, 2022-01-13:
This seems specific to stm32wl55jc low power STOP2 mode, probably because "The content of all other peripherals is reset and must be reprogrammed."
```
&cpu0 {
	cpu-power-states = <&stop0 &stop1>;
};
```

sujayeshdev, 2022-01-13:
@FRASTM Thank you for your work.
I have understood that ADC registers are not retained in STOP2 mode and decided to use HAL instead of Zephyr PM.
If I stick to using Zephyr PM, can we have some hook for it? E.g. if we can register the callbacks for PM (maybe in `power.c`), users can register their custom code for ADC and other peripheral for STOP2 mode, and PM will call the callbacks before suspending and after wake up. So user can handle all peripheral without any issue with `CONFIG_PM`. I am not sure if Zephyr PM has such specifications.

honestech74, 2022-01-13:
@FRASTM Thank you for your work.
I have understood that ADC registers are not retained in STOP2 mode and got working by using HAL instead of Zephyr PM.
If I stick to using Zephyr PM, can we have some callbacks for it? E.g. if we can register the callbacks for PM (maybe in power.c), users can register their custom code for ADC and other peripheral for STOP2 mode, and PM will call the callbacks before suspending and after wake up etc. So user can handle all peripherals without any issue with CONFIG_PM and pure Zephyr style. I am not sure if Zephyr PM has such specifications.

FRASTM, 2022-01-17:
Yes, more or less, for each peripheral, entering/exiting STOP2 mode. 
I guess this is the purpose of the `pm_device` _Pointer to the device's power management resources,_ 
which appears in the `DEVICE_DEFINE`
like `PM_DEVICE_DT_DEFINE(SPI(idx), gpio_stm32_pm_action);`

nobodywasishere, 2022-02-22:
I believe I am also running into this issue with the LoRa E5 Dev Board (STM32WLE5JC), where it will just halt during `adc_read()` and fails to print anything (including `ADC Reading:`). Is this related to this issue?

erwango, 2022-02-23:
@nobodywasishere Do you enable PM ?

nobodywasishere, 2022-02-23:
@erwango PM is not enabled, no

erwango, 2022-02-23:
> @erwango PM is not enabled, no

So likely it is another issue. Please raise a dedicated ticket giving any information that would help.

Djammall, 2024-04-15:
The issues seems to still be present.

Djammall, 2024-04-15:
The issues seems to still be present on STM32WL series. I managed to solve the issue based on what @sense-Jo has done. 
 Changes are minor since the file adc_II_stm32.c evolved: 

-  Adding `
	/* enable device runtime power management */
	err = pm_device_runtime_enable(dev);
	if ((err < 0) && (err != -ENOSYS)) {
		return err;
	}`
at the end of `static int adc_stm32_calibrate(const struct device *dev)` in `adc_stm32_pm_action`.

- Adding `
	error = pm_device_runtime_get(dev);
	if (error < 0) {
		return error;
	}`
in `adc_stm32_read`

- Adding `pm_device_runtime_put(dev);` 
at the end of ` adc_stm32_isr(const` struct device *dev)`

What do you think about the changes ? Can it be interesting to make a PR ?

erwango, 2024-04-15:
@Djammall Sure, go ahead, please make a PR. @gautierg-st will be able to provide guiding during the review