zephyrproject-rtos/zephyr #37365: STM32 :DTCM: incorrect buffer size utilization

manoj153, 2021-08-02:
**Describe the bug**
I have allocated 128 kb of `uint8_t` array in which is the max size of the DTCM region but compilation memory usage is reporting utilization is 125kb/97.66%.   

I have tried memory validation routine using `memcmp` in which all the 128K iteration PASSED.. 

SoC : `STM32H723ZG` 

**Impact**
```
wrong info of mem usage
```
**Logs and console output**
Code to test (main.c)

```
#define size_my 128000
__dtcm_data_section uint8_t hello_buffer[size_my];

void main(void)
{
	LOG_WRN("start transfer");
	for (size_t i = 0; i < size_my; i++) {
		hello_buffer[i] = 0xAA;
	}
	LOG_WRN("done transfer");
	uint8_t first = 1;
	uint8_t *tmp = hello_buffer;
	uint8_t base_check = 0xAA;
	//hello_buffer[size_my - 5] = 0xCC;
	for (size_t i = 0; i < size_my; i++) {
		if (memcmp(tmp, &base_check, 1) != 0) {
			if (first) {
				LOG_ERR("I:%d", i);
				first = 0;
			}
		}
		tmp = tmp + 1;
	}
	LOG_WRN("done memcompare checking");
}
```
memory usage:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      298716 B         1 MB     28.49%
            SRAM:       80880 B       128 KB     61.71%
            DTCM:        125 KB       128 KB     97.66%
     BACKUP_SRAM:          81 B         4 KB      1.98%
        IDT_LIST:          0 GB         2 KB      0.00%

```

logs:
```
Booting Zephyr OS build v2.6.0-rc2-109-gf022d0a853a7  ***
[00:00:00.001,200] <wrn> MAIN: start transfer
[00:00:00.001,700] <wrn> MAIN: done transfer
[00:00:00.011,300] <wrn> MAIN: done memcompare checking

```

FRASTM, 2021-08-03:
could this be a question of 
when setting 
`#define size_my 128*1024`

memory usage is
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      146740 B         1 MB     13.99%
            SRAM:        4288 B       128 KB      3.27%
            DTCM:        128 KB       128 KB    100.00%
        IDT_LIST:          0 GB         2 KB      0.00%

```

manoj153, 2021-08-04:
Guys, I think it's not a bug,

I am truly sorry.

As @FRASTM mentioned, the unit KB is referring to = 1024 bytes.

One excercise I have done to prove this statement is take the memory address say :`octospi2`
`0x7FFFFFFF-0x70000000` = `268435455bytes` == `256Mbytes` so the generation is correct

Image shows address for reference
[image]

FRASTM, 2021-08-05:
I agree to close ! thanks @manoj153 for clarification

closed 2021-08-05 by manoj153