zephyrproject-rtos/zephyr #37368: Bluetooth: Host: Support receiving long advertising data

rugeGerritsen, 2021-08-02:
**Is your enhancement proposal related to a problem? Please describe.**
I want to be able to receive 1650 bytes of advertising data as supported by a given Bluetooth Controller. Currently the host forwards all advertising reports directly to the application regardless of the data status. Therefore, if the advertiser is transmitting more than 251 bytes, the application is unable to decode the data as it does not have the context describing if the data is first, intermediate, or last.

The `data_status` is present in both HCI LE Extended Advertising Report event (Core v5.3, Vol 4, Part E, Section 7.7.65.13) and HCI LE Periodic Advertising Report event (Core v5.3, Vol 4, Part E, Section 7.7.65.15) 

**Describe the solution you'd like**
To simplify the application layer design, I would like the host to recombine the received advertising data. The data shall be forwarded once the controller has indicated that there is no more data to follow. 

This requires the host to allocate a scan buffer to store the intermediate received advertising data. If the host supports receiving periodic advertising data, it requires one more.

Once intermediate advertising data has been received, this data shall be copied over to the scan buffer allocated in the host.

The size of the scan buffers should be configurable.

**Describe alternatives you've considered**
Instead of recombining advertising data in the host, this can be done in the application. That requires the data status to be forwarded. This approach is more error prone than letting the host recombine the data. If the application forgets to validate the data status flags, it may try to parse an intermediate fragment as if the data is complete.

Forwarding the data status to the application is implemented in https://github.com/zephyrproject-rtos/zephyr/pull/37336.

cvinayak, 2021-08-05:
@rugeGerritsen Add the Fixes with this issue number in the PR so that this issue gets linked and gets closed when the PR is merged.

rugeGerritsen, 2022-02-07:
Fixed as part of https://github.com/zephyrproject-rtos/zephyr/pull/41337

closed 2022-02-07 by rugeGerritsen